ReFirmLabs/binwalk #213: jffs2 mtd images fail to extract for Altera SOCFPGA

jameshilliard, 2016-06-05:
I have a device with this partition layout.

    cat /proc/mtd
    dev:    size   erasesize  name
    mtd0: 01000000 00020000 "NAND Flash Boot Area 16MB"
    mtd1: 01000000 00020000 "NAND Flash Boot Area backup1 16MB"
    mtd2: 01000000 00020000 "NAND Flash Boot Area backup2 16MB"
    mtd3: 08000000 00020000 "NAND Flash jffs2 Root Filesystem 128MB"
    mtd4: 05000000 00020000 "NAND Flash jffs2 Root Filesystem 80MB"

I dumped the partitions by using dd on [/dev/mtd0](https://bitbucket.org/jhilliar/s9rev/raw/6e460e4f948e453124ab77c3f5c80ef7baa05ddd/mtd0) [/dev/mtd1](https://bitbucket.org/jhilliar/s9rev/raw/6e460e4f948e453124ab77c3f5c80ef7baa05ddd/mtd1) [/dev/mtd2](https://bitbucket.org/jhilliar/s9rev/raw/6e460e4f948e453124ab77c3f5c80ef7baa05ddd/mtd2) [/dev/mtd3](https://bitbucket.org/jhilliar/s9rev/raw/6e460e4f948e453124ab77c3f5c80ef7baa05ddd/mtd3) and [/dev/mtd4](https://bitbucket.org/jhilliar/s9rev/raw/6e460e4f948e453124ab77c3f5c80ef7baa05ddd/mtd4)

I can mount some of the partitions locally on the device using:

    mount -t jffs2 /dev/mtdblock4 /mnt/upgrade/

However binwalk fails to extract any of the partitions though but does at least appear to identify the filesystem

    DECIMAL       HEXADECIMAL     DESCRIPTION
    --------------------------------------------------------------------------------
    0             0x0             JFFS2 filesystem, little endian

When running extraction it seems jefferson gets stuck.

Device is based on a Altera SOCFPGA

Is this using some sort of custom jffs2 packing that jefferson isn't handling properly?

devttys0, 2016-06-05:
If it mounts properly, then I wouldn't expect that they're doing any any custom packing. This should probably be reported to the [jefferson project](https://github.com/sviehb/jefferson) though.

sviehb, 2016-06-23:
I couldn't access to files you linked, but I recently committed some fixes to jefferson. Please let me know if it works with the most recent version.
Jefferson can take a while when processing large files.

jameshilliard, 2016-06-26:
@sviehb oh, I accidentally marked that repo private, should be able to download now, yeah could have just been it taking a while to process, I didn't let it run all that long, it seems to extract the [update](https://bitbucket.org/jhilliar/s9rev/raw/27bf3d5848cf9c6d1cd89bd3489462f99f4b2115/s9-20160608-650M-release/angstrom_rootfs.jffs2) images at least

closed 2016-08-16 by jameshilliard