ReFirmLabs/binwalk #215: Repacking of firmware after extraction

abduct, 2016-06-12:
## Feature Request:

Is it possible to add a repack option so that after running an extraction and modifying the contents we can repack the image back together?

## General Question: 

How could this be done manually? For instance if I have the signature output of:
```

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
13632         0x3540          U-Boot version string, "U-Boot 1.1.4 (Aug 17 2012 - 15:21:07)"
13680         0x3570          CRC32 polynomial table, big endian
14976         0x3A80          uImage header, header size: 64 bytes, header CRC: 0x71A58DBE, created: 2012-08-17 07:21:08, image size: 32798 bytes, Data Address: 0x80010000, Entry Point: 0x80010000, data CRC: 0xF13C6F0C, OS: Linux, CPU: MIPS, image type: Firmware Image, compression type: lzma, image name: "u-boot image"
15040         0x3AC0          LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 93688 bytes
131584        0x20200         LZMA compressed data, properties: 0x6D, dictionary size: 8388608 bytes, uncompressed size: 2753096 bytes
1039828       0xFDDD4         Squashfs filesystem, little endian, version 4.0, compression:xz, size: 2684880 bytes, 786 inodes, blocksize: 262144 bytes, created: 2014-02-04 10:24:21
3735552       0x390000        JFFS2 filesystem, big endian

```

And I wish to extract, modify, and repack the Squashfs filesystem where my data resides, what would be the steps in doing so?

Should I split the binary in three parts: 1) 0x000000 -> 0xFDDD4, 2) 0xFDDD4 -> 0x390000 3) 0x390000 -> end of binary. Proceed to extract the 2nd part (the squashfs file system), modify it, and resquash it with the same settings. Then finally merge the 3 parts back together again?

Or is that not the right way to go about this?

My objective is to modify some configuration files in the squashfs portion of the binary to open up a way into the device since all other options are closed.

Thanks.

benjaminlevine, 2016-08-12:
I would also like some more info on how to repack an image.
I extracted a squashfs, modified some config files and then repacked it with the same settings as the original.
I padded the resulting squashfs with 0xff to make it the same length as the original.
To get it back together again I used dd to copy the squashfs back into the bin. 
I used:
cat modified.squashfs | dd conv=notrunc of=original.bin bs=1 seek=$((0x2A0000))

where 0x2A0000 was the start of my Squashfs (so 0xFDDD4 in your case).

However, the resulting binary would not boot on my device. Still debugging

ArionMiles, 2016-08-13:
@benjaminlevine would you mind explaining to me how you "padded the resulting squashfs with 0xff to make it the same length as the original"?

Apart from that, I also tried the same method. Upon using `binwalk -e firmware.bin` on the final, modified firmware binary, in the `/_firmware.bin.extracted/` directory, I saw that all the files (including the modified files) that were previously neatly organized into different folders like `etc, tmp, var, www, bin, usr` in the stock firmware were present inside the `_firmware.bin.extracted` directory with no order whatsoever. There was also a `squashfs-root` directory present which had the same files (including the modified ones) in perfect order like in the stock firmware binary.

I believe that appending the modified filesystem on the stock binary seems to create this mess and have two copies of each file.

abduct, 2016-08-13:
@ArionMiles I am not sure how @benjaminlevine padded their firmware, but I wrote a quick script to do it for me:

```
original_file_size = File.size '/home/original/firmware/location/blob'
repacked_file_size = File.size '/home/repacked/firmware/location/blob'

File.open('/home/repacked/firmware/location/blob', 'ab') do |file|
  (original_file_size - repacked_file_size).times do
    file.write "\xFF"
  end
end

puts "Padded the repacked binary with (#{original_file_size - repacked_file_size}) bytes of 0xFF"
```

I haven't worked on this much as of late, but I've still been unsuccessful with repacking a modified firmware and loading it back onto the device via flashing with an external programmer.

Any more tips or advice would be appreciated.

devttys0, 2016-08-13:
This isn't a binwalk related question, as binwalk does not re-build binary images. However, you might want to take a look at the firmware mod kit, which uses binwalk for extraction and can re-build many router firmware images automatically (it may even work on this particular firmware): https://code.google.com/archive/p/firmware-mod-kit/

closed 2016-08-13 by devttys0

abduct, 2016-08-14:
So I take it you refuse this feature request as mentioned in the first post?

ArionMiles, 2016-08-14:
Firmware mod kit has long been abandoned. It doesn't work for me, and I can't even submit an issue to anyone. Would be nice if someone created a new firmware mod kit, possibly forking the previous one and adding more features on it.

devttys0, 2016-08-15:
@abduct Correct. Unfortunately what you're asking is not a trivial undertaking; IMHO there are too many oddball firmwares and edge cases to make such a tool generic and reliable. If one were to attempt such a utility, it would best to make it a completely separate project.

The only tool that I'm aware of that has attempted this is the firmware-mod-kit; as @ArionMiles mentioned it has not been updated for some time, but if you're looking for something that can handle just the most common/simple/generic cases it may be just what you're looking for.

ArionMiles, 2016-08-15:
This problem isn't directly binwalk related but please give it a read and if you can help, please reply.

The only problem I'm facing at this point is when I extract the filesystem with binwalk into a `squashfs-root` directory and squash it back into a `filesys.squashfs` using `mksquashfs squasfs-root filesys.squashfs` with lzma compression (as the original filesystem also uses lzma)

Now, I put this modified filesystem by two methods with same results:

#Method 1: 

`dd if=filesys.squashfs of=firmware.bin bs=1 seek=1180160 conv=notrunc` where `1180160` is the starting point of filesystem.

#Method 2:

I use the `tpl-tool` included in firmware mod kit for tp-link firmwares (this is a tp-link firmware I'm trying to modify)

I get a `firmware.bin-rootfs` file along with `-bootloader -kernel -header` files. I rename my `filesys.squashfs` to `firmware.bin-rootfs` and replace it with the firmware.bin-rootfs originally extracted by `tpl-tool` and run the `tpl-tool -b firmware.bin` and get a modifed `firmware.bin-new` file.

#RESULT:

Extracting the modifed firmware with binwalk in both the results gets me a `squashfs-root` folder where all the files previously arranged into `bin usr tmp var sbin www` and the likes are directly placed into the squashfs-root folder with no arrangement whatsoever. There are also the bin, usr, www, etc. folders inside this directory with same files. So my question is.... WHAT IS GOING ON?