ReFirmLabs/binwalk #217: Extraction problem with dev files

gj12, 2016-06-20:
Binwalk terminates if I extract CLX-6260FW printer firmware with binwalk -e -M 
```
Signature Exception: [Errno 6] No such device or address: '/root/mfc/s/_CLX6260FW_V4.00.02.29.hd-1.extracted/_440010C.extracted/dev/hdb12'
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/lib/python3.5/site-packages/binwalk/core/module.py", line 566, in main
    retval = self.run()
  File "/lib/python3.5/site-packages/binwalk/modules/signature.py", line 175, in run
    for fp in iter(self.next_file, None):
  File "/lib/python3.5/site-packages/binwalk/core/module.py", line 393, in next_file
    fp = self.config.open_file(next_target_file)
  File "/lib/python3.5/site-packages/binwalk/modules/general.py", line 200, in open_file
    peek=peek)
  File "/lib/python3.5/site-packages/binwalk/core/common.py", line 489, in BlockFile
    return InternalBlockFile(fname, mode=mode, **kwargs)
  File "/lib/python3.5/site-packages/binwalk/core/common.py", line 326, in __init__
    super(self.__class__, self).__init__(fname, mode)
OSError: [Errno 6] No such device or address: '/root/mfc/s/_CLX6260FW_V4.00.02.29.hd-1.extracted/_440010C.extracted/dev/hdb12'
----------------------------------------------------------------------------------------------------
```

devttys0, 2016-06-21:
This should be fixed in the latest commit to master (a442dc7e5b482225f8c813e09e58cb9bb05a9efd).

gj12, 2016-06-22:
It works! Thank you!

closed 2016-06-22 by gj12