ReFirmLabs/binwalk #220: not working for me

ralo21, 2016-07-25:
i guy , i have to disassemble a firwmare 

this : http://storow.xirodrone.com/attac-UA3500_FCU_FW_1.0.0_20150710.bin

i have installed ubuntu in virtualbox and binwalk 

when i analyze with binwalk i have blank line and when i extract nothing just blank line 

sorry for my english :( 

if you have idea :) 

thank you

devttys0, 2016-07-27:
There's nothing to extract from this firmware. There does not appear to be any compression, encryption, file systems, etc. It's just straight ARM code - load the .bin file directly into any ARM disassembler.

ralo21, 2016-07-27:
its crypted :(

devttys0, 2016-07-27:
I see nothing in the linked firmware image that suggests encryption of any type. There are plain text strings and code. What leads you to believe that it is encrypted?

ralo21, 2016-07-27:
I can't see the string :( what use for see that ?

Thank you

hoover2015, 2016-08-07:
Please don't help him. He's breaking the law. 

Zero Tech USA Co., Ltd
Add: 690 N. Berry St. Suite B,
Brea, CA 92821
Tel： (714) 592-7666

ralo21, 2016-08-07:
You can close , i have found how to thank you guys :)

closed 2016-08-07 by devttys0