ReFirmLabs/binwalk #221: Only a part of the BIN-files extracted

TheGotchi, 2016-08-06:
I would use binwalk for reengineering but only a part of the files are extracted from the firmware-bin. see what I was trying to do:

linux-fftx:/home/torsten/Documents/FirmwareCam # binwalk -Mer CH-sys-48.50.64.74.bin

Scan Time:     2016-08-06 18:45:04
Target File:   /home/torsten/Documents/FirmwareCam/CH-sys-48.50.64.74.bin
MD5 Checksum:  1a6583e0e90af1cb5b67ae880aa2e358
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
172           0xAC            Zip archive data, at least v2.0 to extract, compressed size: 7241, uncompressed size: 16802, name: system/system/bin/motogpio.ko
7599          0x1DAF          End of Zip archive
7761          0x1E51          Zip archive data, at least v2.0 to extract, compressed size: 4649, uncompressed size: 9152, name: system/system/bin/cmd_thread
12594         0x3132          End of Zip archive
12756         0x31D4          Zip archive data, at least v1.0 to extract, compressed size: 8, uncompressed size: 8, name: system/system/bin/fwversion.bin
12954         0x329A          End of Zip archive
13116         0x333C          Zip archive data, at least v1.0 to extract, compressed size: 9, uncompressed size: 9, name: system/system/bin/sysversion.txt
13317         0x3405          End of Zip archive
13479         0x34A7          Zip archive data, at least v2.0 to extract, compressed size: 1852, uncompressed size: 9692, name: system/system/bin/grade.sh
15511         0x3C97          End of Zip archive
15673         0x3D39          Zip archive data, at least v2.0 to extract, compressed size: 431629, uncompressed size: 958036, name: system/system/bin/encoder
447480        0x6D3F8         End of Zip archive
447642        0x6D49A         Zip archive data, at least v2.0 to extract, compressed size: 61572, uncompressed size: 147792, name: system/system/bin/wifidaemon
509398        0x7C5D6         End of Zip archive
509560        0x7C678         Zip archive data, at least v2.0 to extract, compressed size: 1874, uncompressed size: 4522, name: system/system/bin/gpio_aplink.ko
511626        0x7CE8A         End of Zip archive
511788        0x7CF2C         Zip archive data, at least v2.0 to extract, compressed size: 8704, uncompressed size: 20212, name: system/system/bin/updata
520668        0x7F1DC         End of Zip archive
520830        0x7F27E         Zip archive data, at least v2.0 to extract, compressed size: 7962, uncompressed size: 18643, name: system/system/lib/libsns_h42.so
528982        0x81256         End of Zip archive
529144        0x812F8         Zip archive data, at least v2.0 to extract, compressed size: 8969, uncompressed size: 19091, name: system/system/lib/libsns_gc1004.so
538309        0x836C5         End of Zip archive
538471        0x83767         Zip archive data, at least v2.0 to extract, compressed size: 7813, uncompressed size: 16341, name: system/system/lib/libsns_ov9712_plus.so
546490        0x856BA         End of Zip archive
546652        0x8575C         Zip archive data, at least v2.0 to extract, compressed size: 90652, uncompressed size: 357356, name: system/system/lib/libOnvif.so
637490        0x9BA32         End of Zip archive
637652        0x9BAD4         Zip archive data, at least v2.0 to extract, compressed size: 9979, uncompressed size: 21088, name: system/system/lib/libsns_sc1045.so
647827        0x9E293         End of Zip archive
647989        0x9E335         Zip archive data, at least v2.0 to extract, compressed size: 43603, uncompressed size: 84480, name: system/system/lib/libvoice_arm.so
691786        0xA8E4A         End of Zip archive
691948        0xA8EEC         Zip archive data, at least v2.0 to extract, compressed size: 136, uncompressed size: 236, name: system/init/ipcam.sh
692252        0xA901C         End of Zip archive

Scan Time:     2016-08-06 18:45:04
Target File:   /home/torsten/Documents/FirmwareCam/_CH-sys-48.50.64.74.bin.extracted/system/init/ipcam.sh
MD5 Checksum:  7166ea2370656b5ba19a67039dc08b75
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
149           0x95            Unix path: /system/system/bin/updata
175           0xAF            Unix path: /system/system/bin/wifidaemon &
207           0xCF            Unix path: /system/system/bin/upgrade &

linux-fftx:/home/torsten/Documents/FirmwareCam/_CH-sys-48.50.64.74.bin.extracted # tree
.
└── system
    └── init
        └── ipcam.sh

2 directories, 1 file
linux-fftx:/home/torsten/Documents/FirmwareCam/_CH-sys-48.50.64.74.bin.extracted # 
 
what am I doing wrong? please help
Kind regards
TheGotchi

devttys0, 2016-08-06:
Do you have the `jar` utility installed? It seems only the last zip file was extracted, which is a common issue with some zip files extracted using the `7z` utility (which binwalk defaults to using if `jar` is not installed): https://github.com/devttys0/binwalk/blob/master/src/binwalk/config/extract.conf#L43

TheGotchi, 2016-08-07:
thanks a lot. just to clarify for me (i'm quite new to linux).

I just have to install Java?

Kind regards

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Kriskras99, 2016-08-07:
If you are on ubuntu you can do the following:
```bash
sudo apt-get update && sudo apt-get install openjdk-8-jre
```
or if you want Oracle's java:
```bash
sudo apt-get install python-software-properties && sudo add-apt-repository ppa:webupd8team/java
sudo apt-get update && sudo apt-get install oracle-java8-installer
```

If you're not on Ubuntu could you please say on which OS you are?

TheGotchi, 2016-08-07:
Wow, I appreciate this. Thanks a lot. I'm on Suse 13.1

Kriskras99, 2016-08-08:
I don' t actually have any experience with Suse, but this tutorial seems good:
https://www.unixmen.com/install-java-8-opensuse-13-x-sles-servers/
And here's the official documentation:
https://en.opensuse.org/SDB:Installing_Java
And sorry for the late reply, I missed the notification.

TheGotchi, 2016-08-09:
thanks a lot for your assistance but I have no luck. java-installation worked in my opinion:

linux-fftx:/home/torsten/Downloads # update-alternatives --config java
There are 2 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                       Priority   Status
------------------------------------------------------------
  0            /usr/lib64/jvm/jre-1.7.0-openjdk/bin/java   17147     auto mode
* 1            /usr/java/jre1.8.0_102/bin/java             1         manual mode
  2            /usr/lib64/jvm/jre-1.7.0-openjdk/bin/java   17147     manual mode

Press enter to keep the current choice[*], or type selection number:

But same issue here:

linux-fftx:/home/torsten/Downloads # binwalk -Mer CH-sys-48.50.64.74.bin

Scan Time:     2016-08-09 17:25:28
Target File:   /home/torsten/Downloads/CH-sys-48.50.64.74.bin
MD5 Checksum:  1a6583e0e90af1cb5b67ae880aa2e358
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
172           0xAC            Zip archive data, at least v2.0 to extract, compressed size: 7241, uncompressed size: 16802, name: system/system/bin/motogpio.ko
7599          0x1DAF          End of Zip archive
7761          0x1E51          Zip archive data, at least v2.0 to extract, compressed size: 4649, uncompressed size: 9152, name: system/system/bin/cmd_thread
12594         0x3132          End of Zip archive
12756         0x31D4          Zip archive data, at least v1.0 to extract, compressed size: 8, uncompressed size: 8, name: system/system/bin/fwversion.bin
12954         0x329A          End of Zip archive
13116         0x333C          Zip archive data, at least v1.0 to extract, compressed size: 9, uncompressed size: 9, name: system/system/bin/sysversion.txt
13317         0x3405          End of Zip archive
13479         0x34A7          Zip archive data, at least v2.0 to extract, compressed size: 1852, uncompressed size: 9692, name: system/system/bin/grade.sh
15511         0x3C97          End of Zip archive
15673         0x3D39          Zip archive data, at least v2.0 to extract, compressed size: 431629, uncompressed size: 958036, name: system/system/bin/encoder
447480        0x6D3F8         End of Zip archive
447642        0x6D49A         Zip archive data, at least v2.0 to extract, compressed size: 61572, uncompressed size: 147792, name: system/system/bin/wifidaemon
509398        0x7C5D6         End of Zip archive
509560        0x7C678         Zip archive data, at least v2.0 to extract, compressed size: 1874, uncompressed size: 4522, name: system/system/bin/gpio_aplink.ko
511626        0x7CE8A         End of Zip archive
511788        0x7CF2C         Zip archive data, at least v2.0 to extract, compressed size: 8704, uncompressed size: 20212, name: system/system/bin/updata
520668        0x7F1DC         End of Zip archive
520830        0x7F27E         Zip archive data, at least v2.0 to extract, compressed size: 7962, uncompressed size: 18643, name: system/system/lib/libsns_h42.so
528982        0x81256         End of Zip archive
529144        0x812F8         Zip archive data, at least v2.0 to extract, compressed size: 8969, uncompressed size: 19091, name: system/system/lib/libsns_gc1004.so
538309        0x836C5         End of Zip archive
538471        0x83767         Zip archive data, at least v2.0 to extract, compressed size: 7813, uncompressed size: 16341, name: system/system/lib/libsns_ov9712_plus.so
546490        0x856BA         End of Zip archive
546652        0x8575C         Zip archive data, at least v2.0 to extract, compressed size: 90652, uncompressed size: 357356, name: system/system/lib/libOnvif.so
637490        0x9BA32         End of Zip archive
637652        0x9BAD4         Zip archive data, at least v2.0 to extract, compressed size: 9979, uncompressed size: 21088, name: system/system/lib/libsns_sc1045.so
647827        0x9E293         End of Zip archive
647989        0x9E335         Zip archive data, at least v2.0 to extract, compressed size: 43603, uncompressed size: 84480, name: system/system/lib/libvoice_arm.so
691786        0xA8E4A         End of Zip archive
691948        0xA8EEC         Zip archive data, at least v2.0 to extract, compressed size: 136, uncompressed size: 236, name: system/init/ipcam.sh
692252        0xA901C         End of Zip archive

Scan Time:     2016-08-09 17:25:28
Target File:   /home/torsten/Downloads/_CH-sys-48.50.64.74.bin.extracted/system/init/ipcam.sh
MD5 Checksum:  7166ea2370656b5ba19a67039dc08b75
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
149           0x95            Unix path: /system/system/bin/updata
175           0xAF            Unix path: /system/system/bin/wifidaemon &
207           0xCF            Unix path: /system/system/bin/upgrade &

linux-fftx:/home/torsten/Downloads # ls
CH-sys-48.50.64.74.bin  _CH-sys-48.50.64.74.bin.extracted  jre-8u102-linux-x64.rpm
linux-fftx:/home/torsten/Downloads # cd _CH-sys-48.50.64.74.bin.extracted/
linux-fftx:/home/torsten/Downloads/_CH-sys-48.50.64.74.bin.extracted # ls
system
linux-fftx:/home/torsten/Downloads/_CH-sys-48.50.64.74.bin.extracted # cd system
linux-fftx:/home/torsten/Downloads/_CH-sys-48.50.64.74.bin.extracted/system # ls
init
linux-fftx:/home/torsten/Downloads/_CH-sys-48.50.64.74.bin.extracted/system # cd init
linux-fftx:/home/torsten/Downloads/_CH-sys-48.50.64.74.bin.extracted/system/init # ls
ipcam.sh
linux-fftx:/home/torsten/Downloads/_CH-sys-48.50.64.74.bin.extracted/system/init #

any idea?

Kriskras99, 2016-08-09:
So I tried it on [CH-sys-48.53.64.67.bin](http://cd.gocam.so/FM/system/CH-sys-48.53.64.67.zip) and it failed for on Binwalk v2.1.1. Then I found this issue #185, which failed in the same way. But there @devttys0 mentioned it was fixed in the master branch, so I built Binwalk from source (v2.1.2b) and it succesfully extracted the whole file. So the only way to fix this is to wait for Binwalk v2.1.2 to be released or build from the master branch.

Please take a look at https://github.com/devttys0/binwalk/blob/master/INSTALL.md to compile binwalk from source.
If you can't compile from source, please send me the link from which you downloaded the firmware, then I will extract it for you.

TheGotchi, 2016-08-09:
wow, thanks a lot. I think my skills are actually too low for doing the 
compiling. very sorry.
here's the link to the firmware:

http://45.63.8.70/FM/system/CH-sys-48.50.64.74.zip

thanks again.

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Kriskras99, 2016-08-09:
Here it is: https://mega.nz/#!5d8jUQAK!HQMK_BbvmQIORpijoAVP-cxUP1wG9BEpLOBiW35T7fM
And good luck with reverse engineering :+1:

TheGotchi, 2016-08-09:
wow, thanks a lot ... but ... sad to say.
the cam just upgraded automaticly to a new firmware :-(

http://45.63.8.70/FM/system/CH-sys-48.50.72.76.zip

may i ask you again? so sorry

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Kriskras99, 2016-08-09:
No problem, here you go:
https://mega.nz/#!1J9zUQhJ!QpjGnGI0GVKDVDuZJwONL5dAZjJQ1zDoHx8vbQ-TWjE

TheGotchi, 2016-08-11:
thank you very much. that works.

---
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

Kriskras99, 2016-08-11:
Glad it worked, I think this issue can be closed now. It's fixed in the next version.

closed 2016-08-15 by devttys0