ReFirmLabs/binwalk #224: Strange output from modified android boot image

voltagex, 2016-08-25:
Input is one of @M1cha's efidroid boot images.
Binwalk v2.1.2b from b22c4b3c386c608c0545e4dfa46a2b33e99bec3b

It seems to "loop" at the end of the detection - I've attached a full log to this issue.

```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Android bootimg, kernel size: 444836 bytes, kernel addr: 0x8000, ramdisk size: 0 bytes, ramdisk addr: 0x2000000, product name: ""
302080        0x49C00         SHA256 hash constants, little endian
326712        0x4FC38         CRC32 polynomial table, little endian
336360        0x521E8         Certificate in DER format (x509 v3), header length: 4, sequence length: 720
347220        0x54C54         Android bootimg, kernel size: 0 bytes, kernel addr: 0x6E756F66, ramdisk size: 1849761892 bytes, ramdisk addr: 0x696F7264, product name: " addr:%08x sz:%08x"
348652        0x551EC         Unix path: /home/voltagex/efidroid/uefi/lkmodules/shared/lib/atagparse/atagparse.c
349620        0x555B4         Base64 standard index table
350592        0x55980         Unix path: /home/voltagex/efidroid/modules/libboot/qcdt.c
350708        0x559F4         CRC32 polynomial table, little endian
445284        0x6CB64         gzip compressed data, from FAT filesystem (MS-DOS, OS/2, NT), last modified: 1970-01-01 00:00:00 (null date)
445432        0x6CBF8         Qualcomm device tree container, version: 0, DTB entries: 0
450560        0x6E000         Qualcomm device tree container, version: 3, DTB entries: 53
454656        0x6F000         device tree image (dtb)
463639        0x71317         LZMA compressed data, properties: 0x91, dictionary size: 33554432 bytes, uncompressed size: 50331648 bytes
```

[binwalk-android-image.txt](https://github.com/devttys0/binwalk/files/437070/binwalk-android-image.txt)

[lk_origdtb.img.xz.zip](https://github.com/devttys0/binwalk/files/437072/lk_origdtb.img.xz.zip)

(this is just an xz file, I've renamed it so GitHub will allow me to upload it)

devttys0, 2016-08-26:
The LZMA results are not shown in my output, which suggests that they are likely false positives but that your Python install does not have the `lzma` module installed (if installed, binwalk uses it to validate LZMA signatures). If you either install the `lzma` module for Python2.x, or install/run binwalk in Python 3.x (`lzma` is a default package in Python3), you should see the LZMA results disappear.

There are numerous "device tree image" results, but these all appear valid. In fact, the binwalk output suggests that there should be 53 of DTB entries, although only 44 are detected, so there are possibly some that are being missed (different signatures?).

M1cha, 2016-08-26:
@devttys0 if I got it correctly his point was that he had an defective file, but thinks that no file should be able to cause a infinity loop - no matter how bad they got damaged.

devttys0, 2016-08-26:
@M1cha I agree, but there is no loop, at least not when I run binwalk on the file, and aside from the LZMA differences that I noted earlier @voltagex's output is the same as mine. Examining the raw file, the signatures all appear to be exactly where binwalk reports them to be.

voltagex, 2016-08-26:
Yep, sorry. My mistake.

closed 2016-08-26 by voltagex