ReFirmLabs/binwalk #225: detecting MPEG transport streams

BenGardiner, 2016-09-21:
The application (as in the subject line) might be a little strange for binwalk, but... theoretically just adding a custom signature, right? Unfortunately I haven't had luck getting this to work yet. Please help if you can:

Based on [libmagic's mpeg transport stream signature](https://github.com/threatstack/libmagic/blob/master/magic/Magdir/animation#L703) -- reproduced here for reference:

```
[...]
# iso 13818 transport stream
#
# from Oskar Schirmer <schirmer@scara.com> Feb 3, 2001 (ISO 13818.1)
# syncbyte      8 bit	0x47
# error_ind     1 bit	-
# payload_start 1 bit	1
# priority      1 bit	-
# PID          13 bit	0x0000
# scrambling    2 bit	-
# adaptfld_ctrl 2 bit	1 or 3
# conti_count   4 bit	-
0	belong&0xFF5FFF10	0x47400010
>188	byte			0x47		MPEG transport stream data
[...]
```

 I created the following binwalk custom signature file

```bash
$ cat ~/mpeg_ts.magic
0       belong&0xff5fff10       0x47400010      MPEG transport stream data
>188    byte                    !0x47           {invalid}
```

I have an MPEG ts stream for testing on-hand from [online public sources](https://stackoverflow.com/questions/10104301/hls-streaming-video-url-need-for-testing)

```bash
$ hexdump -C ~/06401.ts |head
00000000  47 40 00 1b 00 00 b0 0d  00 00 c3 00 00 00 01 e1  |G@..............|
00000010  e0 e8 5f 74 ec ff ff ff  ff ff ff ff ff ff ff ff  |.._t............|
00000020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000000b0  ff ff ff ff ff ff ff ff  ff ff ff ff 47 41 e0 1b  |............GA..|
000000c0  00 02 b0 1d 00 01 c3 00  00 e1 e1 f0 00 1b e1 e1  |................|
000000d0  f0 00 0f e1 e2 f0 06 0a  04 65 6e 67 00 9c 04 25  |.........eng...%|
000000e0  f0 ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000000f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
$ r2 -n ~/06401.ts
 -- Learn pancake as if you were radare!
[0x00000000]> s 188
[0x000000bc]> px
- offset -   0 1  2 3  4 5  6 7  8 9  A B  C D  E F  0123456789ABCDEF
0x000000bc  4741 e01b 0002 b01d 0001 c300 00e1 e1f0  GA..............
0x000000cc  001b e1e1 f000 0fe1 e2f0 060a 0465 6e67  .............eng
0x000000dc  009c 0425 f0ff ffff ffff ffff ffff ffff  ...%............
0x000000ec  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x000000fc  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x0000010c  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x0000011c  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x0000012c  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x0000013c  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x0000014c  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x0000015c  ffff ffff ffff ffff ffff ffff ffff ffff  ................
0x0000016c  ffff ffff ffff ffff ffff ffff 4741 e131  ............GA.1
0x0000017c  0710 0006 b3f4 7e3e 0000 01e0 0000 84c0  ......~>........
0x0000018c  0a31 0037 c4ab 1100 37ad 3500 0000 0109  .1.7....7.5.....
0x0000019c  1000 0000 0167 4280 1e96 5405 017f cb80  .....gB...T.....
0x000001ac  a040 0000 fa40 003a 9838 8800 493c 0002  .@...@.:.8..I<..
[0x000000bc]> q
```

I've installed binwalk from sources with ```sudo python3 setup.py install``` from git

```bash
± git log -n1
commit fa0d7cfba69184ff95471631be6026063ca5346d
Merge: b22c4b3 4d320c6
Author: devttys0 <heffnercj@gmail.com>
Date:   Fri Aug 26 10:05:18 2016 -0400

    Merge pull request #222 from 0xItx/master

    Add signatures for qnx6fs & qnx ifs
```

I expected to be able to get binwalk to find the mpeg transport stream signature at offset 0 of that file -- but I get silence. I've also tried disabling smart and showing invalid to see more. No joy

```bash
$ binwalk -v -B -b -I --magic=$HOME/mpeg_ts.magic ~/06401.ts |cat -v|grep -i mpeg
$
```

Please help me figure out where I've gone wrong.

devttys0, 2016-09-21:
Binwalk doesn't use libmagic, although its file format is mostly compatible ("mostly" being the key word there :). Internally, binwalk converts the magic bytes to a regex string and uses regex to do the pattern matching, which is orders of magnitude faster than using libmagic. The down side is that some features, like bit masking, aren't supported for the magic bytes (maybe bit-masked regex strings possible?? TBH, my regex foo is weak :P). For indented signature lines, bit masking may still be used.

To get around this, you can generate multiple signatures, one for each variation of the MPEG header. Looking at the first three bytes, there are only two bit positions that are "don't care" bits (`error_ind` and `priority`), so four signatures will cover all variants:

```
0       string  \x47\x40\x00    MPEG transport stream data
>188    byte    !0x47           {invalid}
0       string  \x47\xC0\x00    MPEG transport stream data
>188    byte    !0x47           {invalid}
0       string  \x47\x60\x00    MPEG transport stream data
>188    byte    !0x47           {invalid}
0       string  \x47\xE0\x00    MPEG transport stream data
>188    byte    !0x47           {invalid}
```

These are reasonably strong signatures in that they shouldn't generate many false positives. The only downside is that you need four signature entries to match the exact same file signature, though I find this a relatively rare occurrence.

```
$ binwalk --magic=magic.txt ./06401.ts 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             MPEG transport stream data
```

BenGardiner, 2016-09-21:
Thanks, Craig. :) I think multiple magic byte matches is a totally reasonable approach.

closed 2016-09-21 by BenGardiner