ReFirmLabs/binwalk #229: Wiki describes --disasm option

nicovell3, 2016-10-02:
At [https://github.com/devttys0/binwalk/wiki/Usage](https://github.com/devttys0/binwalk/wiki/Usage) the -Y / --disasm option is described but when running binwalk that option isn't even listed. I don't know if it was an option in the past or it's going to be implemented, but it's a little confusing.

devttys0, 2016-10-04:
The `--disasm` option is only available if you have capstone installed: https://github.com/devttys0/binwalk/blob/master/INSTALL.md#dependencies

closed 2017-02-14 by nicovell3

blshkv, 2017-11-19:
This option is still visible for me even when there is a capstone with bindings installed

binwalk - latest git
dev-libs/capstone-4.0_alpha5-r1
dev-python/capstone-python-4.0_alpha5

```
equery f capstone-python 
 * Searching for capstone-python ...
 * Contents of dev-python/capstone-python-4.0_alpha5:
/usr
/usr/lib64
/usr/lib64/python2.7
/usr/lib64/python2.7/site-packages
/usr/lib64/python2.7/site-packages/capstone
/usr/lib64/python2.7/site-packages/capstone-4.0.0-py2.7.egg-info
/usr/lib64/python2.7/site-packages/capstone/__init__.py
/usr/lib64/python2.7/site-packages/capstone/__init__.pyc
/usr/lib64/python2.7/site-packages/capstone/__init__.pyo
/usr/lib64/python2.7/site-packages/capstone/arm.so
/usr/lib64/python2.7/site-packages/capstone/arm64.so
/usr/lib64/python2.7/site-packages/capstone/arm64_const.so
/usr/lib64/python2.7/site-packages/capstone/arm_const.so
/usr/lib64/python2.7/site-packages/capstone/ccapstone.so
/usr/lib64/python2.7/site-packages/capstone/include
/usr/lib64/python2.7/site-packages/capstone/include/capstone
/usr/lib64/python2.7/site-packages/capstone/include/capstone/arm.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/arm64.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/capstone.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/m68k.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/mips.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/platform.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/ppc.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/sparc.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/systemz.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/tms320c64x.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/x86.h
/usr/lib64/python2.7/site-packages/capstone/include/capstone/xcore.h
/usr/lib64/python2.7/site-packages/capstone/lib
/usr/lib64/python2.7/site-packages/capstone/lib/libcapstone.a
/usr/lib64/python2.7/site-packages/capstone/lib/libcapstone.so
/usr/lib64/python2.7/site-packages/capstone/m68k.so
/usr/lib64/python2.7/site-packages/capstone/m68k_const.so
/usr/lib64/python2.7/site-packages/capstone/mips.so
/usr/lib64/python2.7/site-packages/capstone/mips_const.so
/usr/lib64/python2.7/site-packages/capstone/ppc.so
/usr/lib64/python2.7/site-packages/capstone/ppc_const.so
/usr/lib64/python2.7/site-packages/capstone/sparc.so
/usr/lib64/python2.7/site-packages/capstone/sparc_const.so
/usr/lib64/python2.7/site-packages/capstone/systemz.so
/usr/lib64/python2.7/site-packages/capstone/sysz_const.so
/usr/lib64/python2.7/site-packages/capstone/x86.so
/usr/lib64/python2.7/site-packages/capstone/x86_const.so
/usr/lib64/python2.7/site-packages/capstone/xcore.so
/usr/lib64/python2.7/site-packages/capstone/xcore_const.so
/usr/lib64/python3.4
/usr/lib64/python3.4/site-packages
/usr/lib64/python3.4/site-packages/capstone
/usr/lib64/python3.4/site-packages/capstone-4.0.0-py3.4.egg-info
/usr/lib64/python3.4/site-packages/capstone/__init__.py
/usr/lib64/python3.4/site-packages/capstone/__pycache__
/usr/lib64/python3.4/site-packages/capstone/__pycache__/__init__.cpython-34.pyc
/usr/lib64/python3.4/site-packages/capstone/__pycache__/__init__.cpython-34.pyo
/usr/lib64/python3.4/site-packages/capstone/arm.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/arm64.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/arm64_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/arm_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/ccapstone.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/include
/usr/lib64/python3.4/site-packages/capstone/include/capstone
/usr/lib64/python3.4/site-packages/capstone/include/capstone/arm.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/arm64.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/capstone.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/m68k.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/mips.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/platform.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/ppc.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/sparc.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/systemz.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/tms320c64x.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/x86.h
/usr/lib64/python3.4/site-packages/capstone/include/capstone/xcore.h
/usr/lib64/python3.4/site-packages/capstone/lib
/usr/lib64/python3.4/site-packages/capstone/lib/libcapstone.a
/usr/lib64/python3.4/site-packages/capstone/lib/libcapstone.so
/usr/lib64/python3.4/site-packages/capstone/m68k.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/m68k_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/mips.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/mips_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/ppc.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/ppc_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/sparc.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/sparc_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/systemz.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/sysz_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/x86.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/x86_const.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/xcore.cpython-34m.so
/usr/lib64/python3.4/site-packages/capstone/xcore_const.cpython-34m.so
/usr/share
/usr/share/doc
/usr/share/doc/capstone-python-4.0_alpha5
/usr/share/doc/capstone-python-4.0_alpha5/README.txt.bz2
```

blshkv, 2017-11-19:
sorry, please ignore it. I found out that capstone binding was not installed properly