ReFirmLabs/binwalk #232: D-Link DIR 600 firmware

varunmadhavam, 2016-10-17:
Hi,
doesnt seem to support D-Link DIR 600 firmware. Not detecting any filesystems.

ftp://ftp.dlink.eu/Products/dir/dir-600/driver_software/DIR-600_fw_revC1_3-05B15__all_en_20120216.zip

Kriskras99, 2016-10-17:
Which version of binwalk are you on?
I'm on v2.1.2b and it is detecting a file system.
```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             SEAMA firmware header, big endian, meta size: 52, image size: 1037873
80            0x50            uImage header, header size: 64 bytes, header CRC: 0x8BB760D9, created: 2012-02-01 10:31:41, image size: 691029 bytes, Data Address: 0x80008000, Entry Point: 0x800080BC, data CRC: 0x30BD93D6, OS: RTEMS, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "DIR-600_wrgn34"
144           0x90            LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 2081220 bytes
691280        0xA8C50         uImage header, header size: 64 bytes, header CRC: 0x30382FDA, created: 2012-02-01 10:31:47, image size: 346608 bytes, Data Address: 0x0, Entry Point: 0x0, data CRC: 0xE07CBE2B, OS: Linux, CPU: PowerPC, image type: OS Kernel Image, compression type: lzma, image name: "rootfs_with_lzma"
691344        0xA8C90         LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 470309 bytes
```
It does not seem to extract it correctly however: https://gist.github.com/Kriskras99/b2b68c6b66269ea04d9b360ed5789833

varunmadhavam, 2016-10-17:
hi,
i have reached the above stage. What i meant was that it was not able to extract any proper file systems. Sorry for the confusion. I will edit the post. And its detecting two kernel images, one MIPS and one PowerPC. Is it possible or just a false positive.

devttys0, 2016-11-16:
There are not two kernel images, there are two uImage headers; the first claims to be a MIPS kernel (this appears to be correct), the second claims to be a PowerPC file system. It is possible to have multiple processors on the same device, but in this case it is unlikely; the developers probably just made a typo, or were lazy.

The file system, however, appears to be custom (note the `ROMFS` strings, but this is *not* a Linux romfs file system):

```
$ hexdump -C A8C90 | head
00000000  2e 6d 6f 52 a1 02 00 00  40 2d 07 00 04 03 02 01  |.moR....@-......|
00000010  52 4f 4d 46 53 20 76 39  2e 30 00 bf e0 e7 b8 00  |ROMFS v9.0......|
00000020  01 00 7f 05 05 00 00 00  22 02 e8 03 a0 00 00 00  |........".......|
00000030  6f 14 29 4f 40 54 00 00  00 00 00 00 20 20 20 30  |o.)O@T......   0|
00000040  01 00 7f 05 03 00 00 00  22 02 e8 03 60 00 00 00  |........"...`...|
00000050  6f 14 29 4f e0 54 00 00  00 00 00 00 20 20 20 31  |o.)O.T......   1|
00000060  01 00 7f 05 09 00 00 00  22 02 e8 03 20 01 00 00  |........"... ...|
00000070  71 14 29 4f 40 55 00 00  00 00 00 00 20 20 20 32  |q.)O@U......   2|
00000080  01 00 7f 05 09 00 00 00  22 02 e8 03 40 01 00 00  |........"...@...|
00000090  6f 14 29 4f 60 56 00 00  00 00 00 00 20 20 20 33  |o.)O`V......   3|
```

Google didn't turn up anything, but I reverse engineered enough of it to figure out the data structure and extract the files. They are almost all web files (php, js, images, etc), along with some other misc stuff; no executable ELFs though, so this appears to be some sort of partial firmware update.

I've updated binwalk to identify this "file system", and I've attached a script, [romfs.py.txt](https://github.com/devttys0/binwalk/files/595652/romfs.py.txt), to extract the files, but have not integrated the extraction script into binwalk so it must be run manually (the code is super crappy, but seems to work, at least for this firmware image).

```
$ binwalk A8C90

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             D-Link ROMFS filesystem, version 9.0, little endian, size: <= 470336
```

```
$ ./romfs.py A8C90 romfs-root
Extracted romfs-root/htdocs/phplib/inf.php
Extracted romfs-root/htdocs/phplib/inet.php
Extracted romfs-root/htdocs/phplib/wifi.php
Extracted romfs-root/htdocs/phplib/trace.php
Extracted romfs-root/htdocs/phplib/dumplog.php
Extracted romfs-root/htdocs/phplib/html.php
Extracted romfs-root/htdocs/phplib/upnp.php
Extracted romfs-root/htdocs/phplib/xnode.php
Extracted romfs-root/htdocs/phplib/phyinf.php
Extracted romfs-root/htdocs/phplib/fatlady/DHCPS4.LAN-2.php
Extracted romfs-root/htdocs/phplib/fatlady/DEVICE.DLOB.php
Extracted romfs-root/htdocs/phplib/fatlady/VSVR.NAT-1.php
Extracted romfs-root/htdocs/phplib/fatlady/INET.BRIDGE-1.php
Extracted romfs-root/htdocs/phplib/fatlady/INET.WAN-2.php
Extracted romfs-root/htdocs/phplib/fatlady/DEVICE.EASYSETUP.php
Extracted romfs-root/htdocs/phplib/fatlady/DHCPS4.LAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/ICMP.WAN-2.php
Extracted romfs-root/htdocs/phplib/fatlady/HTTP.WAN-3.php
Extracted romfs-root/htdocs/phplib/fatlady/DEVICE.TIME.php
Extracted romfs-root/htdocs/phplib/fatlady/DEVICE.ACCOUNT.php
Extracted romfs-root/htdocs/phplib/fatlady/WAN.php
Extracted romfs-root/htdocs/phplib/fatlady/PHYINF.WAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/WIFI.WLAN-2.php
Extracted romfs-root/htdocs/phplib/fatlady/PORTT.NAT-1.php
Extracted romfs-root/htdocs/phplib/fatlady/DMZ.NAT-1.php
Extracted romfs-root/htdocs/phplib/fatlady/ROUTE.STATIC.php
Extracted romfs-root/htdocs/phplib/fatlady/RUNTIME.TIME.php
Extracted romfs-root/htdocs/phplib/fatlady/WIFI.WDS-1.php
Extracted romfs-root/htdocs/phplib/fatlady/QOS.php
Extracted romfs-root/htdocs/phplib/fatlady/URLCTRL.php
Extracted romfs-root/htdocs/phplib/fatlady/DNS4.LAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/INET.LAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/SCHEDULE.php
Extracted romfs-root/htdocs/phplib/fatlady/INET.WAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/UPNP.LAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/WIFI.WLAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/ICMP.WAN-1.php
Extracted romfs-root/htdocs/phplib/fatlady/HTTP.WAN-1.php
[... 565 lines cut ...]
Extracted romfs-root/etc/events/SEALPAC-SAVE.sh
Extracted romfs-root/etc/events/SENDMAIL.sh
Extracted romfs-root/etc/events/reboot_delay.sh
Extracted romfs-root/etc/events/SENDMAIL.php
Extracted romfs-root/etc/events/freset.sh
Extracted romfs-root/etc/events/start_inter_connect_detect.sh
Extracted romfs-root/etc/events/INET4-DHCP_WAN-2_bound.sh
Extracted romfs-root/etc/events/INET4-DHCP.php
Extracted romfs-root/etc/events/DHCPS-RESTART.sh
Extracted romfs-root/etc/events/AUTO_WAN_GO.sh
Extracted romfs-root/etc/events/DHCPS-RESTART.php
Extracted romfs-root/etc/events/AUTO_WAN_STATE.sh
Extracted romfs-root/etc/events/SEALPAC.php
Extracted romfs-root/etc/events/scanap.sh
Extracted romfs-root/etc/events/UPDATERESOLV.php
Extracted romfs-root/etc/events/INET4-DHCP_BRIDGE-1_renew.sh
Extracted romfs-root/etc/events/LOGFULL.sh
Extracted romfs-root/etc/events/end_inter_connect_detect.sh
Extracted romfs-root/etc/events/SEALPAC-LOAD.sh
Extracted romfs-root/etc/events/INET4-DHCP_WAN-1_bound.sh
Extracted romfs-root/etc/events/INET4-DHCP_BRIDGE-1_bound.sh
Extracted romfs-root/etc/events/UPDATERESOLV.sh
Extracted romfs-root/etc/events/SEALPAC-CLEAR.sh
Extracted romfs-root/etc/events/checkfw.sh
```

devttys0, 2016-11-17:
Extraction capabilities for this firmware file system have been integrated into the latest binwalk master branch.

closed 2016-11-22 by devttys0