ReFirmLabs/binwalk #234: Binwalk recursively extracting to the extracted folder

adi0x90, 2016-10-30:
Binwalk is recursively extracting all the files inside the subfolders to the `_firmware-name.extracted/ `folder instead of respecting the directory structure and then placing the files inside `squashfs-root`. 

Output below : 

```
root@kali:# binwalk -e Dlink_firmware.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
48            0x30            Unix path: /dev/mtdblock/2
96            0x60            uImage header, header size: 64 bytes, header CRC: 0x7FE9E826, created: 2010-11-23 11:58:41, image size: 878029 bytes, Data Address: 0x80000000, Entry Point: 0x802B5000, data CRC: 0x7C3CAE85, OS: Linux, CPU: MIPS, image type: OS Kernel Image, compression type: lzma, image name: "Linux Kernel Image"
160           0xA0            LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 2956312 bytes
917600        0xE0060         PackImg section delimiter tag, little endian size: 7348736 bytes; big endian size: 2256896 bytes
917632        0xE0080         Squashfs filesystem, little endian, non-standard signature, version 3.0, size: 2256151 bytes, 1119 inodes, blocksize: 65536 bytes, created: 2010-11-23 11:58:47

root@kali:~/lab/firmware# ls _Dlink_firmware.bin.extracted/
[                                            h_adv_dmz.php                          qos.sh
0                                            h_adv_firewall.php                     qos.xml
0ro                                          h_adv_mac_filter.php                   random
1                                            h_adv_network.php                      rcS
1ro                                          h_adv_port.php                         reboot
2                                            h_adv_qos.php                          __rebooting_msg.php
2ro                                          h_adv_routing.php                      Reboot.php

```
The above is happening with all the firmwares and not just for this specific example. At the same time, the `extract-firmware.sh` is working perfectly and extracting the files correctly in the subfolder tree structure. 

Binwalk version : `Binwalk v2.1.2b`
Tried using both the github version, as well as the kali repo one. Any pointers will be highly appreciated.

devttys0, 2016-11-01:
This is not an issue that I am able to reproduce. The extracted files that you listed should be in a `squashfs-root` directory, along with the appropriate directory tree for that file system, but binwalk is not responsible for that - it merely kicks off one of the external SquashFS extraction tools (`unsquashfs` or `sasquatch`). Those tools would be responsible for creating the appropriate directory structure and placing the extracted files in the correct directories.

Please provide a firmware image that you know will reproduce the issue so that I can look into it further.

vincentcox, 2018-02-10:
Same issue as @adi0x90. I used it on: [a tp-link firmware image](https://static.tp-link.com/ArcherC2600(US)_V1_170306.zip). 

[image]

closed 2018-04-12 by CoffeeExpress

vincentcox, 2018-04-12:
Ok then. ¯\_(ツ)_/¯

Dec0y-jb, 2019-10-01:
@vincentcox - I believe this has to do with which distribution you're using. I was experiencing the same issue on Linux Mint; however I extracted the same firmware in Ubuntu 18.04 - and the file structure remained intact.

dissect0r, 2019-12-06:
I just experienced this with Binwalk v2.2.0 under both macOS 10.14.6 and Kali 2019.4 dealing with some decompressed LZO data, but under Ubuntu it worked as expected (put files in the correct subdirectories) oh well I guess :man_shrugging:

LuKePicci, 2020-05-11:
Same thing here on Ubuntu 18.04 under WSL. It worked just fine on the same files before I upgraded from ubuntu 16.04

LuKePicci, 2020-05-18:
I think I have found the culprit, it's p7zip

This is the binary:
```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
38            0x26            LZMA compressed data, properties: 0x6D, dictionary size: 4194304 bytes, uncompressed size: 5570588 bytes
2359296       0x240000        Squashfs filesystem, little endian, version 4.0, compression:xz, size: 37989694 bytes, 6069 inodes, blocksize: 262144 bytes, created: 2020-02-14 03:46:01
```

When I run binwalk -e here is what htop reveals during the first messy part of the extraction:
[image]

Meanwhile I see the squashfs being extracted as a flat files and folders structure.

vincentcox, 2020-05-28:
@LuKePicci How did you fixed it ultimately? It's an issue I had 2,5 years ago, but it's quite possible I face the same issue somewhere in the future. 
BTW, how did you get that view (screenshot)? Is it a (debugging) command?

Take care and stay healthy!

LuKePicci, 2020-05-28:
I simply uninstalled p7zip, so it's clearly something that should be fixed in binwalk.

The screens hot is from the good old htop in tree view.

vincentcox, 2020-05-29:
Thanks for the reply! 

> The screenshot is from the good old htop in tree view.

I still learn Linux every day!