ReFirmLabs/binwalk #235: LZMA header at 0x0 vs 0x514 -different results

kamil-HFC, 2016-11-03:
I have firmware with false-positive LZMA header:
$ hexdump -C firware.bin | grep '5d 00.00'
```
00000510  70 3b 11 c5 5d 00 00 00  01 00 20 20 0e 00 0d 3a  |p;..].....  ...:|
00357730  50 5d ec 5d 00 00 54 c0  bb 25 b4 1d 15 dd e3 95  |P].]..T..%......|
```
but Binwalk revelas nothing:
```
$ binwalk firware.bin -I | grep LZMA
$
```
When i cut possible LZMA file ith dd:
```
$ dd if=firware.bin of=lzma bs=$((0x514)) skip=1
```
Now we have this:
```
$ binwalk lzma -I | grep LZMA
0             0x0             LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 2898643604054482944 bytes

Even if i append one byte before \x5D Binwalk can't find LZMA header. Why? 
```

devttys0, 2016-11-03:
Hmmm, I cannot reproduce this issue: 

```
$ binwalk test.lzma 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 13130048 bytes

$ echo "abcde" > test2.lzma
$ cat test.lzma >>test2.lzma 
$ binwalk test2.lzma 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
6             0x6             LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 13130048 bytes
```

Can you provide a file that will reproduce the bug?

kamil-HFC, 2016-11-03:
Here it is:
http://www86.zippyshare.com/v/58PQvZvi/file.html

devttys0, 2016-11-04:
You have to be careful when using the `-I` option, as you are telling binwalk to treat invalid results as if they are valid. This means if there is an invalid signature that occurs before your LZMA data, and that signature says it is x number of bytes in size, binwalk will skip ahead x bytes, potentially skipping over the LZMA header that you're interested in. It looks like that's what is happening with your command:

```
$ binwalk -I firware.bin | grep LZMA
$
```

You can tell binwalk to not trust the signature data by using the `--dumb` option:

```
$ binwalk -I firware.bin --dumb | grep LZMA
1259          0x4EB           LZMA compressed data, properties: 0x6E, dictionary size: 0 bytes, uncompressed size: 0 bytes
1300          0x514           LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 2898643604054482944 bytes
```

Or, if you're specifically looking for LZMA results that were marked as invalid, use the `-y` option to tell binwalk to only look for LZMA signatures:

```
$ binwalk -I firware.bin -y lzma

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
1259          0x4EB           LZMA compressed data, properties: 0x6E, dictionary size: 0 bytes, uncompressed size: 0 bytes
1300          0x514           LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 2898643604054482944 bytes

$
```

With that said, the LZMA compression at offset 0x514 does appear to be valid, even though its uncompressed size field is incorrect. Since binwalk now validates LZMA compression before reporting LZMA results, the size field checks that it performs are not really necessary, so I've just checked in a patch that removes that check; the LZMA result is now reported by default:

```
$ binwalk firware.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
56            0x38            Certificate in DER format (x509 v3), header length: 4, sequence length: 748
1300          0x514           LZMA compressed data, properties: 0x5D, dictionary size: 16777216 bytes, uncompressed size: 2898643604054482944 bytes
```

kamil-HFC, 2016-11-04:
Thanks for your comprehensive explanation. Now Binwalk reports LZMA as you say.

closed 2016-11-04 by kamil-HFC