ReFirmLabs/binwalk #244: Best improvements for binwalk

numind06, 2016-12-18:
Hi, I'd just want to recognize the big effort of this project but we still have some unsupported headers and compressed systems that I think we could fix with the help of all people interested.

In my particular case I hav some troubles with UBI headers, seems like a custom build, too much headers, the data has been extracted from NAND flash, after remove dummy blocks we get a clean 256MB bin dump, binwalk returns this data "UBI erase count header, version: 1, EC: 0x2, VID header offset: 0x800, data offset: 0x1000" but binwalk is only able to extract two empty folders, "0" and "webapp"
[image]
Entropy shows many headers and splitted chuncks of compressed data
[image]

So maybe if some more advanced user is willing to help with did we coul show all info neded and implement solution for each particular case and save all in a kind of library or database.

Many thanks in advance.

devttys0, 2017-01-10:
Can you provide a copy of the firmware image?

numind06, 2017-01-11:
Sure, If you tell me other way to upload file here (about 70MB), maximum size is 10 MB, I could paste a link from wetransfer for example...but need to be downloaded before 10 days or link will die.

devttys0, 2017-01-11:
That would be fine!

numind06, 2017-01-24:
Hi, sorry for delay, here is the link, only available for 7 days :(
https://we.tl/A0d78PSp9d

closed 2018-04-12 by CoffeeExpress