ReFirmLabs/binwalk #255: Extraction files from firmware failed

futex, 2017-03-11:
There is a weird behavior when i want to extract files from a firmware with binwalk 2.1.1.

```
binwalk -Mer CH-sys-48.61.75.81.bin                                                                                                                                                                                           

Scan Time:     2017-03-11 13:01:44
Target File:   /tmp/test/CH-sys-48.61.75.81.bin
MD5 Checksum:  266d22c102850f707b8c4246871ff639
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
172           0xAC            Zip archive data, at least v2.0 to extract, compressed size: 2091, uncompressed size: 12128, name: system/system/bin/load3518ev200
2453          0x995           End of Zip archive
2615          0xA37           Zip archive data, at least v2.0 to extract, compressed size: 59761, uncompressed size: 140027, name: system/system/bin/wifidaemon
62560         0xF460          End of Zip archive
62722         0xF502          Zip archive data, at least v1.0 to extract, compressed size: 9, uncompressed size: 9, name: system/system/bin/sysversion.txt
62923         0xF5CB          End of Zip archive
63085         0xF66D          Zip archive data, at least v2.0 to extract, compressed size: 4649, uncompressed size: 9152, name: system/system/bin/cmd_thread
67918         0x1094E         End of Zip archive
68080         0x109F0         Zip archive data, at least v1.0 to extract, compressed size: 8, uncompressed size: 8, name: system/system/bin/fwversion.bin
68278         0x10AB6         End of Zip archive
68440         0x10B58         Zip archive data, at least v2.0 to extract, compressed size: 1728, uncompressed size: 4362, name: system/system/bin/gpio_aplink.ko
70360         0x112D8         End of Zip archive
70522         0x1137A         Zip archive data, at least v2.0 to extract, compressed size: 1852, uncompressed size: 9692, name: system/system/bin/grade.sh
72554         0x11B6A         End of Zip archive
72716         0x11C0C         Zip archive data, at least v2.0 to extract, compressed size: 292784, uncompressed size: 722528, name: system/system/bin/encoder
365678        0x5946E         End of Zip archive
365840        0x59510         Zip archive data, at least v2.0 to extract, compressed size: 255539, uncompressed size: 524032, name: system/system/bin/wpa_supplicant
621571        0x97C03         End of Zip archive
621733        0x97CA5         Zip archive data, at least v2.0 to extract, compressed size: 1570, uncompressed size: 5877, name: system/system/bin/load3518
623483        0x9837B         End of Zip archive
623645        0x9841D         Zip archive data, at least v2.0 to extract, compressed size: 8704, uncompressed size: 20212, name: system/system/bin/updata
632525        0x9A6CD         End of Zip archive
632687        0x9A76F         Zip archive data, at least v2.0 to extract, compressed size: 12088, uncompressed size: 29447, name: system/system/lib/libsns_gc1004.so
644971        0x9D76B         End of Zip archive
645133        0x9D80D         Zip archive data, at least v2.0 to extract, compressed size: 16893, uncompressed size: 40146, name: system/system/lib/libsns_sc1135.so
662222        0xA1ACE         End of Zip archive
662384        0xA1B70         Zip archive data, at least v2.0 to extract, compressed size: 11569, uncompressed size: 28153, name: system/system/lib/libsns_gc1024.so
674149        0xA4965         End of Zip archive
674311        0xA4A07         Zip archive data, at least v2.0 to extract, compressed size: 12847, uncompressed size: 30829, name: system/system/lib/libsns_ov9732.so
687354        0xA7CFA         End of Zip archive
687516        0xA7D9C         Zip archive data, at least v2.0 to extract, compressed size: 17306, uncompressed size: 43188, name: system/system/lib/libsns_sc2135.so
705018        0xAC1FA         End of Zip archive
705180        0xAC29C         Zip archive data, at least v1.0 to extract, name: system/system/lib/libOnvif.so
705366        0xAC356         End of Zip archive
705528        0xAC3F8         Zip archive data, at least v1.0 to extract, name: system/system/lib/libsns_ov9712_plus.so
705734        0xAC4C6         End of Zip archive
705896        0xAC568         Zip archive data, at least v2.0 to extract, compressed size: 100918, uncompressed size: 245972, name: system/system/lib/libnl.so.1
806998        0xC5056         End of Zip archive
807160        0xC50F8         Zip archive data, at least v2.0 to extract, compressed size: 43603, uncompressed size: 84480, name: system/system/lib/libvoice_arm.so
850957        0xCFC0D         End of Zip archive
851119        0xCFCAF         Zip archive data, at least v2.0 to extract, compressed size: 340, uncompressed size: 1137, name: system/init/seq_ap6181.sh
851637        0xCFEB5         End of Zip archive
851799        0xCFF57         Zip archive data, at least v2.0 to extract, compressed size: 166, uncompressed size: 302, name: system/init/ipcam.sh
852133        0xD00A5         End of Zip archive

Scan Time:     2017-03-11 13:01:44
Target File:   /tmp/test/_CH-sys-48.61.75.81.bin.extracted/system/init/ipcam.sh
MD5 Checksum:  2bcaf9df32c1f005cf43a5c13ab13e77
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
215           0xD7            Unix path: /system/system/bin/updata
241           0xF1            Unix path: /system/system/bin/wifidaemon &
273           0x111           Unix path: /system/system/bin/upgrade &

$  tree _CH-sys-48.61.75.81.bin.extracted                                                                                                                                                                                         ✓ 
_CH-sys-48.61.75.81.bin.extracted
[... 59 lines cut ...]
851119        0xCFCAF         Zip archive data, at least v2.0 to extract, compressed size: 340, uncompressed size: 1137, name: system/init/seq_ap6181.sh
851637        0xCFEB5         End of Zip archive
851799        0xCFF57         Zip archive data, at least v2.0 to extract, compressed size: 166, uncompressed size: 302, name: system/init/ipcam.sh
852133        0xD00A5         End of Zip archive

Scan Time:     2017-03-11 13:04:51
Target File:   /tmp/test/_CH-sys-48.61.75.81.bin.extracted/system/init/ipcam.sh
MD5 Checksum:  2bcaf9df32c1f005cf43a5c13ab13e77
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
215           0xD7            Unix path: /system/system/bin/updata
241           0xF1            Unix path: /system/system/bin/wifidaemon &
273           0x111           Unix path: /system/system/bin/upgrade &

$ tree _CH-sys-48.61.75.81.bin.extracted                                                                                                                                                                                         ✓ 
_CH-sys-48.61.75.81.bin.extracted
└── system
    └── init
        └── ipcam.sh

2 directories, 1 file
```
Thank

devttys0, 2017-04-14:
This sounds like a common problem with tools like `unzip` and `7z`, which binwalk uses to extract ZIP files. That's why it is [suggested that you install java](https://github.com/devttys0/binwalk/blob/master/INSTALL.md#dependencies) on your machine so that binwalk can instead use the `jar` utility, which does a better job of unzipping some zip files than `unzip` or `7z`. Binwalk only falls back to `7z` if `jar` does not exist.

I've actually tested binwalk on basically the same camera firmware, and without `jar` I [had the same issues](https://github.com/devttys0/binwalk/blob/master/src/binwalk/config/extract.conf#L52).

ruiruige, 2018-12-13:
Feel very sorry to reopen the issue, I have meet the same problem, but I can not find a doc about how to config java with binwalk.

I hava tried to add java's bin path to PATH, but when I run binwalk, it still said 

    Extractor.execute failed to run external extractor '7z e -y '%e''

So, how can I make java work with binwalk, so that binwalk can use java's ability to handle .7z files?

CoffeeExpress, 2018-12-17:
What's the value of PATH on your system?

ruiruige, 2018-12-17:
I re-checked my environment just now to reproduce the issue, but  I found it is a mistake.

If I do not install p7zip, it will says 

    Extractor.execute failed to run external extractor '7z e -y '%e''

If I do not add jar into PATH, it will says

    Extractor.execute failed to run external extractor 'jar xvf '%e'': [Errno 2] No such file or directory,

So now I found that the two warnings are as expected because I indeed did not install the  corresponding dependencies.

The solution is to install p7zip, (for example rpm: p7zip-plugins-16.02-10.el7.x86_64.rpm and p7zip-16.02-10.el7.x86_64.rpm) , and should also add jar into PATH

Feel very sorry to disturb everyone. I will close the issue.

CoffeeExpress, 2018-12-20:
Thank you for the update. Glad you got it working.

closed 2018-12-20 by CoffeeExpress