ReFirmLabs/binwalk #256: FiOS-G1100-Quantum-Gateway GPG Encrypted Firmware Identification/Decryption

jameshilliard, 2017-03-14:
This is an odd router I picked up which appears to use gpg encrypted and signed firmware, the firmware images don't seem to get identified by binwalk correctly. I've been trying to figure out a way to extract the gpg decryption keys but so far I have been coming up empty handed, it appears the ttl debug interface is disabled and I haven't managed to get a root shell any other way, it has a very limited chroot shell available over ssh but without read/write I couldn't figure out how break out of it. My guess is I would need to dump the NAND unless I can come up with some sort of exploit.
It seems my router only has the decryption keys for one of these images as well(I was able to see some limited log output over the chroot shell when uploading them).
http://bitcast-a.bitgravity.com/2wire/cms/DOWNLOAD/upgrade/frontier/D4A928/1.03.02.02/bhr4_release_01.03.02.02-FTR_firmwareupgrade.bin.signed
http://bitcast-a.bitgravity.com/2wire/cms/DOWNLOAD/upgrade/frontier/D4A928/1.2.0.36.98.0/bhr4_stepstone_release_1.2.0.36.98.0_firmwareupgrade.bin.signed

Gpg2 at least seems to identify the keys needed:
```
$ gpg2 --list-packets bhr4_stepstone_release_1.2.0.36.98.0_firmwareupgrade.bin.signed 
:pubkey enc packet: version 3, algo 1, keyid EDDA2E82EDC7030C
	data: [2048 bits]
:encrypted data packet:
	length: unknown
	mdc_method: 2
gpg: encrypted with RSA key, ID EDC7030C
gpg: decryption failed: No secret key
```
```
$ gpg2 --list-packets bhr4_release_01.03.02.02-FTR_firmwareupgrade.bin.signed 
:pubkey enc packet: version 3, algo 1, keyid F18B47DF3F881C75
	data: [2046 bits]
:encrypted data packet:
	length: unknown
	mdc_method: 2
gpg: encrypted with RSA key, ID 3F881C75
gpg: decryption failed: No secret key
```

jameshilliard, 2017-03-21:
I managed to find, extract and decrypt the pgp decryption key on the router for one of the firmware images(bhr4_release_01.03.02.02-FTR_firmwareupgrade.bin.signed).
```
-----BEGIN PGP PRIVATE KEY BLOCK-----

lQOXBFUrhvcBCACYLresTT4+S7YAoqctW3VWXtoiFFc/hR+kHZvhpKdQjYirorRy
aYv9xCYc5Y+6Rh/mpQFYIbZoMqxtTZ5kf02kQyXXR7mDhiWu0b3S8q4dUJ/hyy4E
Q1oYZDMX9t9El60AeiB9AEzEpiPlOrT2s77PfexR34e4uQ5GIMZVoSM5WB734ZUW
qumyhPeGg3108m4NpuMitSoRUJW683J0oWFf/b/rXEle+onYaafeAAuZTkFwD8s3
7WwWGPPyOKDUgHi1qpvB3kTs9R+OOJeBFA5Rppx+01BmGdImVXgmfF+VH/rPd6ox
fWHBJ72Quct9oZdufm+d/FNmHFmwJzUwlEPJABEBAAEAB/jEK3SYpvmVVANIzmKy
FTMsIxkM1SuitfgTlhdaxuTm8Ys7tIDm+yd5918p4MFlXP/CUPFqqgp4Rtn+DBAh
e/iZxfUBjXOWF1Z8A+KuCiZno4Z1iXPICwoYZxF10sX7pYldFBDNEZXj6EZdN1AO
s6VD0w7Oe1Z4yBOeUqFXwF+nifN81GwyO3RnUWIpbmeEE94Vz9U/cSnyxwXywChm
dQ8NT+CEP0o+ypkcf1v2KNIcUdrgJ995Z3sLVhsqOc8X+fWfW04HQuEpIX2bLF6+
lQdKKbF0oK15Zf/8FmmGrh9Uh07n87kwDXNpGtDiTNsQgOngYNqL6iPytLKv6xAJ
hiEEAMN2o5RcHwX1y4djcMvNUumfSsmACpnDPBA3mPfwXcmzi5DhWWnWav/waIAC
3jE2bBmVRcHZf5w/P0D86xXOoZMcqG88siAL7Q5xZIS2BWXx07k8UmMkJcRGEcUf
JJzf9gz3cwO4eiSmAeINAuWVALSKTWhACISGtq6K4i2BQ51VBADHUIoYwcvFw+Hx
UPyIlKGMD/dQqvx5gVNhQ+qv7Uo33EFdt689yzZytYGRAL3zAEDTJUDsLVdZk4Be
yZpdAWH1bIuKFpNkKfHXIoT27gDextniARdSBu/OvhAxAEoR07BqZ1TXPK1o9wgz
9aki6SAgqDtSrHvO+qr8S5RZ8yBspQP/ebhgtORfaylmUNVkHg1JEE0xQkyIZp4X
zrw2JmeBWJG1ej/oX5PcAHzKzWV2gdvy9gyvCEcEjYK1X07X59mVLIK3gGEwBJpv
bXqCoTW59r8OdeOMpzzWV22AvdOrLz/LV54CWE8bPibxg1wnKUd823W70MSMSevN
IOubTecJP2E5brQmVmVyaXpvbiBCSFI0IDxldUBncmVlbndhdmVzeXN0ZW1zLmNv
bT6JATgEEwECACIFAlUrhvcCGwMGCwkIBwMCBhUIAgkKCwQWAgMBAh4BAheAAAoJ
EJRf3PS924d/yW4IAJI7t+D40QWz5UXeiSoYb7U7ULFGWqvFRj+14oQgIDwyKvJA
xw2cElESHu1RmbhJ+hKDxm/IMxjRMp8gWyMK5r+HcKJ+S8+9G5I2s3xRmI+Pi3rE
AuYoHws7Pjj60lWQ8M1rT1+k3tkee/ozZ9kGpfrP5FBH1/fOOErUm0gdktA5FmDI
xBLR9gNYJyKyjqng5sKiMZNdf/u5qfC68kHj42UOQXHoKgzIwzdK5vr3UlqQR9Be
k9sitlavGZ5WVs2Y9fHXH7chCvlZxPc+P/WH6lWQJXjJo5t+rjSpONnkjPTg9wrp
O/hyeckVFcJynOwDXG9tIZDh0zzmXQPnsn+LfW+dA5gEVSuG9wEIAMWA1s80e0xk
vw12632EonrOgzdZ2Y5vWWOYta2trWWzB9hlmN1Ae5AFgXI7LTBXqpCfSZVcvP6P
W9BvUjHrZuL5X8lQtaKMIMpvtb2iWVW7HQtDPRH9coMoimpvuY5ZXv2vmqurTnTW
9I/AklyV6evt6Sruug581LIV/WFgmaznPD3dAHmgPPcJiuXYnEQQZlm8iO6rlxAQ
pBxCwly/5pWZ56M0bKCyM3qVgEYi/pjJjrg9ndmDdUV2ADvzMWQ5EUAxYaTGVI8J
2AEUR9nEP7WXAMJt9hpM++oH12OX3ZLzFPnOuj1F50Q+VAZMwq1yQyx8CY6oQqSa
DXzs9wz6M9cAEQEAAQAH/jBkgz2+BEARp2ZrLwRQTWd91lTnpRDrY6Gtt0ZY+dWj
alaxfiUoOZ5uWutcaJQhxt8syGDamkxdYAfQXvlwToNqyveO2RJ890Pi30sZzn3d
HR63WO1hhn9wnYm62mJwr3/FWUaa8NxcFwxqCPK6oNh4MNueJuSJ3avNC4qimsTs
bJ25v3orm91ggVggdIkPIXjc/gozGASR29t73VnJaz/Y88bDc0VQzwUkaTEZKOhp
K++PGkIc+f3iKs86TwmeAOCoYFexMUgWdNRHDl4KK+yA1J6o5Mv1KKzy8JuMKfGk
7qfoi+7WplnLQagfsv1BibH7n/LO8FQ1b2DMRB53ykEEAMZjGFvAGVUTjGhVQDvB
CChtvX2YWNZzFdDzEpGKElNPmeBAagcjBszboaNMX0/GuTYeJbSa+sEz+dKwrMaW
cjESL5HKMIjbvt2k7tkDeJEBo9dITU5bGJ6Nk3uik1vHuebAi5ALGjqKQDIskFQX
y4C5bwF//QbI6o52w0ZSv1pHBAD+3Amnb6pULy0wdNO8OBvwH7y9/5yHi6L5+YaF
OgDPGa10RpXRiPSR/AhDjj6Y5BPIuk/fBZnCh7XAB9yOg/rOunhHSrK7Mhn0/d2m
ngr9UbOxsgdZAryIb26QATh0b/TIoGOoVy3bFFPYp15zanCKd9ABe3HCjTCaUDxs
e6yR8QQAw5GO61wKGaXRIazfJ2K0VYV39APfCTncjS/TJvC0nwy8RzVL5sH4f/uX
VdJkvgMExc6+u30XIaIePVaFQe4LBCZrdxfSLVYEhxZd0pBEl454JXy35CZu92KR
Nq04FpX5mC3VJQgREaBo3JvRCdNECka0dtum8ZD1uirsMZBg2G1Ce4kBHwQYAQIA
CQUCVSuG9wIbDAAKCRCUX9z0vduHf4cWB/9FFUHaJiRYTwnXnPAx/7s/fjrFE+cu
QjCnMhGlJIaAwRJROxIisfnT0J2Myryo+wr1cBxZQOvHHq+llPD4tqmUfQQMsyyy
Fp4pI/o0bRTmzsifCfRNbVU3zbg/WiD6RV90SwVDjVnu+zQDN68XPuWeWGpS5JZS
i0/48qbyCtToMSLlsRCObZKCIARzw3ulqSic8aF6Q3xff3paSj3TZAuwbDmtYNIz
ecT8nfMlydQ9WigmtkOPXCU19J2RlPSKt37ZC3VB51oqu1BSi+q5ObmaXSVUfX0y
KjN9HiHVNBWJakFTAcTsDrCVm3WpTKZkDLS0IQTps/eB46vF7V97AIQ6
=UtFV
-----END PGP PRIVATE KEY BLOCK-----
```

alexis4, 2017-04-20:
i was able to install bhr4_release_01.03.02.02-FTR_firmwareupgrade.bin.signed what i did was i went to http://myfiosgateway.com/#/advanced/fwrestore and i restored to the backup firmware first then i installed the http://myfiosgateway.com/#/advanced/fwupgrade firmware

jameshilliard, 2017-04-20:
Yeah, that's the only one I've been able to install as well, I suspect the keys were rotated at some point so I would need to extract the other keys from a router with older firmware.

alexis4, 2017-04-20:
do you know of any site to get more firmware of compatible firmware for the fios-g1100
i only found some on frontier any others

jameshilliard, 2017-04-20:
Nope, I got these links by calling frontier, if you have Verizon FIOS maybe try calling their customer support and asking if they have any. They give these out for routers that can't be automatically updated due to being behind another router(they are normally remotely updated).

alexis4, 2017-04-20:
i tried they wont's because i put a new cooling system in to help it stay cooled

alexis4, 2017-04-20:
it alway got to hot and started to slow down so i put in a new heat sink and used thermal past to help it transfer heat they use the thermal pad like in old laptops to help them stay cool thermal paste is better than the pad

alexis4, 2017-04-20:
here is a link to other version of the firmware i just found it http://myplace.frontier.com/~firmware/

jameshilliard, 2017-04-20:
> i tried they wont's because i put a new cooling system in to help it stay cooled

Maybe try for a different customer service rep.

> here is a link to other version of the firmware i just found it http://myplace.frontier.com/~firmware/

That looks like a [frontier personal website](https://frontier.com/helpcenter/categories/internet/other-services/upload-my-web-site/get-started/visit-my-site-online) to me.

alexis4, 2017-04-20:
yeah it is from them and i have tried mutile times to get help but no luck same thing every time do you know all of the hard brands and spec because i would have to look into it but dd-wrt does have custom for specific hardware they do not have any for fios-g1100 put they have way build a firmware it's not recommend you could brick router if one falls but this that info it may be a possibility

jameshilliard, 2017-04-20:
Yeah, there's currently no custom firmware available for these, OpenWRT/LEDE would probably be the easiest to port. These have protection to prevent flashing unsigned firmware but I have a way around that at least(although it's currently rather complicated). They firmware images are both signed and encrypted with PGP, the [signing key](https://pgp.mit.edu/pks/lookup?op=vindex&search=0xC32877552D7B4FA1) is also different from the encryption key.

alexis4, 2017-04-20:
i'm not good a coding at all. is it possible to just modify the key to trick the router into think it acually signed. like just past they key of the signed firmware into a modded one

jameshilliard, 2017-04-21:
No, PGP is a strong signing method and is not breakable directly, I was able to get the PGP encryption private key simply due to the fact that the router has to have the ability to decrypt firmware updates and thus has to have the encryption private key stored somewhere on the flash, the same is not true for the signing key(the signing key and encryption keys are not the same), changing the firmware at all would mean the signature would not validate and the router would refuse to flash the firmware. However, the signature validation can be bypassed entirely since I have a way to get a root shell.

Brandonv101, 2017-05-04:
@jameshilliard If you don't mind can you explain how you got shell access. I can only SSH into the admin account I still can't find the root password I already tried ThinkGreen.

jameshilliard, 2017-05-04:
@Brandonv101 You have to enable ssh using tr-069 on the WAN side(there's a built in remote activate-able root ssh backdoor), I set up a local [genieacs](https://genieacs.com/) server to do that. Redirecting the router to a local acs server is a bit tricky though, I originally tried to mitm it but that's not possible since the router verifies the acs server ssl certificate.

You can however change the config file to disable ssl and point it at your own acs server, the config file is aes encrypted but I have some python scripts that can decrypt and re-encrypt the config file so that it can be edited(I had to get some help with reversing the encryption scheme from the assembly for that).

Brandonv101, 2017-05-06:
@jameshilliard There is an option to enable SSH access though in the web UI. If you don't mind can you explain how you changed the config file? Via the web UI or some other method. 

I also found a few hidden firmware rollback and update links assuming that the router is using the 192.168.1.1 IP: http://192.168.1.1/#/advanced/fwupgrade & http://192.168.1.1/#/advanced/fwrestore

Hopefully soon we can decrypt the firmware update and customize it.

jameshilliard, 2017-05-06:
By downloading the config file from the webui decryting it, editing it then re-encrypting it and uploading it back I was able to get tr-069 access and enable root ssh.

I can already decrypt one of the firmware updates with the key posted above but not the other since my router didn't have the key for it. However even if the firmware is decrypted it's not possible to sign it since that signing private key is not on the router at all. It should however possible to bypass the signature checks by flashing a new firmware over the root ssh directly.

What firmware version does your router have? If it has an older firmware I might be able to pull the decryption key from it for the other firmware update file.

kingoflions, 2017-05-06:
jameshilliard
how did u dicrypt ur config.bin file. i had tried everything "binwalk..." with mine from the zte "ZXHN H108N V2.5" router and didn't succed 
this is the config.bin link https://www.dropbox.com/s/ebw8tcleiznbcu3/config.bin?dl=0

jameshilliard, 2017-05-06:
These are the config file encryption/decryption scripts I'm using:
[gwdecrypt.py](https://gist.github.com/jameshilliard/7112235b62dd929d69d7980c979ae7c0)
[gwencrypt.py](https://gist.github.com/jameshilliard/99191b2a2877220041dc8789fa07339a)

kingoflions, 2017-05-06:
when i run it i get this
./gwdecrypt.py
Traceback (most recent call last):
File "./gwdecrypt.py", line 10, in 
from Crypto import Random
ImportError: No module named 'Crypto'

jameshilliard, 2017-05-07:
That's just a python [import error](http://stackoverflow.com/questions/30738083/importerror-no-module-named-crypto). Also make sure you use python 3, pretty sure those scripts do not work with python 2.

0BuRner, 2017-05-07:
Would be awesome to implements these tricks in https://github.com/reverse-shell/routersploit project :)

jameshilliard, 2017-05-07:
The hardest part is probably doing the tr-069 server emulation needed in order to activate the root shell.

Brandonv101, 2017-05-08:
@jameshilliard Ok so my current firmware is 01.04.00.10 and I can downgrade to 01.03.02.03 

GW Internal Router Build Number : 01.04.00.822

I have attached the config file here it's Google Drive since I am not able to use dropbox due to it being full. If you can change it and I can get SSH access I can poke around and see if I can pull anything else. [https://drive.google.com/drive/folders/0Bw7iUjIheIK2bktUWW9xaUczSnc?usp=sharing](url)

The5heepDev, 2017-05-22:
I have a router from 2015, I got it a few days ago completely unopened. The firmware is the original. I may have to dump the NAND unless I can get the TR-069 server to work correctly. Let me know if any of you want files from the older firmware

jameshilliard, 2017-05-22:
@The5heepDev Yeah, if you can get a NAND dump that would be great, that way I can probably pull the PGP key needed for decryption of the stepstone firmware. What firmware version is it on right now? Make sure you don't connect it to the internet otherwise it might auto-update.

The5heepDev, 2017-05-22:
Yup, I dont plan on connecting it to the internet. Let me see if uBoot says the fm version

jameshilliard, 2017-05-22:
You can see the firmware version in the webui.

The5heepDev, 2017-05-22:
I don't have the router on me at the moment, I can check for sure when I get home later today

Here's a line from uBoot:

14:36:29.515  Booting:  Bhr4 ,  Version:  01.04.00.10 (build:  01.04.00.822)

jameshilliard, 2017-05-22:
That may be the newer firmware version.

The5heepDev, 2017-05-22:
hmm ok, I'll see if I can roll it back

The5heepDev, 2017-05-22:
Also, what information do you have on the actual hardware? Any datasheets etc? I'm trying to get as much data as possible, and sharing resources is better than finding the same thing all over again lol

jameshilliard, 2017-05-22:
Not a whole lot, seems to be a cortina ARM SoC.

The5heepDev, 2017-05-22:
Cortina G4, ARMv7

I can get the specific model later

The5heepDev, 2017-05-22:
Also, I have some c files from cortina that uBoot is running. It seems to be opening a rw console on UART0. There is apparently UART0, UART1, UART2, and UART3

http://snapon.lab.bufferbloat.net/~d/verizon_firmware/linux/drivers/tty/serial/serial_cortina.c

The5heepDev, 2017-05-22:
In addition, this may be very useful: http://snapon.lab.bufferbloat.net/~d/verizon_firmware/

Brandonv101, 2017-05-23:
@The5heepDev You can downgrade I left a link to the hidden WebUI page in an earlier comment.

The5heepDev, 2017-05-24:
I made a git repo with all the information from this thread written up. Add info to the repo as you get it. I can give access to the main contributors.

https://github.com/The5heepDev/FIOS-G1100

The5heepDev, 2017-05-24:
there is a bunch of new info I've found in my repo, check it out

https://github.com/The5heepDev/FIOS-G1100

Brandonv101, 2017-05-25:
@The5heepDev Thanks this is going to be really useful. Just to ask has this router been out since 2014 because some of the firmware dumps are from 2013/2014 unless it's an old compile/linux kernel.

mhenigma, 2017-05-26:
Exciting to see this comment thread. Would be great to get a custom firmware running on the G-1100.

The5heepDev, 2017-05-26:
@mhenigma https://github.com/The5heepdev/FIOS-G1100

vido89, 2017-05-30:
@kingoflions Did you was able to decrypt your configs ? I have ZTE ZXV10 H201LV2 and I cant find aes key ?

alexis4, 2017-07-14:
@Brandonv101 the firmware you listed is older mine was last update 1 years ago it has version 1.2.1.36.84 i put open source bhr4_release_01.03.02.02-FTR_firmwareupgrade.bin on it that was the only one that i got that was accepted

Brandonv101, 2017-07-14:
@alexis4 I was not able to decrypt the firmware because my internet was out for a week and I got a new router with a newer web UI.

mzpqnxow, 2017-11-13:
I've not had any luck binwalking the decrypted firmware.. binwalk does seem to identify a large amount of Java class files, but they don't get successfully get dumped to disk. I am working with a decrypted version of bhr4_release_01.03.02.02-FTR_firmwareupgrade.bin.signed

jameshilliard, 2017-11-13:
Strange, it worked fine for me once it was decrypted. Did you install all binwalk dependencies?

mzpqnxow, 2017-11-13:
As far as I know, I use binwalk very regularly.. I’ll take a look and see
if there’s anything new I might have missed

CoffeeExpress, 2018-04-23:
This has been a busy thread. Should I leave it open?

jameshilliard, 2018-04-23:
Sure, I don't think support for identifying and extracting the firmware images is supported by binwalk yet, so may be good to leave it open until someone gets around to that at least.

minanagehsalalma, 2019-04-17:
@kingoflions  where are you from dude ???
Can we know each other  ????????

jameshilliard, 2019-07-07:
FYI looks like someone [found](https://medium.com/tenable-techblog/verizon-fios-router-authenticated-command-injection-f6d2ddec30fd) a command injection vulnerability [here](https://github.com/tenable/poc/blob/master/verizon/verizon_g1100_cmd_injection.py). This probably allows for getting a root shell much easier than my method of using the tr-69 backdoor.

minanagehsalalma, 2019-07-08:
> These are the config file encryption/decryption scripts I'm using:
> [gwdecrypt.py](https://gist.github.com/jameshilliard/7112235b62dd929d69d7980c979ae7c0)
> [gwencrypt.py](https://gist.github.com/jameshilliard/99191b2a2877220041dc8789fa07339a)

@jameshilliard  i got this while trying to decrypt
ValueError: Input strings must be a multiple of 16 in length
it's ZXHN H108N V2.5
and got this while trying to encrypt a decrypted one !
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
....
and router pass view does it with no problem !!

jameshilliard, 2019-07-08:
> it's ZXHN H108N V2.5

How is that relevant? The decryption/encryption scripts are specific to these greenwave routers since they use a custom encryption scheme, they aren't going to magically work on routers made by ZTE.

minanagehsalalma, 2019-07-08:
@jameshilliard  sorry i though it was for zte because your reply was for 

> jameshilliard
> how did u dicrypt ur config.bin file. i had tried everything "binwalk..." with mine from the zte "ZXHN H108N V2.5" router and didn't succed
> this is the config.bin link https://www.dropbox.com/s/ebw8tcleiznbcu3/config.bin?dl=0

Nostradamus1973, 2019-10-04:
It's been a while since I've seen any progress on this thread.
Has everyone given up?
I'm not a programmer, or an Electrical Engineer, so, I can't attest to the difficulty of this exercise and I don't want to come off as someone who know's anything about technical side of this. However, that being said maybe there's another way. If there's still interest maybe someone following this issue has contacts that work at GreenWave Systems and maybe they can throw a bone(hint) this way by saying how they'd go about it, or even to go as far as giving someone a boot-loader( on the DL). This router has been around for a while, so, maybe the possibility of finding a (disgruntled)employee with knowledge of how to go about things is higher.
I joined just to add to this discussion, I hope this came off as constructive, if not, pardon my rant and have a great day.

jameshilliard, 2019-10-04:
@Nostradamus1973 I made a [pull request](https://github.com/ReFirmLabs/binwalk/pull/418) to handle firmware decryption.

minanagehsalalma, 2019-10-04:
> @Nostradamus1973 I made a [pull request](https://github.com/ReFirmLabs/binwalk/pull/418) to handle firmware decryption.

@jameshilliard  take a look on nirsoft's router pass view .. i am sure it would help .

jameshilliard, 2019-10-04:
> i am sure it would help

@minanagehsalalma help with what exactly? I already have python scripts to encrypt/decrypt g1100 config files, they use a hard coded AES encryption key.

minanagehsalalma, 2019-10-04:
@jameshilliard okay ... i though  it would help as it Identifies  more than just GPG ..
Sorry if I was wrong .