ReFirmLabs/binwalk #257: Unable to extract firmware

dooves, 2017-03-14:
Hi all,

I'm pretty new to Binwalk but was able to open a firmware from my router with now issues but i have another firmware I am trying to look at but seems to have many issues. one thing i did notice is that the extension was different in the router firmware being a .bin format where the one I am having issues with is .ota format, so unsure if this it the problem.

I have tried a few of the commands and one being garbage unless someone can read it, and was hoping someone can see where I was going wrong.

```
root@kali:~# binwalk -B firmwaretest.ota

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------

root@kali:~#

root@kali:~# binwalk -B --signature firmwaretest.ota

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------

root@kali:~# 

root@kali:~# binwalk -I firmwaretest.ota

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
6             0x6             Broadcom firmware header Corporatio ock FW Version 1.18 foÆP«ãßD©>ju©Uª}ÁÚ2ör¨5"
28201         0x6E29          PC bitmap,
35042         0x88E2          Private key in DER format (PKCS header length: 4, sequence length: -19837
35042         0x88E2          Certificate in DER format (x509 v3), header length: 4, sequence length: -19837
39091         0x98B3          BFF volume entry, AIXv3, file name: "ùúË-~5³æÚpß,$¨¨yÚ¥å¼3eÞ8°	.Ã¹d®­Uî"
44862         0xAF3E          Microsoft executable,
52916         0xCEB4          Private key in DER format (PKCS header length: 4, sequence length: 1237
52916         0xCEB4          Certificate in DER format (x509 v3), header length: 4, sequence length: 1237
62695         0xF4E7          Private key in DER format (PKCS header length: 4, sequence length: 24832
62695         0xF4E7          Certificate in DER format (x509 v3), header length: 4, sequence length: 24832
74579         0x12353         Linux EXT filesystem, invalid state invalid error behavior invalid major revision rev -1052847673.-16066, ext2 filesystem data, UUID=a080fe76-6213-6dff-f315-e13b0fc70fc7, volume name "­"
77664         0x12F60         VxWorks symbol table, little endian, first entry: [type: function, code address: 0xD9D32028, symbol address: 0xD575704C],,,,,,,,,
86209         0x150C1         Microsoft executable,
87169         0x15481         Microsoft executable,
90806         0x162B6         Private key in DER format (PKCS header length: 4, sequence length: -14601
90806         0x162B6         Certificate in DER format (x509 v3), header length: 4, sequence length: -14601
102550        0x19096         Microsoft executable,
114235        0x1BE3B         ARJ archive data, header size: -13260, version 83, minimum version to extract: 107, , original name: "¸¿z", original file date: 1924-06-29 13:01:00, compressed file size: 1978558718, uncompressed file size: 1530400409,
135111        0x20FC7         BFF volume entry, compressed, file name: "¿3Á¢¤E
                                                                               ßÃwù0Ôï-[bèð]zÇ©Óä NgË&=x             y)+CÊ¨Fã¿§²mâvØÞz.ïpGøÃóÝÌÓAÚù³ðãàÆ"                   è4	¡Ý~
137663        0x219BF         Linux EXT filesystem, invalid state invalid error behavior invalid major revision rev -123704865.-1888, ext4 filesystem data, UUID=02f53106-f863-0f06-f9e6-3e4e61da61da, volume name ""
140356        0x22444         BFF volume entry, file name: "ÄÍj2
                                                                   v.râm©[àÓ³<©íéúYåìfw¨ýfÚJº3iÎ£«¯ÔRÃÊEË`k¿¹ÿªYh§ãÿ=ÿÜè{¢J'"
147444        0x23FF4         Private key in DER format (PKCS header length: 4, sequence length: -7907
147444        0x23FF4         Certificate in DER format (x509 v3), header length: 4, sequence length: -7907
152867        0x25523         Microsoft executable,
159070        0x26D5E         ARJ archive data, header size: -8837, version 90, minimum version to extrac@5P", original file date: 2024-06-07 18:50:56, compressed file size: -1755795992, uncompressed file size: -1126761428,
162499        0x27AC3         PC bitmap,
162726        0x27BA6         PC bitmap,
195902        0x2FD3E         BFF volume entry, file size: 1898047593, com<éo@UÏ" size: -1178797762, file name: "Çï32ÁúÅÌâÙ¾¿yswÐß7á¼`Æû>E£åM¿2öA
200993        0x31121         BFF volume entry, compressed, file size: 71808697, compressed size: 626242046, file name: "å²0¦Õe¢4#0$.îÙð
                                                           ²¨¥ÛV
                                                                 +âóÈYÇÎtf2¥¢îJå©ë\6»Ëª8"
205209        0x32199         BFF volume entry, file size: 888027438, compressed size: 406968329, file name: "9DRöâ¿vc ýìmÁ"
212544        0x33E40         BFF volume entry, file size: 1888594126, compressed size: -339225507, file name: "0³h"
225155        0x36F83         Microsoft executable,
247791        0x3C7EF         Microsoft executable,
256424        0x3E9A8         ARJ archive data, header size: 28482, version 115, minimum version to extract: -71, flags, , original name: "´o4H<ú4'uºoT{ôVª²1s¿(Äi$±«õCi$bö²ôryi¸C
                             ^
`nö!_0RÖxð¢°Ë|\ÒÃ7äÄgDX®&¸
                                ", original file date: 2035-08-25 08:41:35, compressed file size: -401043983, uncompressed file size: 1795588124,
257424        0x3ED90         ARJ archive data, header size: 25143, version 126, minimum versiõO?'çxLÅÍæz¡ùz@2xÐÒ¦zÑCVýÐ¶)ú[k3tà origina6l&ìb9g¥!Åé]ziÊe[Üz!Í&                                                                EZr?", original file date: 1966-08-08 17:06:17, compressed file size: 604682379, uncompressed file size: -1024286453,
260560        0x3F9D0         Linux EXT filesystem, invalid state invalid error behavior invalid major revision rev 301990183.4608, ext4 filesystem data, UUID=eb59e322-4bbc-8af3-fd94-49048be18be1, volume name "çÊºp8ÍÜ:Vú·ãÆf6£æøI!öA»}2Ü']", original file date: 1959-12-11 16:28:09, compressed file size: 2146998628, uncompressed file size: 986496361,
321189        0x4E6A5         BFF volume entry, compressed, file name: "ì)kèq'ÍÃeÛç|´ö%Ù9Ùº±ý \Õ÷^­xøåù]Ê²Ë`¦É +$¼|Åsß"
328680        0x503E8         ARJ archive data, header size: -25273, version 105, minimum version to extract: -53, flags, multi-volume, backup, , original name: "<4õQ±bFûNÿbp¨M95.NÔ3Q¯½C@ÓÛ±ý`µ¼
[... 16 lines cut ...]
429705        0x68E89         Microsoft executable,
434681        0x6A1F9         Microsoft executable,
461265        0x709D1         Linux EXT filesystem, invalid state invalid error behavior invalid major revision rev 979726833.14949, ext4 filesystem data, UUID=ba9a9154-033e-aa24-8aaf-02aec1c2c1c2, volume name "çÎÇA	"
462693        0x70F65         Microsoft executable,
463434        0x7124A         ARJ archive data, header size: 16706, version -120, minimum version to extract: 42, flags, slash-switched, backup, , original name: "xRÆà", original file date: 1942-05-09 14:11:49, compressed file size: -1179029425, uncompressed file size: 1008391850, os: Amiga
482943        0x75E7F         Linux EXT filesystem, invalid state invalid error behavior invalid major revision rev -1058964674.-16159, ext2 filesystem data, UUID=d20ccf9d-3d2b-0081-f8c2-64f847804780, volume name "2TÞ>±xÝqïâ=(7îdÎ:áÐiÌ ÅûÖnª|Nì®uö ®ñ:¯|íBY$'}Þ$Ãqê±
                                        êXÎqØw
Èå=ï!üºÙ©m¯Ø&:Ó
                      `SCÛ"
488483        0x77423         ARJ archive data, header size: 14543, version -92, minimum version to extract: -109, multi-volume, , file type: 7-bit text, original name: "¹×ûñ:#òa/¾0c±ãý ¥p", original file date: 1957-02-20 21:17:03, compressed file size: -201829846, uncompressed file size: 820535770,

root@kali:~#

root@kali:~# binwalk --verbose firmwaretest.ota

Scan Time:     2017-03-14 15:16:37
Target File:   /root/firmwaretest.ota
MD5 Checksum:  c1a30c172e538ad916d2307fcfc81542
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------

root@kali:~# 
```

devttys0, 2017-04-14:
What firmware image are you  providing binwalk, and what do you expect binwalk to find? Obviously you typically don't know the specific answer to this, but generally if the firmware is running Linux, you can expect to find some compression (LZMA or gzip typically) and a file system (SquashFS, CramFS, etc). Binwalk can typically identify and extract these easily.

If the firmware is not running Linux, there may still be some compressed sections (again, Binwalk should be able to identify and extract most of these), but sometimes the file is just a bunch of code and data. No file systems, no compressed files, nothing for binwalk to do. Everything is already "extracted"! You'll need a disassembler to start looking at the code.

So, what is the size of your `firmwaretest.ota` file? Linux images are usually at least several MB in size, while non-Linux firmwares (usually running a smaller RTOS) may only be a few hundred kB, maybe less.

Have you run `strings` against the file? Lots of ASCII strings strongly suggests that there is no compression, and you just have raw data and code in the file.

Have you run binwalk with the `-A` option to see if there is any code that binwalk thinks it recognizes? This also can be a good indication that you're dealing with raw data and code in the file.

Have you run binwalk with the `-E` option to view entropy? This is a good way of identifying possible sections of compressed or encrypted data (these sections would have very high, flat entropy).

sundhaug92, 2017-04-21:
-I means show signatures marked as invalid, which is why you're getting so many false-positives (evident by the amount of Mojibake) and unlikely combinations (firmware containing both Microsoft executables and EXT-filesystems). 

It is possible that the firmware doesn't use any currently recognized signatures. If you can you provide us with some information about what kind of router the firmware is for and where to download the firmware, we might be able to deconstruct it and enable binwalk to recognize the signatures.

truyenthongmang, 2017-09-09:
Excuse me, i have some firmware file ip camera. When using binwalk with some option like -A, -E it show some result (image attachment)

But when using with option -e to extract this file firmware, it does not show anything
(link to download the file firmware: https://drive.google.com/file/d/0B_d17G80f5ikQl9pZ1AxT0UxNzA/view?usp=sharing )

[image]
[image]

closed 2018-04-17 by CoffeeExpress