ReFirmLabs/binwalk #260: HUGE uncompressed size in binwalk output

shivang1989, 2017-04-06:
I'm trying to run binwalk on a firmware but the command 'binwalk firmware.bin' outputs LZMA compressed data which has uncompressed size as 3468924096+ Gigabytes. 

> binwalk firmware.bin
`DECIMAL       HEXADECIMAL     DESCRIPTION
----------------------------------------------------------------------

384272        0x5DD10         Base64 standard index table
428268        0x688EC         Base64 standard index table
454224        0x6EE50         LZMA compressed data, properties: 0x6C, dictionary size: 33554432 bytes, uncompressed size: 3468924096053250820 bytes`

What is going wrong ? Am I missing something ?

vido89, 2017-04-06:
Can you provide as a link to actual firmware ?

itdaniher, 2017-04-06:
Dollars to donuts it's not actually LZMA... This sounds like a false positive, but too soon to say without firmware.bin...

shivang1989, 2017-04-13:
Sorry for late reply :( 
Here is the firmware -- http://bit.ly/2pwaRTG

devttys0, 2017-04-14:
To be honest, there doesn't seem to be anything in this firmware image to "binwalk". It looks like straight, uncompressed, unencrypted, code and data (tons of ASCII strings too). Given the size of the firmware, it's definitely not running Linux, and there is unlikely any form of file system in the firmware image. I don't know what the architecture is, but it sounds like you need a disassembler, not binwalk!

As for the LZMA issue, I suspect that you are using an older version of binwalk, or that you are running binwalk under Python2 and do not have the `python-lzma` module installed (this module is packaged with Python3 by default). If this module is available, binwalk uses it to validate possible LZMA compressions, and I do not see any valid LZMA results listed when running binwalk on the firmware image:

```
$ binwalk firmware.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
384272        0x5DD10         Base64 standard index table
428268        0x688EC         Base64 standard index table
```

0BuRner, 2017-04-14:
I guess you're the original author of this post (because it seems to be the same firmware) but someone replied with a so interesting and complete answer that I have to share it : https://reverseengineering.stackexchange.com/questions/15088/lzma-file-format-not-recognized-details-enclosed

shivang1989, 2017-04-19:
Yes that question belongs to me. The answer there is interesting and now I am completely confused. 
I too feel strongly that its not a running linux. It's just 500KB *sighs*. 
The firmware's strings show paths like /application/something/something/<some-file.c>. 
What is this pointing to ? Where are all this directories ? 
Apologies for such silly questions but I have just started with firmware analysis.

vido89, 2017-04-19:
@shivang1989 Those paths do not point to anything those are just comments

shivang1989, 2017-04-20:
@vido89 Oh I see! I think I will have to dig deeper some other way because this bin file contains nothing of interest. 
I will update here (or that stackexchange thread) if I find anything interesting. 

Thank you guys for the pointers.

closed 2017-04-20 by shivang1989