ReFirmLabs/binwalk #263: Binwalk API Scan Function Frozen

ksbickmore, 2017-04-21:
I'm using the binwalk python api to try and find the rootfs of a Netgear Firmware.  
Specifically [DGND3700v2](https://drive.google.com/open?id=0B8zKXtEpeLpWZnJpYmFZcFZHOTQ).

'for module in binwalk.scan(self.item, "-e", "-r", "-y", "compressed", signature=True, quiet=True)`:

I'm using this to recurse into the directory structure and try to find the rootfs.  After a few levels I reach 
 [DGND3700v2.bin](https://drive.google.com/open?id=0B8zKXtEpeLpWem41YUVDRUIwRlk) and attempt to extract the contents.  The scan ends up freezing about half-way through the extraction at around 11000 files.  Admittedly there are over 20000 items to be extracted but it never finishes.  Memory doesn't seem to be the problem as it's barely being used here.

Using Binwalk directly, instead of the api, does extract all the files.

sundhaug92, 2017-04-21:
That's strange, how's the CPU-usage? Which version of binwalk are you using?

ksbickmore, 2017-04-21:
I've been working on v2.1.2b.

Normally I run this and it freezes and has stayed frozen all day.  Once it got to around 11000 files extracted the process would still exist but use no CPU cycles. I reproduced the same problem about 10 times over the past week.

Of course, now that I've posted my problem, I ran the script a few more times and for some reason it is now, partially at least, working.  Instead of freezing it simply stops extracting and falls out of the scan about halfway through.

devttys0, 2017-04-22:
Can you please provide some more information on how you're running binwalk from you Python script? The following Python script works fine for me on the DGND3700v2.bin file that you provided:

```python
#!/usr/bin/env python

import sys
import binwalk

binwalk.scan(sys.argv[1], extract=True, signature=True)
```

If you can provide the actual code that you are using that would be great, if not, can you provide the options being passed to binwalk.scan and possibly any other relevant details about the environment that you're running in?

One thing to check for is to look and see if the `jefferson` utility is running when you execute binwalk. This is the utility that is responsible for performing JFFS2 file system extraction, and you should be able to see it running using the Linux `top` utility. It takes some time to extract this file system, so it runs for a few minutes, and pegs the CPU at ~98% during that time (at least on my system).

sundhaug92, 2017-04-22:
If you're running 2.1.2b, it might be beneficial to try re-pulling from git and re-installing, in case it was accidentally fixed.

ksbickmore, 2017-04-23:
I'm running this in Ubuntu 16.04.  I have mainly been working with python3, though occasionally I switch to python2 just to see if things are different at all.

@devttys0 

I don't every see jefferson running, though I do have it installed.  Using the script you posted above works fine for me and I get 4000.jffs2 out of it, though that is not the part that gives me problems.  The below code has been the problem.  This should pull out over 20000 files, instead it either freezes or stops around 11000 and then goes through the rest of the for loop.

```python
for module in binwalk.scan("DGND3700v2.bin", "-e", "-r", "-y", "compressed", signature=True, quiet=True):
```

devttys0, 2017-04-23:
OK, since you're telling binwalk to only report results that have "compressed" in the name, it will never report the JFFS2 file system, and hence never run the `jefferson` utility.

I'm running some tests with the same command line options as you at the moment, but it is going to take some time; this scan is significantly slowed by the fact that every single zlib compressed block in the JFFS2 file system has to be identified and decompressed individually. I'll let you know what I find.

However, I'm curious as to why you think there should be 20,000+ files extracted by binwalk? Binwalk only finds ~11,000 results with the `-y compressed` flag:

```
$ binwalk -y compressed DGND3700v2.bin | grep compressed | wc -l
11329
```

ksbickmore, 2017-04-23:
Perhaps I'm wrong but when I originally ran binwalk I ended up with over 20000.  This is why I was confused.  The main problem for a while was simply that it would freeze and I couldn't figure out why.

devttys0, 2017-04-23:
If you originally ran binwalk without the `-r` flag, then you'd get two files for every result: the original compressed file that was carved out of the firmware image, and the decompressed version of that file. Otherwise, I'm not sure how there would have been 20,000+ files generated by binwalk for this firmware image.

So far for me, nothing has frozen. Binwalk is still finding, extracting, and decompressing all the Zlib compressed data. But it's only about half way through at the moment, so I'll have to see if it freezes at some point.

ksbickmore, 2017-04-23:
I misunderstood the `-r` option then, thank you for the clarification on that.  It no longer freezes for me either, I don't know why it started working after a week of running it over and over.

devttys0, 2017-04-23:
I tested your script just for good measure, and got the same results, so I think everything is OK. 

Out of curiosity, why limit binwalk to just "compressed" results here? If its the actual file system files you're interested in, you'll want binwalk to find and extract the JFFS2 file system (those Zlib compressed data blobs are just part of the JFFS2 file system).

ksbickmore, 2017-04-23:
It's only "compressed" in this particular instance.  At other points in the script we extract other parts of the firmware while looking for the file system.  Any tips on actually finding the JFFS2 root file system? So far JFFS2 has caused me some problems.  For instance the .jffs2 file never seems to have anything in it. I'm no expert with binwalk so maybe I'm not using the right options.  With this firmware and most other JFFS2 firmwares I've never actually been able to find the file system.

devttys0, 2017-04-23:
Binwalk should drop the `.jffs2` file to disk and then extract its contents to a directory named `jffs2-root`. More specifically, binwalk uses [jefferson](https://github.com/sviehb/jefferson) to extract the contents of the `.jffs2` file. In this case, it generates what appears to be a complete file system:

```bash
$ binwalk -e DGND3700v2.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
16314         0x3FBA          Sercomm firmware signature, version control: 0, download control: 256, hardware ID: "ABL", hardware version: 0x4100, firmware version: 0x8, starting code segment: 0x0, code size: 0x7300
16377         0x3FF9          Sercomm firmware signature, version control: -31463, download control: 480, hardware ID: "", hardware version: 0x37A, firmware version: 0x2, starting code segment: 0x1000, code size: 0xEE44
16384         0x4000          JFFS2 filesystem, big endian

$ ls _DGND3700v2.bin.extracted/jffs2-root/fs_2/
bin  config  dev  etc  home  kernel_cksum  lib  mnt  proc  sbin  sys  tmp  usr  var  vmlinux.lz  www  www.eng
```

As long as you have `jefferson` properly installed, I would expect that you would get the same results.

ksbickmore, 2017-04-23:
Interesting, I will need to reinstall everything and try again.  I get the same binwalk output:

`$ binwalk -e DGND3700v2.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
16314         0x3FBA          Sercomm firmware signature, version control: 0, download control: 256, hardware ID: "ABL", hardware version: 0x4100, firmware version: 0x8, starting code segment: 0x0, code size: 0x7300
16377         0x3FF9          Sercomm firmware signature, version control: -31463, download control: 480, hardware ID: "", hardware version: 0x37A, firmware version: 0x2, starting code segment: 0x1000, code size: 0xEE44
16384         0x4000          JFFS2 filesystem, big endian`

But I do not have a rootfs:
`ls _DGND3700v2.bin.extracted/
4000.jffs2`

devttys0, 2017-04-23:
I suspect it's an issue with the `jefferson` installation; if `jefferson` was not installed at all, binwalk should have spit out a warning to that effect when it tried to extract the `4000.jffs2` file. Maybe try to run `jefferson` manually against the `4000.jffs2` file and see if it gives any warnings or errors?

ksbickmore, 2017-04-23:
Yea it was a problem with jefferson.  After fixing that it works fine.

sundhaug92, 2017-04-23:
Doesn't binwalk also check the return-code from external extractors @devttys0?

closed 2018-04-17 by CoffeeExpress