ReFirmLabs/binwalk #265: Feature-request: General support for packet-based formats

sundhaug92, 2017-04-22:
A lot of formats are based on "packets", where a header might or might not come after another at non-fixed positions. Examples include GPG-signed/encrypted data and network-packets. With the current offset-required format, such signatures are either difficult or impossible to implement.

CoffeeExpress, 2018-04-23:
Do you have an example firmware where this is an issue?

sundhaug92, 2018-04-23:
@CoffeeExpress 

1. #256 could benefit from it, since it'd mean you could get more details out of binwalk (for example something like `gpg --list-packets --verbose`)
2. Adding support would benefit x509-certificates and packaging-formats that use per-file and/or per-directory headers (instead of a single directory-listing)

Could this be solved by using basic signatures and unique python plugins? Sure but I think the codebase would benefit from a unified implementation, which would also make it easier for new signatures.

jameshilliard, 2019-07-06:
I did manage to get gpg detection/decryption working in #418.

devttys0, 2024-09-23:
Binwalk v3 provides much more flexibility in signature parsing; signatures are free to parse data however they see fit (though, you do have to write code - not as simple as a text file definition - but these are the trade offs...).

closed 2024-09-23 by devttys0