ReFirmLabs/binwalk #272: binwalk cant walk through this bin file

kingoflions, 2017-05-02:
hello . i have this issue , i cant do anything with this config file "zte router config backup" ZXHN H108N V2.5
i tried all the command and no thing show up is it encrypted or somthing ?
this is my config.bin from my router
https://www.dropbox.com/s/ebw8tcleiznbcu3/config.bin?dl=0

devttys0, 2017-05-02:
This file does appear to be encrypted, or at the least obfuscated or using a compression that binwalk is unaware of. Do you have a link to the firmware image for this router?

kingoflions, 2017-05-02:
yes i think this is the firmware http://www.mediafire.com/file/6kdg6ew0v99hzaf/ZXHN+H108N+V2.5+TE+Data.rar
thanks brother

AlexHilgertBRA, 2017-11-11:
I know it is old, however, someone can find it useful in the future.
The config file is encrypted using AES, and for this router the key is "GrWM2Hz&LTvz&f^5"

You can decode it using this tool:
https://forum.antichat.ru/threads/proga-router-scan.398971/page-180#post-4089778

Cheers.

kotfarepra, 2018-03-18:
@AlexHilgertBRA how did you found the key ? i tried some keys i found but couldnt decode a file i have
can you explain how to get the key ?
Thank you

codezeroo, 2018-03-22:
hey guys  i wana to ask how about ZXHN H218N can decode the config.bin with binwalk or offzip and can't get the router key to use decode_zte_config.py .
anyhelp ?

E3V3A, 2018-11-26:
@AlexHilgertBRA  
Do you still have that tools somewhere?

minanagehsalalma, 2019-04-16:
@kingoflions @kotfarepra 
@E3V3A  
did any one find the tool !!!!!!!!!!!!!!!!!!!!!!!!!!!

JpegXguy, 2019-08-21:
Try with RouterPassView. It has been updated with ZTE AES capabilities since the OP was made.

minanagehsalalma, 2019-08-21:
> Try with RouterPassView. It has been updated with ZTE AES capabilities since the OP was made.

@JpegXguy  mate  i found it ... but neither of them can encrypt the config file after the decryption !

JpegXguy, 2019-08-21:
What model do you have?

minanagehsalalma, 2019-08-21:
@JpegXguy  "What model do you have?"
ZXHN H108N V2.5
 .... 
"It probably could be done "
i can't find proper steps or enough info to do it 
here is the tool any way .... but i couldn't even extract the script from it to edit it !
https://yadi.sk/d/qg4nOG6MU6Dfnw

JpegXguy, 2019-08-22:
Yeah that's the script I originally used as well. Try asking around in the antichat forum that's linked above for possible encryption.

minanagehsalalma, 2019-08-22:
> Yeah that's the script I originally used as well. Try asking around in the antichat forum that's linked above for possible encryption.

i have already did but no one answered .

JpegXguy, 2019-08-22:
Good. Other than that you can get a understanding of the process from the python script (change extention to .py if you want to run) and maybe reverse it. It's not as simple as I make it sound because you have to know where the segments would be when you have a plaintext file and stuff. 
[decode_zte_config.txt](https://github.com/ReFirmLabs/binwalk/files/3527913/decode_zte_config.txt)

It might actually be easier to open the router casing and see if you can change stuff in the command line, connected with serial cable. (though that needs login as well)

minanagehsalalma, 2019-08-22:
> Good. Other than that you can get a understanding of the process from the python script (change extention to .py if you want to run) and maybe reverse it. It's not as simple as I make it sound [...]

Really ! Thanks for the text file I have tried to extract it with so many tools and failed .. Never though of converting it to txt . I would try my best to see if I could do anything although this would be a bit hard for me ... I will ask on sites and forums too " how can I add encrypting function to a decrypting script " 
@JpegXguy  if you found any thing new  keep me updated ;) 
See you later .

JpegXguy, 2019-08-22:
A note: The exe is completely different from a python script. I just uploaded the plaintext .py script I originally got from the forum as txt because of github upload policy

minanagehsalalma, 2019-08-22:
> A note: The exe is completely different from a python script. I just uploaded the plaintext .py script I originally got from the forum as txt because of github upload policy

You mean this won't work on my router ?

JpegXguy, 2019-08-22:
You should try it anyway. No, what I mean is that I didn't convert the .exe. The exe you found is probably the .py file bundled with python already, for Windows.

minanagehsalalma, 2019-08-22:
> You should try it anyway. No, what I mean is that I didn't convert the .exe. The exe you found is probably the .py file bundled with python already, for Windows.
@JpegXguy 
Yup it's the same and it worked ... still missing the encryption function ..  
thanks again :)

benghanou, 2023-01-01:
Key decode config.bin Router ZTE F680 and F660
(ZTE F680 config.bin https://mega.nz/file/0ioB0JJT#XRp_NClzwL36_JqIdoqUfKlGWvqfRP_Wjmc-TPCJWJ8)
(ZTE F660 config 1 https://mega.nz/file/92xhiLpB#-YxniqzoQeNEGkMegtXr9kdQ1z-I5ibPBwDUctQhGms)
(ZTE F660 config 2 https://mega.nz/file/Bi4n1RBI#64vRHP24N6_dU7XuZROZL3svaLB4xa9O4f1Z6JMbWMY)

closed 2024-09-23 by devttys0