ReFirmLabs/binwalk #275: plot_entropy fails in pyqtgraph ImageExporter

Grazfather, 2017-05-23:
```python
  File "/usr/local/lib/python3.6/site-packages/pyqtgraph/exporters/ImageExporter.py", line 70, in export
    bg = np.empty((self.params['width'], self.params['height'], 4), dtype=np.ubyte)
TypeError: 'float' object cannot be interpreted as an integer
```

In the entropy model I can see that the plt object's `height()` and `width()` methods both return floats.

I worked around it in ImageExported by converting to int explicitly, but not sure if it can/should be fixed in binwalk.

devttys0, 2017-09-14:
The lastest code in master now uses matplotlib instead of pyqtgraph, so this should not be a problem anymore!

Grazfather, 2017-09-19:
I wish I remember what this was from (and I had the code), so I could confirm, but I'll take your word for it.

closed 2017-09-19 by Grazfather