ReFirmLabs/binwalk #280: Cannot save entropy plot in the absense of X

fake-name, 2017-07-05:
```
vagrant@vagrant:/code/Misc/Trimble/MSMobile5_503_1250_OperatingSystem_TSC2⟫ binwalk -E -J wat.txt

DECIMAL       HEXADECIMAL     ENTROPY
--------------------------------------------------------------------------------
0             0x0             Falling entropy edge (0.198120)
QXcbConnection: Could not connect to display
Aborted
```

`wat.txt` is a text file containing the three bytes `wat`. 

This is a VM without any UI components. I understand that displaying the file entropy is probably not reasonable, but there's no reason *saving* a file should fail.

devttys0, 2017-07-06:
This is a known issue, addressed in the [FAQ](https://github.com/devttys0/binwalk/wiki/Frequently-Asked-Questions#how-do-i-plot-entropy-graphs-without-an-x-server). Unfortunately, the pyqtgraph graphing library relies on QT functionality that requires an X server, even if just generating an output file and not displaying it (there is a StackOverflow discussion on the matter [here](https://stackoverflow.com/questions/23739129/exporting-a-plot-in-gui-less-pyqtgraph)).

The FAQ describes how to use X virtual frame buffers to satisfy this X server requirement on GUI-less systems.

fake-name, 2017-07-07:
Whoops, I missed that in the FAQ. Derp.

FWIW, it's not that I don't know how to use a XVFB to work around the issue, it's that installing any windowing environment makes a huge mess and installs a giant number of additional packages, and I'm running headless to (among other reasons) keep the linux install small.

Have you considered another graphing backend? If you're just throwing up simple static graphs (rather then dynamic stuff), matplotlib is quite nice, and doesn't have the attachment to QT that pyqtgraph does.

devttys0, 2017-09-14:
For this, and various other reasons, the latest commits of  binwalk have switched to using matplotlib instead of pyqtgraph. This should alleviate any X server issues!

fake-name, 2017-09-14:
Fancypants! Matplotlib is really nice too.

closed 2017-09-14 by fake-name