ReFirmLabs/binwalk #289: How to extract invalid results?

kbron, 2017-08-24:
hello all

I'm doing some tests to check how well binwalk can identify certificates inside a nonvol_cfg file from a cable modem.

I know the file has 5 certs, but binwalk only displays them when the -i option is on

How can I extract those invalid results?

thank you

devttys0, 2017-08-27:
Binwalk doesn't have any extraction rules for certificate files, but if you want binwalk to carve these files out of the original binary, you can use the [--dd](https://github.com/devttys0/binwalk/wiki/Usage#-d---ddtypeextcmd) option. 

You can also build your own [extraction rule file](https://github.com/devttys0/binwalk/blob/master/src/binwalk/config/extract.conf) rather than specifying the --dd option every time you run binwalk.

kbron, 2017-08-27:
thank you for the response

in case of using the --dd option, would ".*" do the trick for cert files?

binwalk extracted some certs (with -e) but not others...so what's going on exactly? 
when you say there's no extraction rule for certificate does that means there's no signature for certificates? but if that's the case, how come binwalk extract some certs but not others? thank you

kbron, 2017-08-27:
this is another issue, maybe you can help me. I installed binwalk on ubuntu using sudo apt-get install binwalk and everything was running fine. I tried to run lzma extractor from w00tsec and I noticed it needed to import binwalk, I thought there was a binwalk python module so I tried to install it cloning it and running python setup.py install. Now binwalk is broken! when I run I get:
```

Traceback (most recent call last):
  File "/usr/local/bin/binwalk", line 25, in <module>
    import binwalk
  File "build/bdist.linux-x86_64/egg/binwalk/__init__.py", line 3, in <module>

```

I tried to remove an reinstall but got no luck

kbron, 2017-08-27:
this is another issue, maybe you can help me. I installed binwalk on ubuntu using sudo apt-get install binwalk and everything was running fine. I tried to run lzma extractor from w00tsec and I noticed it needed to import binwalk, I thought there was a binwalk python module so I tried to install it cloning it and running python setup.py install. Now binwalk is broken! when I run I get:
```

Traceback (most recent call last):
  File "/usr/local/bin/binwalk", line 25, in <module>
    import binwalk
  File "build/bdist.linux-x86_64/egg/binwalk/__init__.py", line 3, in <module>

```

I tried to remove and reinstall but got no luck

devttys0, 2017-08-31:
> in case of using the --dd option, would ".*" do the trick for cert files?

It should, provided your `--dd` option is properly fomatted. I would do something like this, provided the word "certificate" is in binwalk's output for the certificates you are trying to extract:

```
binwalk --dd=certificate:crt
```

> binwalk extracted some certs (with -e) but not others...so what's going on exactly?

The current version of binwalk in github has no extraction rules for certificate files, so I'm assuming that you specified some extraction rules, either via `--dd` or using your own extraction rule file? In which case, I'd have to see the extraction rule(s) and the binwalk output to really answer this question.

> when you say there's no extraction rule for certificate does that means there's no signature for certificates?

Signatures and extraction rules are two completely different things. Just because binwalk can identify a file signature, doesn't mean it knows how to extract that file type. So binwalk can have signatures for certificate file types X and Y, but only have an extraction rule that matches type X. In which case, you'll see binwalk report both file types X and Y to the screen, but only extract type X when run with the `-e` option.

kbron, 2017-09-18:
thank you so much for your atention

kbron, 2017-11-04:
is there any tutorial on how to write Extraction Rules? I found detailed information how to create signatures but the section for extraction rules is lacking explanation for n00bs like me. thank you

closed 2024-09-23 by devttys0