ReFirmLabs/binwalk #292: Allow extraction with --raw

haasn, 2017-09-11:
The goal: un-split a concatenation of matroska files into separate .mkv files

This finds all of the offsets but I can't find a way to actually save the found segments:
```
binwalk -R '\x1a\x45\xdf\xa3' example.mkv
```

As a work-around, I can use this:
```
echo '0 belong 0x1a45dfa3 ebml' > ebml; binwalk -m ebml -D 'ebml:mkv' /mem/test.mkv
```

Would it be possible to make --dd or --extract work directly with --raw, so I don't have to go through the libmagic indirection?

(Also, the libmagic file for [matroska](https://raw.githubusercontent.com/file/file/master/magic/Magdir/matroska) fails parsing with --magic, so I had to construct a simpler file that only identifies ebml)

devttys0, 2017-09-11:
I've pushed a commit to master that should fix this. You should now be able to extract raw signatures with the `--dd` option:

```
binwalk -R '\x1a\x45\xdf\xa3' --dd='raw:mkv' test1.mkv
```

Binwalk's magic file format is not completely compatible with libmagic's. I've added an equivalent EBML signature to the master branch.

haasn, 2017-09-11:
Thanks, that was fast!

closed 2017-09-11 by haasn