ReFirmLabs/binwalk #293: Some configuration questions

Chenyuxin, 2017-09-15:
I am working on macOS.
My python version is 2.7.when i was execute 'binwalk -Me xxx.bin',it was shown some warnings.
"The Python LZMA module could not be found.",then I try to install lzma for python by brew/ easy_install / pip. But all of this way does't work!

other problems, It's seem couldn't build sasquatch on macOS....

Thank you .

`➜  firmware binwalk -e DIR100_v5.0.0EUb3_patch02.bix

WARNING: The Python LZMA module could not be found. It is *strongly* recommended that you install this module for binwalk to provide proper LZMA identification and extraction results.

WARNING: The Python LZMA module could not be found. It is *strongly* recommended that you install this module for binwalk to provide proper LZMA identification and extraction results.

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
4             0x4             Realtek firmware header, ROME bootloader, image type: RUN, header version: 1, created: 9/17/2011, image size: 1690484 bytes, body checksum: 0xA, header checksum: 0x67
5256          0x1488          LZMA compressed data, properties: 0xA2, dictionary size: 65536 bytes, uncompressed size: 36046402092079104 bytes
9253          0x2425          LZMA compressed data, properties: 0x66, dictionary size: 16777216 bytes, uncompressed size: 6875982256127962083 bytes
9286          0x2446          LZMA compressed data, properties: 0x65, dictionary size: 33554432 bytes, uncompressed size: 7022364572722946380 bytes
9338          0x247A          Unix path: /home/release/teresa/110915_DIR100_release/RealTek865x_062/linux-2.4.x/arch/mips/inflater
13046         0x32F6          mcrypt 2.2 encrypted data, algorithm: blowfish-256, mode: CBC, keymode: 8bit

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -le -d 'squashfs-root' '%e'': [Errno 2] No such file or directory, 'sasquatch -p 1 -le -d 'squashfs-root' '%e'' might not be installed correctly

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -be -d 'squashfs-root' '%e'': [Errno 2] No such file or directory, 'sasquatch -p 1 -be -d 'squashfs-root' '%e'' might not be installed correctly
646032        0x9DB90         Squashfs filesystem, big endian, version 2.0, size: 1041011 bytes, 528 inodes, blocksize: 65536 bytes, created: 2011-09-15 09:11:28`

devttys0, 2017-09-16:
I don't really run binwalk on OSX much, I've never used bew, and I rarely use easy_install/pip, so I don't have a lot of insight into this problem but the bottom line is that the LZMA modules you've installed are either being installed to locations where your Python interpreter cannot find them, OR they are some other LZMA modules that are not API compatible with the standard Python LZMA module. The easiest thing to do I think would be to install binwalk using Python3, which has LZMA built-in.

The sasquatch tool is really just a set of patches for the unsquashfs utility to support vendor-customized SquashFS variations. IIRC, the unsquashfs utility is not supported for OSX, so it likely will not build properly. Since binwalk's auto-extraction feature relies heavily on external utilities, it's really [best to run binwalk on a Linux machine/VM](https://github.com/devttys0/binwalk/wiki/Supported-Platforms) if you want to use that feature.

domints, 2024-12-19:
Almost 7 years later, binwalk got version 3, which depends on sasquatch (earlier it worked with unsquashfs, I think, or it delivered its own binary), and it broke extracting sqashfs :) 

This: https://formulae.brew.sh/formula/squashfs provides unsquashfs, this: https://formulae.brew.sh/formula/binwalk provides binwalk. But they are not happy together. 
I am pretty sure binwalk worked for me in older version. How can this be fixed?