ReFirmLabs/binwalk #295: Installation Error

invicta88, 2017-09-22:
Hello everyone! 

I am having trouble installing binwalk. This is supposed to be a very simple, one-line process, but somehow my computer does not like it. I am running a VirtualBox on Mac OS X Mavericks and have installed Ubuntu 16.04 in the VirtualBox. When I tried to install binwalk using Python 3, I get the following error:
```
Traceback (most recent call last):
  File "setup.py", line 27, in <module>
    label = subprocess.check_output(["git", "describe"], stderr=DEVNULL).decode('utf-8')
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 693, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'git'
```
Can anyone share some insights on this issue? Thanks!

devttys0, 2017-09-23:
Sorry, this is a bug in a recent change to `setup.py` which occurs when the `git` command is not installed on the system. I just committed a patch to the master branch which should fix this.

invicta88, 2017-09-23:
Thanks! For future reference, I used 
```
sudo apt-get install git
```
to install git and now binwalk installs just fine.

closed 2017-09-23 by invicta88