ReFirmLabs/binwalk #296: binwalk extracts empty cramfs-root directory lot of *.zlib files

prasanna510, 2017-10-06:
Hi,
You can find the .bin file in the following location:
https://www.phoenixcontact.com/online/portal/us?uri=pxc-oc-itemdetail:pid=2832658&library=usen&pcck=P-08-08-10-01-01&tab=5

Under Software
Firmware Package 
FW_3.49.zip

I used binwalk -e <file.bin> and got empty "cramfs-root" directory and many zlib files. I want to extract the file system in this firmware.

Advise please.

Clerkosh, 2018-05-04:
Same here, any new information on that problem?

CoffeeExpress, 2018-05-22:
No updates as of right now, but I can put it on my to-do list to try and take a look at some point.

decidedlygray, 2018-06-05:
Temporary workaround - If you just need to extract the CramFS filesystem, a workaround is to use p7zip directly:

```
$ 7z x FL_SWITCH_LM_3_49.bin -o_FL_SWITCH_LM_3_49.bin.extracted_cramfs
```

This should successfully extract the `/bin`, `/en` and `/img` directories for that firmware image into a local folder named `_FL_SWITCH_LM_3_49.bin.extracted_cramfs`

## Additional Debugging Info

Here are some additional steps I took while trying to figure out what was going wrong.

### Swapping Extractor with -D
For whatever reason I'm not able to swap out the extractor using `-D`. I tried:
```
$ binwalk -e -D 'cramfs:cramfs:7z x %e' FL_SWITCH_LM_3_49.bin
$ binwalk -e -D 'cramfs filesystem:cramfs:7z x %e' FL_SWITCH_LM_3_49.bin
$ binwalk -e -D 'cramfs filesystem:cramfs:7z x %e -o%%cramfs-root%%' FL_SWITCH_LM_3_49.bin
$ binwalk -e -D 'cramfs filesystem:cramfs:7z x %e -o'%%cramfs-root%%'' FL_SWITCH_LM_3_49.bin
```

### Running with `sudo` Privs

It appears that one of the fallback cramfs extraction options only works when you run with sudo:
https://github.com/ReFirmLabs/binwalk/blob/2e5a6af9d4643e206bf773858e100084ffbae5db/src/binwalk/config/extract.conf#L70

Testing this, running `sudo binwalk -e FL_SWITCH_LM_3_49.bin` did not result in successful extraction. An empty volume gets mounted, and no additional information is extracted.

As noted above, using the `7z` tool directly worked fine.

tostercx, 2020-08-23:
Got stuck with a similar problem, running `cramfsck` directly should reveal more details. For ex:

    $ cramfsck -v -x out_folder FL_SWITCH_LM_3_49.bin
    cramfsck: superblock magic not found

or in my case:

    $ cramfsck -v -x out_folder 1F4000.cramfs
    ...
    l 0777         7   501:245 out_folder/bin/wget -> busybox
    d 0755      1900   501:245 out_folder/dev
    c 0644    5,   1   501:245 out_folder/dev/console
    cramfsck: mknod failed: out_folder/dev/console: Operation not permitted

It's kind of bad that it fails silently, had no idea extraction had failed halfway trough since it produced some files.

devttys0, 2024-11-27:
This appears to extract correctly with Binwalk v3

closed 2024-11-27 by devttys0