ReFirmLabs/binwalk #297: Cant Unpack Kernel of media player

mrvanity, 2017-10-20:
Hello!
I have used your tool in unpacking the whole image of media player 2.6.32.59.
I got the whole file system unpacked and it great.
Uboot is used, and after unpacking uboot, i get the vmlinux.bin file. This is the compiled kernel and in there there is initrd with some custom files. 
Binwalk cant unpack it.Only an empty cpio.
The vmlinux.bin file is publicly available at:
https://goo.gl/GN1FKm

Could you give me a hand?

My regards.

devttys0, 2017-10-22:
Do you know what custom files you are expecting to find in the kernel image? As far as I can see, the CPIO image is the only thing in there besides the kernel code/data (there is no other data that looks like it would be compressed/encrypted either), and it just contains a special device file `/dev/console`. This file is extracted correctly if you run binwalk as root (creation of devices files requires root privileges).

mrvanity, 2017-10-24:
Hello !
Thank you for your time.
This media player can be created with some custom files that are copied at boot.
I couldn't find them anywhere else, and so i guessed that the kernel file, might have had initrd.gz embedded.

It was not the case.
I rerun binwalk -I on the unpacked firmware file, and i saw some hex addresses (marked as possibly invalid) that really helped me proceed.

Thank you for your fantastic tool and please excuse my bad english

closed 2018-04-11 by CoffeeExpress