ReFirmLabs/binwalk #301: problems extracting squashfs-image

schoerg, 2017-11-13:
I am trying to extract a squashfs image:

`Scan Time:     2017-11-13 01:51:24
Target File:   /mnt/hdd/temp/fritz/var/tmp/filesystem.image
MD5 Checksum:  9cbfa202cc5acd8c13788d1b4978268a
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Squashfs filesystem, big endian, version 4.0, compression:xz, size: 32122637 bytes, 5246 inodes, blocksize: 65536 bytes, created: 1971-01-07 18:57:17
`

However, the output folder is empty. I tried it with `sasquatch` and `unsquashfs` to no avail.

Some hexdump from the beginning of the file:

`00000000  73 71 73 68 00 00 14 7e  01 ea 27 0d 00 01 00 00  |sqsh...~..'.....|
00000010  00 00 01 6b 00 04 00 10  02 c0 00 01 00 04 00 00  |...k............|
00000020  00 00 00 00 9c 0e 1c 84  00 00 00 00 01 ea 27 0d  |..............'.|
00000030  00 00 00 00 01 ea 27 05  ff ff ff ff ff ff ff ff  |......'.........|
00000040  00 00 00 00 01 e8 9d 17  00 00 00 00 01 e9 3f 6b  |..............?k|`

The firmware is from a Fritzbox router: https://ftp.avm.de/fritz.box/fritzbox_6890_lte/firmware/deutsch/FRITZ.Box_6890_LTE.162.06.84.image

decidedlygray, 2018-06-05:
This still happens with the latest `squashfs-tools` as well as `sasquatch`. Not technically a `binwalk` bug though, as it just uses those for extraction.

Here is a workaround that should solve your problem: use 7zip as your squashfs extractor:
```
dg@ubuntu:~/workspace/binwalk_issue_301/var/tmp$ binwalk -e -D 'squashfs:squashfs:7z x %e' filesystem.image

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Squashfs filesystem, big endian, version 4.0, compression:xz, size: 42501273 bytes, 5280 inodes, blocksize: 65536 bytes, created: 1971-05-07 21:54:33

dg@ubuntu:~/workspace/binwalk_issue_301/var/tmp$ binwalk _filesystem.image-0.extracted/0

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             ELF, 32-bit MSB MIPS64 executable, MIPS, version 1 (SYSV)
681592        0xA6678         End of Zip archive, footer length: 22
689384        0xA84E8         Unix path: /sys/bus/usb/devices
704128        0xABE80         SHA256 hash constants, big endian
828872        0xCA5C8         ELF, 32-bit MSB MIPS64 executable, MIPS, version 1 (SYSV)
846299        0xCE9DB         Neighborly text, "neighbor_sa_imageet_neighbour_by_name"
849028        0xCF484         Neighborly text, "neighbor_all_isakmp_sasmalloc"
849062        0xCF4A6         Neighborly text, "neighbor_all_ipsec_sassock_fastclose"
1034424       0xFC8B8         Unix path: /var/tmp/ike.log
... rest of results truncated ...
```

## Additional Info
Doing a hexdump on the file, I am seeing the xz signature (`fd 37 7a 58`), so I don't think it is an edge case like http://www.devttys0.com/2011/08/extracting-non-standard-squashfs-images/, but I could be wrong.

In any case, 7zip seems to be able to handle it. There may still be more data in the archive though. When you run `7z x filesystem.image`  get get the extracted `filesystem` file which then binwalk can read, but the 7zip extraction ends with an error message:
```
ERROR: There are some data after the end of the payload data : filesystem

Sub items Errors: 1

Archives with Errors: 1

Open Errors: 1

Sub items Errors: 1
```

Fckroun, 2023-02-15:
Did you guys found a solution to extract Squashfs filesystem, big endian, version 4.0 ?
7z doesn't work neither.
My case is this FRITZ firmware : https://drive.google.com/file/d/1CB2-td7iDnMegNPQXzuUVsiSAb57LtbH/view

decidedlygray, 2023-02-16:
@Fckroun I haven't.

These tools may be useful:
can convert squashfs endianness - https://github.com/project-magpie/jffs2dump
unsquashfs may be able to extract it out of the box - https://tldp.org/HOWTO/SquashFS-HOWTO/mksqoverview.html

The file you posted is an executable, and it doesn't seem like there is an easy way to extract firmware from it.

Fckroun, 2023-02-17:
@decidedlygray The executable file is used to update the firmware using a windows device, and going through a hexdump we can clearly identify the magic bytes '73 71 73 68' of a squashfs filesystem.
The problem was that the squashfs-tools do not support version 4 big endian filesystems.

What I found later, was this fork of squashfs-tools by onekey-sec :
https://github.com/onekey-sec/sasquatch
where they developed 'sasquatch-v4be' an alternative to 'sasquatch' for this particular case and it works like a charm.

decidedlygray, 2023-02-17:
@Fckroun  ok, excellent. Good find!

Unsolicited plug: I've used their online scanner service and it's pretty great. I haven't tried out their [unblob ](https://github.com/onekey-sec/unblob) yet, which i think is their replacement for binwalk

Glad you were able to find a solution

devttys0, 2024-09-23:
Binwalk v3 supports this variation of SquashFS

[image]

closed 2024-09-23 by devttys0