ReFirmLabs/binwalk #307: '%%squashfs-root%%' '%e'': [Errno 2] No such file or directory

vido89, 2018-01-04:
Binwalk cant extract firmware for DLink Dir605L. [Firmware] (ftp://ftp2.dlink.com/SECURITY_ADVISEMENTS/DIR-605L/REVB/)
```
$ binwalk -e DIR-605L_v2.11BetaB06_hbrf.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
509           0x1FD           LZMA compressed data, properties: 0x88, dictionary size: 1048576 bytes, uncompressed size: 65535 bytes
11280         0x2C10          LZMA compressed data, properties: 0x5D, dictionary size: 8388608 bytes, uncompressed size: 2531328 bytes

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -le -d '%%squashfs-root%%' '%e'': [Errno 2] No such file or directory

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -be -d '%%squashfs-root%%' '%e'': [Errno 2] No such file or directory
653346        0x9F822         Squashfs filesystem, big endian, version 2.0, size: 89100 bytes, 8 inodes, blocksize: 65536 bytes, created: 2017-11-27 07:39:07

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -le -d '%%squashfs-root%%' '%e'': [Errno 2] No such file or directory

WARNING: Extractor.execute failed to run external extractor 'sasquatch -p 1 -be -d '%%squashfs-root%%' '%e'': [Errno 2] No such file or directory
743476        0xB5834         Squashfs filesystem, big endian, version 2.0, size: 2592128 bytes, 512 inodes, blocksize: 65536 bytes, created: 2017-11-27 07:39:23
```

devttys0, 2018-01-06:
Binwalk is telling you that it can't execute the [sasquatch](https://github.com/devttys0/sasquatch) extraction utility, because it is not installed on your system. Once you install it, those squashfs file systems should be extracted successfully.

vido89, 2018-01-06:
Ok tnx a lot and happy new year

closed 2018-04-05 by CoffeeExpress

mirh, 2018-07-03:
A more.. verbose/straightforward/evident message would be helpful. 
Especially to point out to any windows peasant, that he's not gonna get any further on such OS.