ReFirmLabs/binwalk #308: Unable to recognize samsung firmware

DAddYE, 2018-01-05:
Hi all,

first, thanks for this amazing software!

I'm trying to "walk" through the firmware of the Samsung Iconx (2018). It's publicly available here: https://www.samsungimaging.com/file/download?XmlIdx=320&file=ALL_R140XXUAQL2.bin

Or if you prefer from here: https://az335309.vo.msecnd.net/xmlcontainer/customer/data/firmware/xml/SM-R140FOTA_global_201712210608_s.xml

When I use binwalk on it I don't see anything.

```
foo $ binwalk firmware.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
```

Then I scanned the opcodes and as I imagined I see:

```
foo $ binwalk -A firmware.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
46468         0xB584          ARM instructions, function prologue
386749        0x5E6BD         ARM instructions, function prologue
```

And here the entropy:

[image]

Do you have any idea how I can extract this bin?

Thanks so much!

vido89, 2018-01-05:
I dont think its compressed firmware, binwal says it see ARM assembly instruction

devttys0, 2018-01-06:
Based on the entropy analysis and a quick run of strings, I agree that this doesn't appear compressed. Compressed data would usually have an entropy closer to 1.0, and there are a lot of plain-text strings which are clearly intended for print statements and the like. I don't think this is ARM code though, if it was, I'd expect there to be a lot more ARM assembly instructions identified. 

Based on the size of the firmware, (less than 1MB), this is certainly not a Linux firmware image. It's probably an RTOS of some sort, which means there isn't going to be any file systems with executables in them.

Your best bet is to try to figure out what CPU architecture this device runs, and then disassemble the firmware image.

fei-ke, 2018-02-05:
@DAddYE I got a firmware from [here](http://dl.pars-hamrah.com/index.php?file=/Samsung%20Flash%20Files/R%20Series/R140%20Firmware).  Any idea can flash it to my  Iconx (2018). I want to downgrade my device. 
You can download the file from google drive https://drive.google.com/open?id=1_XIIGUhGJkjMvIfVfjg-Qnbibi0YAoBy

almirus, 2018-02-22:
ALL_R140XXUAQL2.bin it is only patch file I think

closed 2018-04-11 by CoffeeExpress