ReFirmLabs/binwalk #309: Binwalk does not show strings like used to ?

vido89, 2018-01-07:
Im not sure which version it was but here is what binwalk show on 11100 file 

```
$ binwalk 11100

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
7812          0x1E84          JPEG image data, JFIF standard 1.01
8058          0x1F7A          Unix path: /www.w3.org/1999/02/22-rdf-syntax-ns#">
8220          0x201C          Unix path: /ns.adobe.com/xap/1.0/g/img/">
25303         0x62D7          Unix path: /kfdf9k2R8cMvysvJWg/5xP/ADLSzurWTVNMMU/p&#xA;yJFHeXSRevE1EkljNowk4xSSqv2SC1a0qrPjhfyslWy/5xX/ADLgUxTX+iT27U9SGSa5dGoQRVWt&#xA;qH
27374         0x6AEE          Unix path: /ns.adobe.com/xap/1.0/mm/"
27432         0x6B28          Unix path: /ns.adobe.com/xap/1.0/sType/ResourceRef#"
27505         0x6B71          Unix path: /ns.adobe.com/xap/1.0/sType/ResourceEvent#"
27580         0x6BBC          Unix path: /ns.adobe.com/xap/1.0/sType/ManifestItem#">
38474         0x964A          Unix path: /purl.org/dc/elements/1.1/">
772652        0xBCA2C         U-Boot version string, "U-Boot 1.1.6 (Jan 16 2015 - 02:23:52)"
777583        0xBDD6F         POSIX tar archive, owner user name: "s", owner group name: "o_cmd"
```
And here is latest version of binwalk on same file
```

$ binwalk 11100

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
920           0x398           LZMA compressed data, properties: 0xC0, dictionary size: 0 bytes, uncompressed size: 3934209 bytes
972           0x3CC           LZMA compressed data, properties: 0xC0, dictionary size: 50331648 bytes, uncompressed size: 1114112 bytes
7812          0x1E84          JPEG image data, JFIF standard 1.01
175375        0x2AD0F         LZMA compressed data, properties: 0x92, dictionary size: -1845362688 bytes, uncompressed size: 165178275 bytes
185571        0x2D4E3         LZMA compressed data, properties: 0x90, dictionary size: -1872625664 bytes, uncompressed size: 344064005 bytes
254442        0x3E1EA         LZMA compressed data, properties: 0x64, dictionary size: -2090860544 bytes, uncompressed size: 64 bytes
399637        0x61915         LZMA compressed data, properties: 0x90, dictionary size: 1167851520 bytes, uncompressed size: 1075907033 bytes
410458        0x6435A         LZMA compressed data, properties: 0xA2, dictionary size: 1107296256 bytes, uncompressed size: 1073744012 bytes
523673        0x7FD99         LZMA compressed data, properties: 0x88, dictionary size: 786432 bytes, uncompressed size: 2686521344 bytes
725350        0xB1166         LZMA compressed data, properties: 0xA2, dictionary size: 1694498816 bytes, uncompressed size: 16643000484 bytes
726326        0xB1536         LZMA compressed data, properties: 0xC7, dictionary size: 0 bytes, uncompressed size: 706136244389 bytes
754922        0xB84EA         LZMA compressed data, properties: 0x64, dictionary size: 1694498816 bytes, uncompressed size: 16643000480 bytes
772652        0xBCA2C         U-Boot version string, "U-Boot 1.1.6 (Jan 16 2015 - 02:23:52)"
792007        0xC15C7         LZMA compressed data, properties: 0x65, dictionary size: 0 bytes, uncompressed size: 139392413506 bytes
792027        0xC15DB         LZMA compressed data, properties: 0x64, dictionary size: 0 bytes, uncompressed size: 435728772941 bytes
807776        0xC5360         Unix path: /home/stb/PERFORCE/THEALE/utopia/project/kriti_nos_mips_mboot/../../mxlib/hal/kriti/gpio/halGPIO.c
811600        0xC6250         Unix path: /home/stb/PERFORCE/THEALE/utopia/project/kriti_nos_mips_mboot/../../mxlib/hal/kriti/flash/serial/halSERFLASH.c
834604        0xCBC2C         CRC32 polynomial table, little endian
835631        0xCC02F         LZMA compressed data, properties: 0x64, dictionary size: 0 bytes, uncompressed size: 1684809006 bytes
838903        0xCCCF7         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 253403070495 bytes
865352        0xD3448         Unix path: /home/stb/PERFORCE/THEALE/utopia/project/kriti_nos_mips_mboot/../../mxlib/hal/kriti/gop/halGOP.c
868324        0xD3FE4         Unix path: /home/stb/PERFORCE/THEALE/utopia/project/kriti_nos_mips_mboot/../../mxlib/hal/kriti/dac/halDAC.c
876392        0xD5F68         Unix path: /home/stb/PERFORCE/THEALE/utopia/project/kriti_nos_mips_mboot/../../mxlib/hal/kriti/ve/mhal_tvencoder.c
877856        0xD6520         LZMA compressed data, properties: 0x92, dictionary size: 184549376 bytes, uncompressed size: 520093696 bytes
877876        0xD6534         LZMA compressed data, properties: 0xA2, dictionary size: 150994944 bytes, uncompressed size: 520093696 bytes
878056        0xD65E8         LZMA compressed data, properties: 0x51, dictionary size: -184549376 bytes, uncompressed size: 520093696 bytes
878116        0xD6624         LZMA compressed data, properties: 0x91, dictionary size: -318767104 bytes, uncompressed size: 520093696 bytes
902940        0xDC71C         LZMA compressed data, properties: 0x5C, dictionary size: 1276116992 bytes, uncompressed size: 164922687 bytes
903660        0xDC9EC         LZMA compressed data, properties: 0x75, dictionary size: 1276116992 bytes, uncompressed size: 164922943 bytes
915027        0xDF653         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 12884901888 bytes
915071        0xDF67F         LZMA compressed data, properties: 0xBF, dictionary size: -1090519040 bytes, uncompressed size: 90844562705 bytes
915131        0xDF6BB         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 274882101248 bytes
915219        0xDF713         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 549764202496 bytes
915263        0xDF73F         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 137441050624 bytes
915351        0xDF797         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 549764202496 bytes
915439        0xDF7EF         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 34360262656 bytes
915571        0xDF873         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 68720525312 bytes
915615        0xDF89F         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 274882101248 bytes
915659        0xDF8CB         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 274882101248 bytes
915703        0xDF8F7         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 549764202496 bytes
915747        0xDF923         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 68720525312 bytes
915791        0xDF94F         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 549764202496 bytes
915879        0xDF9A7         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 34360262656 bytes
915923        0xDF9D3         LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 274882101248 bytes
[... 51 lines cut ...]
960499        0xEA7F3         LZMA compressed data, properties: 0xC0, dictionary size: 0 bytes, uncompressed size: 584448738565 bytes
961144        0xEAA78         LZMA compressed data, properties: 0x5A, dictionary size: -16777216 bytes, uncompressed size: -1 bytes
1022399       0xF99BF         LZMA compressed data, properties: 0xBF, dictionary size: 16842752 bytes, uncompressed size: 4295033089 bytes
1039421       0xFDC3D         LZMA compressed data, properties: 0x5B, dictionary size: 0 bytes, uncompressed size: 391078096504 bytes
1046047       0xFF61F         LZMA compressed data, properties: 0x87, dictionary size: -2023096320 bytes, uncompressed size: 2272198656 bytes
1052931       0x101103        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 2272321624 bytes
1053311       0x10127F        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 2271409592 bytes
1053759       0x10143F        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 2271492792 bytes
1054311       0x101667        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 2271682736 bytes
[... cut ...]
1058491       0x1026BB        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 6566909468 bytes
1058515       0x1026D3        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 23746768128 bytes
1058587       0x10271B        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 6566912412 bytes
1058659       0x102763        LZMA compressed data, properties: 0x87, dictionary size: 0 bytes, uncompressed size: 6566913308 bytes

```
Where strings are go ?

devttys0, 2018-01-08:
The strings are still in the file of course, but you'll notice that the strings that binwalk specifically was looking for were "Unix paths". In the old version of binwalk, the "Unix paths" it found were actually URLs. The new version has better filtering for identifying actual Unix paths, and the "Unix path" results appear legitimate.

Also, many of those LZMA results appear invalid, false-positive results. I suspect that you are running binwalk in Python2 and don't have the `python-lzma` module installed. Installing this will allow binwalk to validate LZMA results and remove those false-positive signatures that you're seeing.

vido89, 2018-01-08:
Ok I see, I have python 2.7 and python3 dont know which python binwalk uses, probably 2.7. If I run just python I have instaled pylzma

```
$ python
Python 2.7.12 (default, Nov 20 2017, 18:23:56) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pylzma
>>> 
```
Dont know why binwalk not validate LZMA results.
Thank you very much for nice explanation

devttys0, 2018-01-09:
Binwalk can be installed under either Python2 or Python3. Python3 includes an `lzma` module by default, but Python2 does not. Binwalk [expects](https://github.com/ReFirmLabs/binwalk/blob/master/src/binwalk/plugins/lzmavalid.py#L23) there to be a Python module named `lzma`. I don't know what API interface `pylzma` uses, but most Linux package managers have an LZMA module for Python2 named `python-lzma` (e.g., `apt-get install python-lzma`) that provides the same API as the one in Python3.

closed 2018-04-11 by CoffeeExpress