ReFirmLabs/binwalk #311: Unable to Recognize Huawei Modem Firmware (Wrong recognition)

alivelimeli, 2018-01-22:
Hello @devttys0 

I'm trying to identify a Huawei modem firmware binary using binwalk but it doesn't recognize it as jiffs, squash etc

```
$ binwalk -v Huawei.bin 

Scan Time:     2018-01-22 21:34:10
Target File:   /media/data/tmp/Huawei.bin
MD5 Checksum:  e1ddf1d896631b07331b5188d5b31ca2
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
4992299       0x4C2D2B        MySQL ISAM index file Version 1
```

It's definitely not MySQL ISAM index file :)

So, how can I go further? Is there an option to force binwalk to extract it as jiffs file system or squash?

Here is the file:
 
[Huawei.bin.zip](https://github.com/ReFirmLabs/binwalk/files/1653158/Huawei.bin.zip)

binwalk opcodes result:
```
$ binwalk -A Huawei.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
12713598      0xC1FE7E        ARM instructions, function prologue
16463970      0xFB3862        ARM instructions, function prologue

```

Entropy:
```
$ binwalk -E Huawei.bin

DECIMAL       HEXADECIMAL     ENTROPY
--------------------------------------------------------------------------------
0             0x0             Rising entropy edge (0.996797)

```
[image]

vido89, 2018-01-22:
Based on entropy its compressed image, so based on what do you think its jiffs ?

alivelimeli, 2018-01-23:
Hello @vido89,

I assumed that its jiffs because it was for the modem firmware. 

Hexdump result:

```
$ head -n1 Huawei.bin | hexdump -C
00000000  12 14 b2 71 7a 13 8b c1  4f 4f f5 4d d3 e2 67 91  |...qz...OO.M..g.|
00000010  06 c7 60 0c 3a 90 ca eb  e1 87 2f 6f e5 75 70 88  |..`.:...../o.up.|
00000020  e9 ec 8c 0c db 2e d3 89  47 14 b2 b9 b2 83 8e f6  |........G.......|
00000030  88 7c d8 d1 15 4f 8a bf  a7 bc fb fb c8 ce 4d 24  |.|...O........M$|
00000040  c9 f8 36 3d 4f c5 66 81  c1 0d 42 10 89 14 e9 f6  |..6=O.f...B.....|
00000050  af 23 9a 08 38 6f 08 60  3a ca a6 a2 69 62 98 65  |.#..8o.`:...ib.e|
00000060  e9 6a ec c1 72 77 ff ef  8a ee 6e 15 de 2c 55 22  |.j..rw....n..,U"|
00000070  26 af 31 85 f6 51 93 df  c7 77 51 f5 9d b1 96 b8  |&.1..Q...wQ.....|
00000080  30 97 c1 9e 79 88 df 46  72 40 9a ad 85 c9 61 6a  |0...y..Fr@....aj|
00000090  a6 0b a7 e8 7b ea c8 10  9c c9 9f 5f 6e ff b3 11  |....{......_n...|
000000a0  17 89 cb 49 ff b1 41 8c  bc e6 7c 5b fc 6a a9 80  |...I..A...|[.j..|
000000b0  db 38 45 ab 6b db 4c bb  6d 58 9b b8 96 44 61 2d  |.8E.k.L.mX...Da-|
000000c0  26 4e d1 f4 a4 a1 6b 57  17 e3 f3 26 81 73 50 86  |&N....kW...&.sP.|
000000d0  34 e1 84 6b 42 f6 0f 40  81 bf 8a 2d 64 1f 7f cb  |4..kB..@...-d...|
000000e0  62 78 4f 10 ca 1b 2d 19  f7 ab ab a9 cf 12 05 fd  |bxO...-.........|
000000f0  23 da 7f 98 c7 11 02 06  60 20 c9 e7 47 2c d5 8c  |#.......` ..G,..|
00000100  8a d8 52 8e b8 d2 93 36  b3 3c f3 58 71 36 ba d8  |..R....6.<.Xq6..|
00000110  7c d2 27 a2 d8 35 75 99  5a 32 f5 f9 04 c5 a8 44  ||.'..5u.Z2.....D|
00000120  8d 2a 53 c3 54 18 a3 61  ce ea 7b cb 7a d8 0f 83  |.*S.T..a..{.z...|
00000130  fb 1c 40 1e c7 d4 7d ba  a4 fd 36 2a a4 ab ce c0  |..@...}...6*....|
00000140  39 1a 1a a9 0c 73 ec 00  ae 7d b2 69 07 fd 47 00  |9....s...}.i..G.|
00000150  ae 4e 0f ce 53 90 3a aa  5c 61 66 be 95 63 2d 67  |.N..S.:.\af..c-g|
00000160  24 ed 1f fc 75 43 66 79  e0 98 d7 c3 8a 4c ed c1  |$...uCfy.....L..|
00000170  47 35 0f 01 a5 21 de 38  f2 14 ff 7c 3d 9e 2a e9  |G5...!.8...|=.*.|
00000180  9b 64 c9 5a 74 42 15 39  03 c3 a5 9c 85 0f b8 0e  |.d.ZtB.9........|
00000190  03 5e 88 5b 37 a0 f6 75  11 96 b6 c6 d5 6c 55 f9  |.^.[7..u.....lU.|
000001a0  36 92 31 9c 15 cc 61 a3  4c 85 72 07 9b f8 75 ce  |6.1...a.L.r...u.|
000001b0  f6 d3 2d 01 93 e7 e8 ce  30 85 cd da a5 fa ed b3  |..-.....0.......|
000001c0  b1 5d ef 28 ae 64 93 de  fe d3 a1 5b 9f 82 97 25  |.].(.d.....[...%|
000001d0  de ec 09 f3 15 63 ff 8f  dd 1c 57 9d c1 74 9f 7a  |.....c....W..t.z|
000001e0  2d e0 e3 93 24 c2 2c 6d  a7 a9 cb c8 08 81 29 84  |-...$.,m......).|
000001f0  4c 92 5b 88 9c 63 ed b7  fc d8 91 28 7a 9e 2a 18  |L.[..c.....(z.*.|
00000200  1d 04 cb ec 21 de 13 24  93 32 1b 7b fe 44 e9 12  |....!..$.2.{.D..|
00000210  57 60 62 be 8f df f6 7f  e6 83 67 74 44 08 68 78  |W`b.......gtD.hx|
00000220  74 9f cd 76 c2 b4 e3 2e  4d fa 1a cb 0d d2 80 64  |t..v....M......d|
00000230  8f 5e 95 72 e9 62 0a                              |.^.r.b.|
00000237
```
Even I've tried to extract with command `binwalk --dd=".*" Huawei.bin` it didn't worked.

devttys0, 2018-01-26:
With the entropy being as high as it is, this firmware image is either using a very good compression algorithm that binwalk doesn't know about, or it is encrypted. I did a scan for raw LZMA compression streams, in the event that they are using LZMA and just stripped out the LZMA header information, but found nothing. I'd suspect that the firmware is likely encrypted, but without further information it's hard to tell for sure.

E3V3A, 2018-10-13:
@alivelimeli  
How/Where did you find that FW? 
If you somehow extracted it from an update blob, it is likely *encrypted*. But Huawei is know for shitty encryption, so it may be easy to find how it was done. Also for what hardware is it meant? 
(E.g. if it is for a Qulacomm based phone, then we have other options...)

closed 2024-10-05 by devttys0