ReFirmLabs/binwalk #312: Binwalk can't be launched from python with matryoshka depth=1

PerezFederico, 2018-01-24:
for module in binwalk.scan((pth), signature=True, extract=True, matryoshka=True, deph=1):
        #......

This throws the help print on stdout. When I change depth to 2, it works.
I'm really needing this, quick response is apreciated. I can fix it with a bit of guidance.

PerezFederico, 2018-01-24:
As a workaround I changed the matryoshka default value in extractor.py

    Option(short='M',
               long='matryoshka',
               kwargs={'matryoshka': 1},
               description='Recursively scan extracted files'),

devttys0, 2018-01-24:
Hmmmm, I'm not getting this issue. I tried to reproduce it with this code:

```
import sys
import binwalk

for module in binwalk.scan(sys.argv[1], signature=True, extract=True, matryoshka=True, depth=1):
    for result in module.results:
        print (result.description)
```

But it seems to be working fine for me. What version of binwalk do you have installed?

PerezFederico, 2018-01-24:
Binwalk v2.1.2-de42daf

Maybe my concept is wrong. 
Depth of 1 means that after first extraction it will try to extract again or the first extraction is considered depth=1 and second depth=2?

devttys0, 2018-01-24:
A depth of 1 would be the same as running just `--extract` and no `--matryoshka` specified.

PerezFederico, 2018-01-24:
Ok, thanks, so my whole point was invalid, even if you can on not reproduce it. 
Thanks for the help! I love this tool and think is one of the most useful things created for reverse engineering.

closed 2018-01-24 by PerezFederico