ReFirmLabs/binwalk #313: Python3 not extracting

SElyPer, 2018-01-27:
Using the latest master from git, when I try to extract a binary using python3, it only performs signature checks (does not seem to process any options).  Python2 using the same binwalk works just fine.
The command line I used for these captures are:
python3 /usr/local/bin/binwalk -e -M firmware.zip > python3.txt
python2.7 /usr/local/bin/binwalk -e -M firmware.zip > python27.txt

[python3.txt](https://github.com/ReFirmLabs/binwalk/files/1670268/python3.txt)
[python27.txt](https://github.com/ReFirmLabs/binwalk/files/1670269/python27.txt)

JessicaFavin, 2018-10-29:
Hello @SElyPer 
I'm trying to work on this issue, could you provide me the zip file for further testing please ?
Thanks

E3V3A, 2018-11-26:
@SElyPer ?

SElyPer, 2018-11-26:
Good morning,
Unfortunately I no longer have that project so I'm not able to recreate the
issue.
Thanks

Sent from a mobile device.

E3V3A, 2018-11-26:
This may be related to #336. 
Otherwise close.

0xd3xt3r, 2019-11-12:
I have provided one such fix for on this pull request #426 see if this works for you

closed 2024-09-23 by devttys0