ReFirmLabs/binwalk #319: IDA 7.0 plugin incompatable

bmourit, 2018-03-08:
The Binwalk IDA plugin is incompatible with versions 7.0+ of IDA. This is due to the requirement of using the new Actions API.

I have updated the plugin, which works fine for me on 7.0+, but it hasn't been tested on older version. If someone using an older version could verify this works (it should) then this could be pulled.

Here is the updated plugin: https://github.com/bmourit/binwalk/blob/patch-1/src/scripts/binida.py

I'll create a pull request as well.

CoffeeExpress, 2018-04-05:
@bmourit Thanks for the patch. I'll see if we can find some time to take a look at it.

bmourit, 2018-04-05:
No prob. Did it for my own use, but figured I'd share.

closed 2018-09-20 by bmourit