ReFirmLabs/binwalk #324: Help!!! binwalk Can't identify file system

rain6851, 2018-04-07:
This is a hp printer firmware:

Firmware address:
  https://drive.google.com/file/d/1c6ndzNIXJgA6fYu0XNO7wDf0pXWsp-gk/view?usp=sharing

sundhaug92, 2018-04-23:
Using binwalk 2.1.2-f636501

I get 
`DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
88            0x58            Copyright string: "Copyright (c) 2000 Hewlett-Packard Company"
65836         0x1012C         Copyright string: "Copyright 1984-1997 Wind River Systems, Inc.Copyright (c) 1999, Hewlett-Packard Company."
65880         0x10158         Copyright string: "Copyright (c) 1999, Hewlett-Packard Company."
81413         0x13E05         Zlib compressed data, default compression
1378468       0x1508A4        VxWorks operating system version "5.4.2" , compiled: "Oct 17 2002, 04:42:41"
1379256       0x150BB8        Copyright string: "Copyright 1984-1996 Wind River Systems, Inc."
1966158       0x1E004E        Private key in DER format (PKCS header length: 4, sequence length: 605
1967572       0x1E05D4        Certificate in DER format (x509 v3), header length: 4, sequence length: 611
1986072       0x1E4E18        Private key in DER format (PKCS header length: 4, sequence length: 605
1987486       0x1E539E        Certificate in DER format (x509 v3), header length: 4, sequence length: 611
2005960       0x1E9BC8        Private key in DER format (PKCS header length: 4, sequence length: 605
2007374       0x1EA14E        Certificate in DER format (x509 v3), header length: 4, sequence length: 611` 

Guessing the zlib might be the root filesystem, I'd try to extract that

sundhaug92, 2018-04-23:
Running `binwalk -eM` on it (Extract, Matryshkha) renders two files, `13E05` and `13E05.zlib`. The first of which appears to be the extracted filesystem but neither binwalk nor libfile appears to have magic for it, so further details are hard to come by.

devttys0, 2018-05-08:
If you're looking for a Linux or Linux-like file system, I don't think you'll find it here. This is a VxWorks Real Time Operating System, not Linux. The Zlib compressed blob at offset `0x13E05` is the compressed RTOS kernel image. Binwalk extracts this fine, but you'll need to load it into a disassembler to analyze the code and figure out how things work.

closed 2018-05-08 by devttys0