ReFirmLabs/binwalk #330: Support RAR 5.0 archive format

UniQP, 2018-05-22:
I just had the case that binwalk could not detect a RAR file. It turns out that it was a RAR 5.0 file, which has slightly different magic bytes, see https://www.rarlab.com/technote.htm#rarsign .

It would be nice if binwalk could also detect RAR 5.0 files in the future.

devttys0, 2018-05-22:
Detection of RAR 5.0 files should be working in commit bec12a27fed390ba31dc8ba0f00dbec046b90c27. Please verify.

UniQP, 2018-05-22:
Wow, thank you for the quick answer. Unfortunately, it does not work.

```
hexdump -C archiv.rar | head -n1
00000000  52 61 72 21 1a 07 01 00  82 6c 02 c3 0c 01 05 08  |Rar!.....l......|
```

Looks like Byte 9 is classified as invalid.

BTW: The first link in the comment is dead.

devttys0, 2018-05-22:
Thanks, I just checked in b28ee174a565339354148dcd266f289874073d7d which should fix this!

UniQP, 2018-05-22:
Now, it works. Thank you.

closed 2018-05-22 by UniQP