ReFirmLabs/binwalk #332: cramfsprogs removed from Ubuntu

decidedlygray, 2018-05-26:
It looks like this might be a duplicate of #242 

The package `cramfsprogs` has been removed from Ubuntu 18.

Currently the workaround I have in place is:
```
git clone https://github.com/ReFirmLabs/binwalk.git
cd binwalk
# Remove CramFS from APT requirements
sed -r 's/cramfsprogs //' deps.sh>deps_fixed.sh
sudo chmod +x deps_fixed.sh
sudo ./deps_fixed.sh --yes
# Download and install it separately
cd /tmp
wget ftp://ftp.si.debian.org/debian/pool/main/c/cramfs/cramfsprogs_1.1-6_amd64.deb
sudo dpkg -i cramfsprogs_1.1-6_amd64.deb
```

Is this package still required?

Thanks!

decidedlygray, 2018-06-04:
Closing, as this is an Ubuntu issue, tracked here: https://bugs.launchpad.net/ubuntu/+source/cramfs/+bug/1737285

closed 2018-06-04 by decidedlygray

davidribyrne, 2020-08-11:
If it helps anyone, I've created a repo for the tools and updated them to work with with Ubuntu 20.04.

https://github.com/davidribyrne/cramfs

ilius, 2020-09-02:
For Debian, manually downloading [deb file](https://packages.debian.org/jessie/amd64/cramfsprogs/download) and installing with `sudo apt install ./cramfsprogs*.deb` worked for me.