ReFirmLabs/binwalk #333: STM32 Controller board fw

washeree, 2018-05-29:
Hello i have a closed fw 3d controller board. I tried to binwalk but no info, with Entropy like this
[image]
i was trying with stlink to dump memory and i didnt know that uncheck read protection erase flash so i need to go from bin file and reflash it.

The strange thing is if i just setup binwalk without plugins it shows LZMA but i think so its wrong info because it went no where too. I couldnt extract with --dd too.

Processor is STM32F103ZET6 and the firmware is ;
[R2_6.zip](https://github.com/ReFirmLabs/binwalk/files/2047741/R2_6.zip)

binwalk shows with -Y parameter ;

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
42026         0xA42A          ARM executable code, 16-bit (Thumb), little endian, at least 529 valid instructions

binwalk invalid packages are;

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
32204         0x7DCC          Private key in DER format (PKCS header length: 4, sequence length: 8
32204         0x7DCC          Certificate in DER format (x509 v3), header length: 4, sequence length: 8
79833         0x137D9         Private key in DER format (PKCS header length: 4, sequence length: 30913
79833         0x137D9         Certificate in DER format (x509 v3), header length: 4, sequence length: 30913
87903         0x1575F         Private key in DER format (PKCS header length: 4, sequence length: 30913
87903         0x1575F         Certificate in DER format (x509 v3), header length: 4, sequence length: 30913
90353         0x160F1         Private key in DER format (PKCS header length: 4, sequence length: 30913
90353         0x160F1         Certificate in DER format (x509 v3), header length: 4, sequence length: 30913
101360        0x18BF0         VxWorks symbol table, little endian, first entry: [type: function, code address: 0xF05F4C5D, symbol address: 0x7D1EDB35],,,,,,,,,
138813        0x21E3D         EFS2 Qualcomm filesystem super block, little endian, , NAND version 0x666f, 1476395008 blocks, 0x74697753 pages per block, 0x3a6863 bytes per page
138873        0x21E79         EFS2 Qualcomm filesystem super block, little endian, , NAND version 0x7075, 1867281522 blocks, 0x5300003a pages per block, 0x6f707075 bytes per page
139665        0x22191         MPEG transport stream data
155298        0x25EA2         VxWorks symbol table, big endian, first entry: [type: function, code address: 0x240EB0F9, symbol address: 0x240EDFF8],,,,,,,,,
205703        0x32387         Broadcom 96345 firmware header, header size: 256, firmware version: "M992", ~CRC32 header checksum: 0xEB011084, ~CRC32 data checksum: 0xA8F9F768
212335        0x33D6F         LBR archive data
223089        0x36771         Linux EXT filesystem, invalid state invalid error behavior invalid major revision rev -1406533306.-21462, ext2 filesystem data (mounted or unclean), UUID=f74efbef-6026-6099-f898-0000f0fef0fe, volume name "��"
265705        0x40DE9         ARJ archive data, header size: -11672, version -46, minimum version to extract: 2, backup, , original name: "b�`jk���E�jk���E�jk���E
                                               �jk���E�jk���E����@��P���", original file date: 1958-04-23 13:58:08, compressed file size: 1114172000, uncompressed file size: -368951056,
265713        0x40DF1         ARJ archive data, header size: 17000, version 13, minimum version to extract: 2, backup, , original name: "��E�jk���E�jk���E
                                     �jk���E�jk���E����@��P���", original file date: 1958-04-23 17:35:44, compressed file size: -362223008, uncompressed file size: 1114172000,
265721        0x40DF9         ARJ archive data, header size: 17000, version 64, minimum version to extract: 2, backup, , original name: "k���E�jk���E
                              �jk���E�jk���E����@��P���", original file date: 2005-04-22 12:13:20, compressed file size: -362643212, uncompressed file size: 23816800,
265729        0x40E01         ARJ archive data, header size: -5528, version -22, minimum version to extract: 104, flags, multi-volume, slash-switched, backup, , original name: "�jk���E
                                                              �jk���E�jk���E����@��P���", original file date: 1970-10-03 15:46:40, compressed file size: 312082834, uncompressed file size: 386876912,
265733        0x40E05         ARJ archive data, header size: 17000, version -128, minimum version to extract: 98, backup, , original name: "k���E
                      �jk���E�jk���E����@��P���", original file date: 1979-11-22 01:40:34, compressed file size: 386876912, uncompressed file size: 23816913,
267407        0x4148F         ARJ archive data, header size: -11672, version -46, minimum version to extract: 2, backup, , original name: "b�`jk", original file date: 1958-04-23 13:58:08, compressed file size: 1114172000, uncompressed file size: -368951056,
267415        0x41497         ARJ archive data, header size: 17000, version 13, minimum version to extract: 2, backup, , original name: "��E�jk", original file date: 1958-04-23 17:35:44, compressed file size: -362223008, uncompressed file size: 1114172000,
267423        0x4149F         ARJ archive data, header size: 17000, version 64, minimum version to extract: 2, backup, , original name: "k", original file date: 2005-04-22 12:13:20, compressed file size: -362643212, uncompressed file size: 7039584,
267431        0x414A7         ARJ archive data, header size: -5528, version -22, minimum version to extract: 104, flags, multi-volume, slash-switched, backup, , original name: "�jk", original file date: 1970-03-23 11:26:24, compressed file size: 312082578, uncompressed file size: 386876912,
267435        0x414AB         ARJ archive data, header size: 17000, version -128, minimum version to extract: 98, backup, , original name: "k", original file date: 1979-11-22 01:36:18, compressed file size: 386876912, uncompressed file size: 7039697,
270347        0x4200B         ARJ archive data, header size: -11672, version -46, minimum version to extract: 2, backup, , original name: "b�`jk", original file date: 1958-04-23 13:58:08, compressed file size: 1114172000, uncompressed file size: -368951056,
270355        0x42013         ARJ archive data, header size: 17000, version 13, minimum version to extract: 2, backup, , original name: "��E�jk", original file date: 1958-04-23 17:35:44, compressed file size: -362223008, uncompressed file size: 1114172000,
270363        0x4201B         ARJ archive data, header size: 17000, version 64, minimum version to extract: 2, backup, , original name: "k", original file date: 2005-04-22 12:13:20, compressed file size: -362643212, uncompressed file size: 7039584,
270371        0x42023         ARJ archive data, header size: -5528, version -22, minimum version to extract: 104, flags, multi-volume, slash-switched, backup, , original name: "�jk", original file date: 1970-03-23 11:26:24, compressed file size: 312082578, uncompressed file size: 386876912,
270375        0x42027         ARJ archive data, header size: 17000, version -128, minimum version to extract: 98, backup, , original name: "k", original file date: 1979-11-22 01:36:18, compressed file size: 386876912, uncompressed file size: 7039697,
282968        0x45158         Private key in DER format (PKCS header length: 4, sequence length: 1544
282968        0x45158         Certificate in DER format (x509 v3), header length: 4, sequence length: 1544
289001        0x468E9         EFS2 Qualcomm filesystem super block, little endian, , NOR version 0x2044, -1110457905 blocks, 0x-49ff5c4d pages per block, 0x-49184a31 bytes per page
312885        0x4C635         BFF volume entry, compressed, file name: "�_"
312963        0x4C683         BFF volume entry, compressed, file name: ""
313029        0x4C6C5         BFF volume entry, compressed, file name: "�_"
313109        0x4C715         BFF volume entry, AIXv3, file size: 1852137583, compressed size: 1559860409, file name: ""
313199        0x4C76F         BFF volume entry, compressed, file name: "�5ءd3�๔�\��_"
313255        0x4C7A7         BFF volume entry, compressed, file name: ""
313305        0x4C7D9         BFF volume entry, compressed, file size: 1425596384, compressed size: 1612775525, file name: "���aeo �������L"
[... cut ...]
412450        0x64B22         Broadcom firmware header Corporatio . Model: ���. Firmware version: .
414722        0x65402         Broadcom firmware header Corporatio . Model: . Firmware version: .
422172        0x6711C         Linux EXT filesystem, invalid state invalid error behavior invalid major revision rev -1094921004.-16708, ext2 filesystem data (mounted or unclean), UUID=6f0075e8-5760-e315-ad7d-9e0267b667b6, volume name "��o�5���"

Board is chinese so i characters looks strange too.

Thank you for help.

devttys0, 2018-06-06:
I don't think binwalk can really help with extracting the `R2_6.bin` file, as there doesn't seem to be anything to extract. Everything appears to be plain text (lots of ASCII strings, no apparent compression, etc), and the processor is known to be ARM so you'll probably want to turn to an ARM disassembler to examine the firmware in detail.

washeree, 2018-06-12:
Thank you for the reply @devttys0 . I tried ida pro but didnt see something relevent. Which disassembler do you suggest? Thank you for your help too.

bpotchik, 2019-03-09:
@washeree, realize this is an older thread but I was able to successfully load/analyze this image in binary ninja.

washeree, 2019-03-15:
@bpotchik Hello brian, later someone crack the encrypted part so i have fw now. But i will check with binary ninja. Thank you so much.