ReFirmLabs/binwalk #334: Help with a vxworks binary

fansanelli, 2018-06-03:
Hello,
there is a binary that I cannot extract.. I tried with both dd and binwalk without success.
The strange thing is that after "binwalk -e" I got a reversed image file. 
[extracted.zip](https://github.com/ReFirmLabs/binwalk/files/2066192/extracted.zip)
Can you please help me?
I attach the entropy, "binwalk -v" output and the image for inspection.

decidedlygray, 2018-06-04:
Are you able to provide a link to the firmware binary itself?

fansanelli, 2018-06-04:
@decidedlygray Sure! And thanks for the prompt reply!
Please find the link: http://www.mediafire.com/file/um3ejq39mb4ubfx/vxWorks.bin

decidedlygray, 2018-06-05:
Did you end up with something like this?

[image]

I believe that is correctly extracted. This being RTOS, you likely won't find a root filesystem like in Embedded Linux type firmware. See #324 

## Additional Analysis/Speculation

Not being that familiar with RTOS systems, it's tough to say, but the .sit files don't appear to be sit compressed data. They seem to share similar file signatures (`0x53 0x49 0x54 0x44`):
```
$ hd -n 16 83CDDF.sit
00000000  53 49 54 44 48 65 61 64  20 25 70 20 70 53 49 54  |SITDHead %p pSIT|
$ hd -n 16 8433AC.sit 
00000000  53 49 54 44 73 20 2d 20  50 61 72 61 6d 65 74 65  |SITDs - Paramete|
$ hd -n 16 A01CD5.sit 
00000000  53 49 54 44 44 61 74 61  00 00 00 75 73 62 45 78  |SITDData...usbEx|
```

Looking at them in hexdump it looks like some kind of blob of null terminated strings. Maybe some kind of string lookup table used by the operating system? All of the files are very similar:
```
$ hd -n 128 8433AC.sit 
00000000  53 49 54 44 73 20 2d 20  50 61 72 61 6d 65 74 65  |SITDs - Paramete|
00000010  72 73 20 61 72 65 20 6e  6f 74 20 76 61 6c 69 64  |rs are not valid|
00000020  0a 00 45 52 52 4f 52 3a  20 75 73 62 45 68 63 64  |..ERROR: usbEhcd|
00000030  20 2d 20 75 73 62 45 68  63 64 55 74 69 6c 2e 63  | - usbEhcdUtil.c|
00000040  5b 34 38 31 32 5d 3a 20  75 73 62 45 68 63 64 47  |[4812]: usbEhcdG|
00000050  65 6e 65 72 61 74 65 53  49 54 44 73 20 2d 20 75  |enerateSITDs - u|
00000060  6e 61 62 6c 65 20 74 6f  20 63 72 65 61 74 65 20  |nable to create |
00000070  68 65 61 64 20 54 44 0a  00 45 52 52 4f 52 3a 20  |head TD..ERROR: |
```

Then the extracted `A1D889.zlib` is maybe the ROTS kernel image? Next step is probably to throw that into a disassembler, and start reverse engineering efforts.

As for why the bitmap is reversed and upside down: my guess is this is simply (for whatever reason) how the manufacturer is storing that resource, and binwalk is working as expected.

All that said, I am just some random person on the internet trying to help, not affiliated with this project. They may have more insight.

decidedlygray, 2018-06-06:
Hey, glad that was helpful! IDA should be able to handle it I believe.

Apologies if I caused any confusion: I am not an official contributor or maintainer of this project. I don't have the power to close issues. If you are happy with where things are, you can go ahead and close the issue.

Cheers!

closed 2018-06-06 by fansanelli