ReFirmLabs/binwalk #335: deps.sh: Add support for python3-only installs

sundhaug92, 2018-06-05:
At least in Ubuntu, `python` always refers to python 2, while python 3 is `python3`. If only `python3` is installed, deps.sh will fail but it's otherwise a supported configuration AFAIK

decidedlygray, 2018-06-06:
Counterpoint, and I may be totally wrong: isn't `python` just a symlink to `/usr/bin/python2.7`?

I thought the use of the `python` symlink in Ubuntu was so that users could easily adjust which version of python is their primary version, without uninstalling or reinstalling anything. For example, just repointing the `python` symlink to `/usr/bin/python3.6` would now call `python3.6` whenever you execute `python`

devttys0, 2018-06-06:
`build.sh` specifically invokes `python2` when installing utilities that are only python 2.x compliant; for utilities that can run under either python 2.x or 3.x it simply invokes `python`. But since some of the external utilities are only supported in python 2, systems that only have python 3 installed won't be able to run those utilities.

Bottom line, if you *only* have `python3` installed, you should expect some of these external utilities to fail.

E3V3A, 2018-11-26:
@devttys0 
> But since some of the external utilities are only supported in python 2,

Which are these?
It should be fairly straight forward to make them support also P3, no?

Alien-AV, 2019-06-02:
@devttys0
Running the `binwalk` script itself is not possible on python3-only ubuntu, as it begins with `#!/usr/bin/env python`, when there's no "python" symlink or executable present anywhere. I'm assuming that on dual-version (python2+python3) system, python2 is always selected as the interpreter :(

@decidedlygray there's an implicit expectation of a lot of scripts, etc., that `python` points to python2, and creating the symlink might backfire later. (when you do add python2 to your system, for example.)

closed 2024-09-23 by devttys0