ReFirmLabs/binwalk #336: Fixing build system

KOLANICH, 2018-06-06:


KOLANICH, 2018-06-15:
@devttys0

CoffeeExpress, 2018-06-28:
@KOLANICH Thanks for the patch. We'll take a look and see what we can do to get this merged.

Xyc0, 2018-07-20:
I got halfway into patching this myself before thinking to check for PRs.
It's a good fix, please consider merging.

E3V3A, 2018-11-29:
@KOLANICH 
I'm trying to fix this patch, and managed to get it installed, but now I get:
```bash
$ binwalk                                                                                                                                              Traceback (most recent call last):                                                                                                                       File "/usr/bin/binwalk", line 7, in                                                                                                              from binwalk.__init__ import main                                                                                                                  ImportError: cannot import name 'main'
```

Something is not found or named properly. 
What's missing here?

KOLANICH, 2018-11-29:
>```from binwalk.__init__ import main```

should be

```from binwalk.__main__ import main```

according to

```ini
[options.entry_points]
console_scripts = binwalk = binwalk.__main__:main
```

in `setup.cfg`

E3V3A, 2018-11-29:
@KOLANICH 
@devttys0 
@CoffeeExpress 

Yeah, I noticed...while you were writing. :)
But there were a bunch of other funny things going on, that would play well. 

Added email, URL and platform and requirements etc. so that `pip show binwalk` is correct. 

But now its all fixed, and you can just do:

```bash
git clone ...
cd binwalk
pip install .
```
and you can run binwalk! 
:tada:

closed 2024-09-23 by devttys0