ReFirmLabs/binwalk #337: Binwalk doesn't extract anything from Microtik SWOS (SwitchOS)

mzpqnxow, 2018-06-27:
I've tried a few flags, using `binwalk -r -M -e swos-rb250-1.17.lzb` did identify 2 lzma dictionaries at adjacent offsets, I specified those offsets but it produced nothing. Any planned support for Microtik SWOS? FWIW the Microtik Router OS extracts just fine.

I'm wondering if it may be encrypted..

CoffeeExpress, 2018-06-28:
What does the entropy graph look like?

devttys0, 2018-06-28:
I took a look at this firmware, and I'm not sure there is anything to extract. Unlike RouterOS, SwOS is not Linux based, and appears to be a custom RTOS. There are lots of plain text strings, and entropy analysis suggests there is no compression or encryption. I'm not sure what CPU architecture this switch uses, but I would expect that you should be able to load it into an appropriate disassembler for further reverse engineering and analysis.

mzpqnxow, 2018-06-29:
Ah, my bad for not at least throwing mips(el) or arm toolchain objdump -D
at this. I’ll take a further look. Thank you

volkirik, 2019-02-19:
> Ah, my bad for not at least throwing mips(el) or arm toolchain objdump -D at this. I’ll take a further look. Thank you

Any news? I also would like to extract SwOS

MrTomek, 2020-09-27:
I don't know if it will be useful to you, but maybe I'll tell someone something. In the middle of bin misp (swos-css317
-2.12.bin.) I found this: https://gist.github.com/MrTomek/3a352befd578e4f2713d27700496c69d

MrTomek, 2020-09-27:
My impression is that SwOS is only an UI to configure switch chip (more or less the /interface ethernet under ROS)

mzpqnxow, 2020-10-02:
I don’t think there’s anything to be done here that is worth the time investment. @devttys0 has it correct, it’s a proprietary RTOS, not a BusyBox-based Linux system. There’s not a squashfs filesystem ir any other sort of standard/typical filesystem to extract

closed 2020-10-02 by mzpqnxow