ReFirmLabs/binwalk #342: It seems that binwalk doesn't identity the libpcap file

wtdcode, 2018-07-24:
Here is my libpcap file : 

[flows.zip](https://github.com/ReFirmLabs/binwalk/files/2223876/flows.zip)

but binwalk found nothing with the magic 0xa1b2c3d4

```
/m/i/C/i/m/flows $ binwalk

Binwalk v2.1.1
Craig Heffner, http://www.binwalk.org

....

/m/i/C/i/m/flows $ binwalk flows.pcap

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
```

devttys0, 2018-07-25:
Thanks, there was a bug in the pcap signature. It should be fixed in the latest commit to master.

wtdcode, 2018-07-25:
It seems that it was fixed in commit [be7df82241c7e706177faf6018f8af1c6eef4dc2](https://github.com/ReFirmLabs/binwalk/commit/be7df82241c7e706177faf6018f8af1c6eef4dc2), so this issue will be closed.

closed 2018-07-25 by wtdcode