ReFirmLabs/binwalk #344: Manually constructing the sites-packages path prevents installation into virtualenv

Xyc0, 2018-07-26:
https://github.com/ReFirmLabs/binwalk/blob/bcdd75a4608407217fd3f5947a8eb0afe2d7bc9e/src/scripts/binwalk#L17

devttys0, 2018-08-03:
Can you provide more details? What is the output/errors you are getting? Installing to a virtualenv seems to work fine for me.

Xyc0, 2018-08-03:
I misspoke concerning virtualenv.
I thought `~/.local/lib/python3.6/site-packages` and typed virtualenv.
I think this problem will be fixed with https://github.com/ReFirmLabs/binwalk/issues/336 but here is my output.

`python3 setup.py install`

    creating ./binwalk/src/binwalk/core/version.py
    running install
    running build
    running build_py
    creating build
    creating build/lib
    creating build/lib/binwalk
    copying src/binwalk/__init__.py -> build/lib/binwalk
    creating build/lib/binwalk/magic
    copying src/binwalk/magic/archives -> build/lib/binwalk/magic
    copying src/binwalk/magic/console -> build/lib/binwalk/magic
    copying src/binwalk/magic/linux -> build/lib/binwalk/magic
    copying src/binwalk/magic/ecos -> build/lib/binwalk/magic
    copying src/binwalk/magic/phones -> build/lib/binwalk/magic
    copying src/binwalk/magic/network -> build/lib/binwalk/magic
    copying src/binwalk/magic/ebml -> build/lib/binwalk/magic
    copying src/binwalk/magic/images -> build/lib/binwalk/magic
    copying src/binwalk/magic/encoding -> build/lib/binwalk/magic
    copying src/binwalk/magic/firmware -> build/lib/binwalk/magic
    copying src/binwalk/magic/code -> build/lib/binwalk/magic
    copying src/binwalk/magic/crypto -> build/lib/binwalk/magic
    copying src/binwalk/magic/bootloaders -> build/lib/binwalk/magic
    copying src/binwalk/magic/hashing -> build/lib/binwalk/magic
    copying src/binwalk/magic/binarch -> build/lib/binwalk/magic
    copying src/binwalk/magic/bincast -> build/lib/binwalk/magic
    copying src/binwalk/magic/sql -> build/lib/binwalk/magic
    copying src/binwalk/magic/animation -> build/lib/binwalk/magic
    copying src/binwalk/magic/vxworks -> build/lib/binwalk/magic
    copying src/binwalk/magic/executables -> build/lib/binwalk/magic
    copying src/binwalk/magic/binwalk -> build/lib/binwalk/magic
    copying src/binwalk/magic/compressed -> build/lib/binwalk/magic
    copying src/binwalk/magic/misc -> build/lib/binwalk/magic
    copying src/binwalk/magic/filesystems -> build/lib/binwalk/magic
    copying src/binwalk/magic/lzma -> build/lib/binwalk/magic
    copying src/binwalk/magic/efi -> build/lib/binwalk/magic
    creating build/lib/binwalk/config
    copying src/binwalk/config/extract.conf -> build/lib/binwalk/config
    creating build/lib/binwalk/plugins
    copying src/binwalk/plugins/gzipvalid.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/unpfs.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/unjffs2.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/lzmamod.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/tar.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/zlibvalid.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/zlibextract.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/hilink.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/gzipextract.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/arcadyan.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/dlromfsextract.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/ubivalid.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/jffs2valid.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/cpio.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/lzmavalid.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/lzmaextract.py -> build/lib/binwalk/plugins
    copying src/binwalk/plugins/ziphelper.py -> build/lib/binwalk/plugins
    creating build/lib/binwalk/modules
    copying src/binwalk/modules/__init__.py -> build/lib/binwalk/modules
    copying src/binwalk/modules/hexdiff.py -> build/lib/binwalk/modules
    copying src/binwalk/modules/disasm.py -> build/lib/binwalk/modules
    copying src/binwalk/modules/extractor.py -> build/lib/binwalk/modules
    copying src/binwalk/modules/entropy.py -> build/lib/binwalk/modules
    copying src/binwalk/modules/general.py -> build/lib/binwalk/modules
    copying src/binwalk/modules/compression.py -> build/lib/binwalk/modules
    copying src/binwalk/modules/signature.py -> build/lib/binwalk/modules
    creating build/lib/binwalk/core
    copying src/binwalk/core/__init__.py -> build/lib/binwalk/core
    copying src/binwalk/core/display.py -> build/lib/binwalk/core
    copying src/binwalk/core/compat.py -> build/lib/binwalk/core
    copying src/binwalk/core/settings.py -> build/lib/binwalk/core
    copying src/binwalk/core/common.py -> build/lib/binwalk/core
    copying src/binwalk/core/exceptions.py -> build/lib/binwalk/core
    copying src/binwalk/core/statuserver.py -> build/lib/binwalk/core
    copying src/binwalk/core/version.py -> build/lib/binwalk/core
    copying src/binwalk/core/idb.py -> build/lib/binwalk/core
    copying src/binwalk/core/module.py -> build/lib/binwalk/core
    copying src/binwalk/core/plugin.py -> build/lib/binwalk/core
    copying src/binwalk/core/magic.py -> build/lib/binwalk/core
    running build_scripts
    creating build/scripts-3.6
    copying and adjusting src/scripts/binwalk -> build/scripts-3.6
    changing mode of build/scripts-3.6/binwalk from 644 to 755
    running install_lib
    creating /usr/local/lib/python3.6/dist-packages/binwalk
    error: could not create '/usr/local/lib/python3.6/dist-packages/binwalk': Permission denied

closed 2024-09-23 by devttys0