ReFirmLabs/binwalk #346: OSError: [Errno 22] Invalid argument

TheElementalOfCreation, 2018-07-29:
I keep getting the following at seemingly random points, usually on really large files:
```
Signature Exception: [Errno 22] Invalid argument
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\site-packages\binwalk\core\module.py", line 597, in main
    retval = self.run()
  File "C:\Program Files\Python36\lib\site-packages\binwalk\modules\signature.py", line 189, in run
    self.scan_file(fp)
  File "C:\Program Files\Python36\lib\site-packages\binwalk\modules\signature.py", line 149, in scan_file
    for r in self.magic.scan(data, dlen):
  File "C:\Program Files\Python36\lib\site-packages\binwalk\core\magic.py", line 811, in scan
    tags = self._analyze(signature, offset)
  File "C:\Program Files\Python36\lib\site-packages\binwalk\core\magic.py", line 674, in _analyze
    ts = datetime.datetime.utcfromtimestamp(dvalue)
OSError: [Errno 22] Invalid argument
----------------------------------------------------------------------------------------------------
```

Sometimes it only happens when I do a specific option like '-I' but not when I call it normally, or vice versa.

TheElementalOfCreation, 2018-07-29:
After adding a few lines to the code to print out what the value of `dvalue` was, the error is occurring because it is receiving a NEGATIVE value somehow.

TheElementalOfCreation, 2018-07-29:
Possible solution: Add check for dates that are negative integers. If so, mark them as invalid and skip them, or if invalid mode is turned on, print them out by the timestamp value instead. I've gone and physically looked at what the values would be if they were LE signed, BE signed, LE unsigned, and BE unsigned, and none of them make any sense for the files. It is most likely that, when the invalid tag was not turned on, it was seeing an invalid entry. When the invalid flag WAS turned on, it was DEFINITELY seeing an invalid entry.

For testing, and additional information, I've added the `pprint` module to the imports and I've temporarily replaced the line 
```
ts = datetime.datetime.utcfromtimestamp(dvalue)
```
with
```
try:
    ts = datetime.datetime.utcfromtimestamp(dvalue)
except:
    print('TIMESTAMP:')
    print(dvalue)
    print('SIGNATURE:')
    pprint.pprint(signature.__dict__)
    print('LINE:')
    pprint.pprint(line.__dict__)
    print('TAGS:')
    pprint.pprint(tags)
    raise
```
The result of which was quite informative. On a file that raised errors when invalid was turned on, I got the following result (followed by the same error as before, but with the line numbers shifted accordingly):

```
TIMESTAMP:
-1721742280
SIGNATURE:
{'confidence': 2,
 'id': 13,
 'lines': [<binwalk.core.magic.SignatureLine object at 0x0000000002F3FDD8>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F3F710>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F3F358>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F3F320>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F3F908>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F3F940>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F3FA20>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F3F550>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F81EF0>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F81EB8>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F81550>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F81F60>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F81320>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F6B518>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F6BDA0>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F6B400>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F03B38>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F03898>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F03828>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F037B8>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F03668>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F03B70>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F03D68>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7A58>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7860>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA76A0>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7668>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7E80>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7CF8>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7898>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7278>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7518>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7F98>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA74A8>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7240>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7D68>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002FA7FD0>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F90C50>,
           <binwalk.core.magic.SignatureLine object at 0x0000000002F90FD0>,
           <binwalk.core.magic.SignatureLine object at 0x000000000305F1D0>,
           <binwalk.core.magic.SignatureLine object at 0x000000000305F4A8>,
[... 81 lines cut ...]
LINE:
{'condition': '=',
 'endianess': '<',
 'fmt': 'i',
 'format': '\\b, last modified: %s',
 'level': 1,
 'offset': 4,
 'operator': None,
 'opvalue': None,
 'pkfmt': '<i',
 'regex': False,
 'signed': True,
 'size': 4,
 'tags': {},
 'text': '>4      ledate          x               \\b, last modified: %s',
 'type': 'date',
 'value': None}
TAGS:
{'id': 65,
 'invalid': False,
 'offset': 859202,
 'once': False}
```

Hope all this information helps.

devttys0, 2018-07-30:
I just pushed a commit to the master branch which:

 1. Adds an exception handler in the code to handle these issues in the future
 2. Makes all existing date fields in the magic signatures files unsigned

closed 2018-08-31 by CoffeeExpress