ReFirmLabs/binwalk #348: Linting?

ariccio, 2018-08-15:
You guys ever consider linting? I just ran pylint against binwalk, and while it was very noisy, it did detect some sketchy things.

Insofar as noise, it complains tons about indentation, formatting, slightly too long lines, missing docstrings, unused variables, etc... 

Otherwise it did pick up several dangerous default arguments, some variable shadowing, a couple redefinitions of built-in types, some sketchy comparisons, and a whole bunch of unused imports. If you know about them and they don't bother you, that's fine. Otherwise, I thought I should let you know.

CoffeeExpress, 2018-08-31:
:+1: Definitely something that needs to get attention. Hopefully we can find some time.

CeruleanSky, 2018-09-05:
Seems the minor cleanup after lint testing in https://github.com/ReFirmLabs/binwalk/commit/cd8ca7a79a5f3ba6af4105d412d413a9a7da8c99 was a bit over eager and removed some initialization. I commented a couple including one that breaks lzma

devttys0, 2024-09-23:
Binwalk v3 has been re-written in Rust; most of these types of issues are simply not allowed by the Rust compiler. As for things like indentation and line length I tend to leave as a personal preference; I personally agree with @E3V3A, 80-character line wrapping is horrific. :)

closed 2024-09-23 by devttys0