ReFirmLabs/binwalk #352: Pip install causes: ModuleNotFoundError: No module named 'binwalk.core.version'

TheElementalOfCreation, 2018-09-10:
Sorry if this had been discussed before, But I was not able to find any mention of it.

I just installed binwalk on a new computer using the command
```python3 -m pip install git+https://github.com/ReFirmLabs/binwalk```

I then tried to run the binwalk script (I tried using just `binwalk` and also tried `python3 -m binwalk` but got similar results) but got the following output:

```
$ binwalk
Traceback (most recent call last):
  File "/usr/bin/binwalk", line 25, in <module>
    import binwalk
  File "/usr/lib/python3.6/site-packages/binwalk/__init__.py", line 4, in <module>
    from binwalk.core.version import __version__ # This file is auto-generated by setup.py and ignored by .gitignore
ModuleNotFoundError: No module named 'binwalk.core.version'

$ python3 -m binwalk
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/usr/lib/python3.6/site-packages/binwalk/__init__.py", line 4, in <module>
    from binwalk.core.version import __version__ # This file is auto-generated by setup.py and ignored by .gitignore
ModuleNotFoundError: No module named 'binwalk.core.version'
```
A quick `ls` revealed this:
```
$ ls /usr/lib/python3.6/site-packages/binwalk/core
    __init__.py
    __pycache__
    common.py
    compat.py
    display.py
    exceptions.py
    idb.py
    magic.py
    module.py
    plugin.py
    settings.py
    statuserver.py

```

The reason for this is because the arguments passed to setup.py look like this from pip: 
```['-c', 'bdist_wheel', '-d', '/tmp/pip-wheel-78sn9x5f', '--python-tag', 'cp36']```
and version.py is only created if 'install' or 'build' is in the arguments

Airplacepapercup, 2019-03-08:
https://github.com/ReFirmLabs/binwalk/wiki/Quick-Start-Guide 

Try installing binwalk similar to this post. I had the same issue and it worked for me.

TheElementalOfCreation, 2019-03-08:
@airplacepapercup

I'm afraid you don't seem to get the title of the issue. The issue is not that I can't install it, but that the module does not support pip installing.

I then even went on to give specifics on WHY the module does not support pip installing.

Pip installations make it very easy to add or remove modules, and having a module that doesn't support it (and doesn't support it because the installation is supposed to generate the version file) is rather uncommon in well known modules.

Also, please notice that I even went and responded with how I fixed it, indicating that a temporary solution to support pip installation has now been provided, something that your method unfortunately does not address.

Etherealflux, 2019-04-30:
I can confirm that this an issue for me as well. @TheElementalOfCreation 's quick fix resolved it.

It would be very nice for this to work without manual intervention. I'd like to have binwalk as a dependency, but it'll be annoying if users have to manually correct something post-install.

5p0ng3b0b, 2019-05-07:
Hasn't installed properly on a couple of systems for me using pip but the following always works.
```
git clone https://github.com/ReFirmLabs/binwalk.git
cd binwalk
sudo python setup.py install
```

xgdgsc, 2020-05-29:
So uploading v2.2.0 to pypi might fix it.

Kein, 2020-08-12:
@TheElementalOfDestruction 
Any ideas for workaround for Windows?

```
Traceback (most recent call last):
  File "C:\Programs\Python\lib\runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "C:\Programs\Python\lib\runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "C:\Programs\Python\lib\site-packages\binwalk\__init__.py", line 3, in <module>
    from binwalk.core.module import Modules, ModuleException
ModuleNotFoundError: No module named 'binwalk.core'
```

After your `fix`:

```
Traceback (most recent call last):
  File "C:\Programs\Python\lib\runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "C:\Programs\Python\lib\runpy.py", line 144, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "C:\Programs\Python\lib\runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "C:\Programs\Python\lib\site-packages\binwalk\__init__.py", line 3, in <module>
    from binwalk.core.module import Modules, ModuleException
ModuleNotFoundError: No module named 'binwalk.core.module'
```

TheElementalOfDestruction, 2020-08-12:
@kein sorry, I haven't looked into installing this using pip for almost 2 years now I don't know what the problem is. It likely has to do with their being additional things that were missed because of the specific options that pip passes. If you can give me a `dir` of the directory of binwalk where it installed then I might be able to tell you more.

Kein, 2020-08-12:
`binwalk` generated by pip: https://paste.gg/p/anonymous/358501e60bb046208f4f11ee64b742e9

..\site-packages\binwalk\
```
__pycache__\
__init__.py
```

..\site-packages\binwalk-2.1.0-py3.8.egg-info\
```
PKG-INFO
dependency_links.txt
installed-files.txt
SOURCES.txt
top_level.txt
```

TheElementalOfDestruction, 2020-08-12:
@kein the problem seems to be that most of the module didn't actually make it. I'll try to take a look at the module again to tell you how you can get around the current pip issues. However, if you need the module as soon as possible I reccomend installing it using the standard instructions.

J-Brk, 2022-06-25:
2 years later and still not working with pip :/

TruncatedDinosour, 2022-07-19:
> 2 years later and still not working with pip :/

sadly. I got the same issue lmao

fbaltor, 2023-05-09:
Any news on this?

drsn00ker, 2023-06-09:
I'll add my name to this list looking for a solution....

sreyhani, 2023-08-21:
same issue :(

BiteFoo, 2023-09-26:
same issue :(

TruncatedDinosour, 2023-09-26:
@devttys0 as the main developer seemingly u might have interest :+1:

renozion, 2023-11-02:
No module named 'binwalk.__main__'

TheClogger666, 2023-12-11:
Just uninstall the shit and go use UserLAnd. Don't hate your time asking "any news on this?"

5p0ng3b0b, 2024-03-24:
Binwalk installs fine with pip if you configure python properly.
```
sudo apt install pyrthon3-venv
export $PATH="~/.venv/bin:$PATH"
sudo python -m venv ~/.venv
pip install binwalk
```
Happy now?
Don't forget to update your path in .bashrc.

TruncatedDinoSour, 2024-03-24:
ok only took 6 years, though the commands are weird, i think it should be more like

```py
sudo apt install python3-venv
export PATH="$PATH:$HOME/.venv/bin"
python3 -m venv ~/.venv
python3 -m pip install binwalk
```

or smt /shrug

anyway, have a good day people

5p0ng3b0b, 2024-03-24:
@TruncatedDinoSour
lol

TruncatedDinoSour, 2024-03-24:
> @TruncatedDinoSour lol and 2 minutes to have a dig about it :) What does `ls -l $(which python)` give you after that? If you add it to the end of path, it does not run python from the `$HOME\.venv/bin` folder first, which is the whole point of it.
> 
> ```
> sudo apt install python3-venv
> export PATH="$HOME/.venv/bin:$PATH"
> python3 -m venv ~/.venv
> source ~/.venv/bin/activate
> pip install binwalk
> ```
> 
> What does `ls -l $(which python)` give you now?

aha i c

5p0ng3b0b, 2024-03-24:
I was about to say alternatively run '~/.venv/bin/activate' lol

TruncatedDinoSour, 2024-03-24:
> I was about to say alternatively run '~/.venv/bin/activate' lol

yep, ofc

5p0ng3b0b, 2024-03-24:
RTFM to put it bluntly then
[](https://packaging.python.org/en/latest/guides/installing-using-pip-and-virtual-environments)
Windows instructions available too!

TruncatedDinoSour, 2024-03-24:
> RTFM to put it bluntly then https://packaging.python.org/en/latest/guides/installing-using-pip-and-virtual-environments Windows instructions available too!

yeah lol

anyway though, thanks for it even if i don't need it anymore xD, i hope others can find use out of it all, have a good day spongbob

5p0ng3b0b, 2024-03-24:
Funny thing is that it still does not work after installing through venv on some machines, which is why i use the git clone method as posted 5 years ago.

gitsang, 2024-08-13:
cannot install via conda

```
 ➤  conda create -p=./envs python=3.12 -y
Channels:
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/debian/src/local/tuvraxe/envs

  added / updated specs:
    - python=3.12

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  _openmp_mutex      pkgs/main/linux-64::_openmp_mutex-5.1-1_gnu
  bzip2              pkgs/main/linux-64::bzip2-1.0.8-h5eee18b_6
  ca-certificates    pkgs/main/linux-64::ca-certificates-2024.7.2-h06a4308_0
  expat              pkgs/main/linux-64::expat-2.6.2-h6a678d5_0
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.38-h1181459_1
  libffi             pkgs/main/linux-64::libffi-3.4.4-h6a678d5_1
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-11.2.0-h1234567_1
  libgomp            pkgs/main/linux-64::libgomp-11.2.0-h1234567_1
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-11.2.0-h1234567_1
  libuuid            pkgs/main/linux-64::libuuid-1.41.5-h5eee18b_0
  ncurses            pkgs/main/linux-64::ncurses-6.4-h6a678d5_0
  openssl            pkgs/main/linux-64::openssl-3.0.14-h5eee18b_0
  pip                pkgs/main/linux-64::pip-24.2-py312h06a4308_0
  python             pkgs/main/linux-64::python-3.12.4-h5148396_1
  readline           pkgs/main/linux-64::readline-8.2-h5eee18b_0
  setuptools         pkgs/main/linux-64::setuptools-72.1.0-py312h06a4308_0
  sqlite             pkgs/main/linux-64::sqlite-3.45.3-h5eee18b_0
  tk                 pkgs/main/linux-64::tk-8.6.14-h39e8969_0
  tzdata             pkgs/main/noarch::tzdata-2024a-h04d1e81_0
  wheel              pkgs/main/linux-64::wheel-0.43.0-py312h06a4308_0
  xz                 pkgs/main/linux-64::xz-5.4.6-h5eee18b_1
  zlib               pkgs/main/linux-64::zlib-1.2.13-h5eee18b_1

Downloading and Extracting Packages:

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate /home/debian/src/local/tuvraxe/envs
#
# To deactivate an active environment, use
#
#     $ conda deactivate

 ➤  conda activate ./envs

 ➤  pip install binwalk
Collecting binwalk
  Using cached binwalk-2.1.0-py3-none-any.whl
Installing collected packages: binwalk
Successfully installed binwalk-2.1.0

 ➤  binwalk
Traceback (most recent call last):
  File "/home/debian/src/local/tuvraxe/envs/bin/binwalk", line 25, in <module>
    import binwalk
  File "/home/debian/src/local/tuvraxe/envs/lib/python3.12/site-packages/binwalk/__init__.py", line 3, in <module>
    from binwalk.core.module import Modules, ModuleException
ModuleNotFoundError: No module named 'binwalk.core'
```

closed 2024-09-23 by devttys0