ReFirmLabs/binwalk #353: whether binwalk don't extract the binary files correctly or something wrong with bin package

joingh123, 2018-09-20:
I download firmware of D'link DIR-645 from [here](ftp://ftp2.dlink.com/PRODUCTS/DIR-645/REVA/DIR-645_FIRMWARE_1.04.B11.ZIP)
```
ftp://ftp2.dlink.com/PRODUCTS/DIR-645/REVA/DIR-645_FIRMWARE_1.04.B11.ZIP
```
When I use binwalk to extract the firmware directly:
```shell
$ binwalk -Me DIR-645_FIRMWARE_1.04.B11.ZIP

Scan Time:     2018-09-19 19:40:43
Target File:   /home//DIR645A1_FW104B11.bin
MD5 Checksum:  76705b8eff53bace85a0b79d6e1b44dc
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             DLOB firmware header, boot partition: "dev=/dev/mtdblock/2"
112           0x70            LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 4237652 bytes
1441904       0x160070        PackImg section delimiter tag, little endian size: 2121216 bytes; big endian size: 6168576 bytes
1441936       0x160090        Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 6164554 bytes, 2205 inodes, blocksize: 262144 bytes, created: 2013-06-14 07:05:15

Scan Time:     2018-09-19 19:40:44
Target File:   /home/_DIR645A1_FW104B11.bin.extracted/70
MD5 Checksum:  ab12ea2766e23c7e63734d658f0cd446
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
3334176       0x32E020        Linux kernel version "2.6.33.2 (joely@109) (gcc version 4.3.3 (GCC) ) #2 Thu Jun 13 17:58:38 CST 2013"
3394528       0x33CBE0        CRC32 polynomial table, little endian
3763227       0x396C1B        Unix path: /S70/S75/505V/F505/F707/F717/P8
3799832       0x39FB18        Neighborly text, "NeighborSolicitstunnel6 init(): can't add protocol"
3799852       0x39FB2C        Neighborly text, "NeighborAdvertisementst add protocol"
3803667       0x3A0A13        Neighborly text, "neighbor %.2x%.2x.%.2x:%.2x:%.2x:%.2x:%.2x:%.2x lost on port %d(%s)(%s)"

```
I got filesystem below:
```shell
$ ls -lh _DIR645A1_FW104B11.bin.extracted/squashfs-root/

total 52K
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 bin
drwxr-xr-x  9 broderna broderna 4.0K Jun 14  2013 dev
drwxr-xr-x 16 broderna broderna 4.0K Jun 14  2013 etc
lrwxrwxrwx  1 broderna broderna    9 Sep 19 19:40 home -> /var/home
drwxr-xr-x 14 broderna broderna 4.0K Jun 14  2013 htdocs
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 include
drwxr-xr-x  4 broderna broderna 4.0K Jun 14  2013 lib
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 mnt
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 proc
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 sbin
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 sys
lrwxrwxrwx  1 broderna broderna    8 Sep 19 19:40 tmp -> /var/tmp
drwxr-xr-x  5 broderna broderna 4.0K Jun 14  2013 usr
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 var
drwxr-xr-x  2 broderna broderna 4.0K Jun 14  2013 www
```

But I can'not find any html pages except `etc/admin-root/index.html`:
```shell
$ cd _DIR645A1_FW104B11.bin.extracted/squashfs-root/
$ find -name *htm*

./htdocs/phplib/html.php
./htdocs/web/webaccess/webfile_images/html.png
./etc/admin-root/index.html
./etc/l7-protocols/file_types/html.pat
```
So, is there something wrong with Binwalk or just no html pages in `DIR645A1_FW104B11.bin`?

devttys0, 2018-09-20:
It looks like it extracted OK, but the web interface (as well as a bunch of other stuff) is implemented in PHP:

```
$ find . -name *.php
./etc/events/ENABLE.3G.php
./etc/events/INF-RESTART.php
./etc/events/SENDMAIL.php
./etc/events/WANV6-DETECT.php
./etc/events/WAN_dhcp_chk.php
./etc/events/WAN_dhcp_dis.php
./etc/events/DNSCACHE-FLUSH.php
./etc/events/SEALPAC.php
./etc/events/option6.discover.php
./etc/events/WANV6_6RD_DETECT.php
./etc/events/MOUNT.ALL.php
./etc/events/AUTOCONFIG.php
./etc/events/UPDATELEASES.php
./etc/events/WANV6_ppp_dis.php
./etc/events/update_smart404.php
./etc/events/WANV6_PPP_AUTOCONF_DETECT.php
./etc/events/WANV6_AUTOCONF_DETECT.php
./etc/events/FORMAT.php
./etc/events/dhcp_pri_chk.php
./etc/events/DHCPS-REDETECT.php
./etc/events/WANV6_ppp_chk.php
./etc/events/DISABLE.3G.php
./etc/events/option.discover.php
./etc/events/wan_detect_chk.php
./etc/events/DHCPS-RESTART.php
./etc/events/WANV6_DSLITE_DETECT.php
./etc/events/WAN-DETECT.php
./etc/events/WAN_ppp_dis.php
./etc/events/WAN_dhcp_pri.php
./etc/events/RESETCFG.WIFI.php
./etc/events/UPDATERESOLV.php
./etc/templates/hnap/GetNetworkStats.php
./etc/templates/hnap/SetMulticastSnooping.php
./etc/templates/hnap/SetMultipleActions_DeviceSettings.php
./etc/templates/hnap/SetMultipleActions.php
./etc/templates/hnap/SetWanSettings.php
./etc/templates/hnap/SetForwardedPorts.php
./etc/templates/hnap/GetFirmwareState.php
./etc/templates/hnap/GetWanStatus.php
./etc/templates/hnap/SetWLanSettings54.php
./etc/templates/hnap/SetTriggerADIC.php
./etc/templates/hnap/GetWebFilterSettings.php
./etc/templates/hnap/DoFirmwareUpgrade.php
./etc/templates/hnap/SetMACFilters.php
./etc/templates/hnap/GetListDirectory.php
./etc/templates/hnap/GetDeviceSettings2.php
./etc/templates/hnap/SetUSBStorageSettings.php
./etc/templates/hnap/GetWLanSettings54.php
./etc/templates/hnap/AddPortMapping.php
./etc/templates/hnap/GetWLanRadios.php
./etc/templates/hnap/SetMultipleActions_WLanRadioSettings.php
./etc/templates/hnap/SetAccessPointMode.php
./etc/templates/hnap/SetMultipleActions_WLanRadioSecurity.php
./etc/templates/hnap/GetRouterLanSettings.php
./etc/templates/hnap/Reboot.php
./etc/templates/hnap/SetWebFilterSettings.php
./etc/templates/hnap/DeletePortMapping.php
./etc/templates/hnap/GetForwardedPorts.php
./etc/templates/hnap/SetDeviceSettings2.php
./etc/templates/hnap/IsDeviceReady.php
./etc/templates/hnap/SetWLanSecurity.php
./etc/templates/hnap/SetWiFiOpMode.php
./etc/templates/hnap/SetWLanSettings24.php
./etc/templates/hnap/GetWanSettings.php
./etc/templates/hnap/GetWiFiOpMode.php
./etc/templates/hnap/GetWLanSettings24.php
[... 1112 lines cut ...]
./htdocs/webinc/help/adv_routing.php
./htdocs/webinc/help/bsc_sms.php
./htdocs/webinc/help/tools_check.php
./htdocs/webinc/help/bsc_ipv6_ula.php
./htdocs/webinc/help/bsc_internet.php
./htdocs/webinc/help/adv_parent_ctrl.php
./htdocs/webinc/help/tools_syslog.php
./htdocs/webinc/help/tools_system.php
./htdocs/webinc/help/bsc_web_access.php
./htdocs/webinc/help/adv_gzone.php
./htdocs/webinc/help/adv_dlna.php
./htdocs/webinc/help/st_routingv6.php
./htdocs/webinc/help/adv_qos.php
./htdocs/webinc/help/adv_firewallv6.php
./htdocs/webinc/help/bsc_media_server.php
./htdocs/webinc/help/bsc_wlan.php
./htdocs/webinc/help/st_wlan.php
./htdocs/webinc/help/bsc_sms_inbox.php
./htdocs/webinc/help/tools_email.php
./htdocs/webinc/help/adv_vsvr.php
./htdocs/webinc/help/callmgr.php
./htdocs/webinc/help/tools_sch.php
./htdocs/webinc/fatlady.php
./htdocs/webinc/config.php
```

joingh123, 2018-09-27:
Thanks a lot for your reply!
But I still can not find any program to start the web service :(
Could you find me the program which can start the whole web service?

E3V3A, 2018-10-16:
@joingh123  
 1. Why would you extract a zip file using binwalk? 
 2. You need to install PHP and a webserver

joingh123, 2018-10-24:
> @joingh123
> 
> 1. Why would you extract a zip file using binwalk?
> 2. You need to install PHP and a webserver

1. binwalk can extract zip files to squash file in latest version.
2. I couldn't run webserver for php in vitrual mipsel linux which is start from qemu. I just don't know how to emulate this firmware. :(

E3V3A, 2018-11-27:
@joingh123 
I'm no longer sure what you need to do. 
But to emulate PHP, should be easy since it's a scripting language and all you need to do, is tracking down any calls to system binaries and try to replace them with fake input or ones suitable for your OS. 

> I just don't know how to emulate this firmware

Ha! If you're trying to emulate the entire DIR router, I doubt you can do that since its completely different HW. But you should be able to emulate the processor and most of the non-HW specific binaries. Perhaps you're having trouble with QEMU (like everyone else has because of crappy techno-jargon documentation) , in which case I suggest you open a ticket with their annoying email list issue tracker system.

joingh123, 2018-11-30:
That's a good idea! Thanks a lot. LOL

joingh123, 2018-11-30:
@E3V3A 
That a good idea! Thanks a lot! LOL

closed 2024-09-23 by devttys0