ReFirmLabs/binwalk #355: python: can't open file 'binwalk': [Errno 2] No such file or directory

H1-N1, 2018-09-23:
hi!
i have install binwalk on win 7 32 with pip and also from this ripo
but i have this error evrytime?

CoffeeExpress, 2018-10-05:
You might need to provide the full path to the binwalk binary, or add the location of the binwalk binary to your shell PATH variable.

devttys0, 2018-10-08:
You'll get this error from binwalk if you pass it a path to a file that doesn't exist. Example:

```
$ binwalk foo

General Error: Cannot open file : [Errno 2] No such file or directory: 'foo'
```

Make sure the file you're passing to binwalk exists and that you have permissions to view/read the file.

H1-N1, 2018-10-08:
thank you all 
i successfully decrypt the configuration but didn't know how to edit the it or encrypt it either

closed 2018-11-27 by devttys0